pymeasure/pymeasure #416: Instrument's classes

msmttchr, 2021-03-20:
It is a reality, for various reasons, that similar instruments from different vendors exist in the market. So when we do a lab activity, we often use generic name, eg: oscilloscope and not Tektronix TDS2000. In general, unless there's a specific need, we consider instruments belonging to same class interchangeable and so, if one is broken or in use, we can use a different one (model or vendor or both) to do the job.
Obviously, human beings have normally the ability to adapt to new situation and they would use the new instrument by reading manual or guessing based on experience.
One may wonder why this is not accomplished in an instrument automation framework.
Indeed, SCPI instrument defines a set of generic commands valid for every instrument and a set of commands valid for some identified instrument  classes (see https://en.m.wikipedia.org/wiki/Standard_Commands_for_Programmable_Instruments and http://www.ivifoundation.org/docs/scpi-99.pdf).
I have also looked at some specific instrument implementation which is often basic and/or partial. I think there's nothing wrong with that, we are often very busy and we tend to implement what's strictly needed, an object oriented language like python allow to manage these situations without any problem.
My proposal is to identify some very well known instrument classes and starts implementing them as a generic one, and for each applicable vendor subclass the generic one.
This would have the advantage of:
- Reducing code duplication
- Making testing simple
- Allowing to grow the list of instruments quickly
- Allowing to use an instrument of a certain class even though its implementation is available
- Allowing to implement only vendor specific enhancements in the specific subclass.
- Define common API in one place and allow instrument reuse/replacement.

If you have time, you could have a look at the example in development for spectrum analyser and RF signal generators (https://github.com/wnluke/pymeasure/blob/d7deaf6c8517788a200762e2c1adb9de8bb564fa/pymeasure/instruments/rf_signal_generator.py or https://github.com/wnluke/pymeasure/blob/6e4d78d71c4d8ad35ae949bd7c7599c83a5d36da/pymeasure/instruments/spectrum_analyzer.py)
Here, even though the instruments are extremely complex and no SCPI class (for spectrum analyser) is defined, it is possible to define a common API.

My proposal on a step by step fashion is:
  1 identify instrument classes
  2 identify difference between SCPI instruments and non SCPI instruments
  3 Implement base classes
  4 Refactor existing specific instrument implementation based on base classes

Obviously hierarchy can be deeper than just one level, for example some time communalities between same vendor instruments exists.

See also @cjermain comment https://github.com/pymeasure/pymeasure/pull/254#pullrequestreview-370783460 and #376

untzag, 2021-03-20:
I don't have any background with pymeasure :confounded:---just a fan---but I've thought about this particular issue for a long time and I just wanted to share my humble thoughts.

I've personally found composition [1] to be the most productive approach for building hardware abstraction layers. In Python you can implement individual "atomic" interfaces as separate small classes and then build your device-specific classes through multiple inheritance. I have found that this approach allows me to enforce consistent behavior where possible while not inconveniencing me when dealing with hardware idiosyncrasies. Individual small pieces of interface can be added or removed at will depending on the hardware. We use a similar approach in yaq [2]. Perhaps a compositional system could be useful in pymeasure :smile:

[1] https://en.wikipedia.org/wiki/Composition_over_inheritance
[2] https://yaq.fyi/

msmttchr, 2021-03-20:
@untzag thanks for sharing your thoughts. I think composition is also a nice proposal. Lately I've seen in the market instruments that are indeed a composition of multiple instruments (e.g
 NI Virtual Bench). In such case, I believe composition is definitely the right way to go, even though I understand your idea/proposal is to use composition at finer granularity level.

paulgoulain, 2021-03-22:
Looking at yaq documentation, there are many great ideas ! One of them, that kinda answer to @msmttchr 's proposal is the ['traits concept](https://yaq.fyi/traits/). It would allow people to keep the (often) not uniform naming of similar functions from one driver to another, while providing a uniform way of accessing same functionalities from a class of instrument.
@untzag is there a complete sample script that use yaq deamons that describe the logic of an experiment ?

bilderbuchi, 2021-03-22:
OP: This is a good thought, and (in some ways) the thinking behind the IVI (Interchangeable Virtual Instrument) standard. Some of these ideas exist in https://github.com/python-ivi/python-ivi, but this is sadly unmaintained, and also crucially did not go as far re: instrument interchangeability that you mention (iirc). See also the related issue https://github.com/python-ivi/python-ivi/issues/50.
I recommend reading the long discussion we had on a similar topic a long time ago: https://github.com/pymeasure/pymeasure/discussions/407

I think many of your points are worthwhile to discuss/investigate, I would also appreciate a more homogeneous API over instrument "classes". Probably, though, the devil is in the details and capability variations between instrument models. Maybe that is why IVI does not seem that prevalent/used much? 
Using the given generic `RFSignalGenerator` class as an example: would all signal generators have "automatic leveling control", and would they all use the query command `":SOURCE:POWER:ALC?;"` (afaik this is not scpi-standardised)? I suspect the answer to both is "no", and that the "common subset" for instrument classes is quite limited.

Re composition vs. inheritance: Yes, in general this seems to be the common wisdom. Pymeasure also uses this in places, e.g. in the way that some instruments define channels for instruments. I wonder though about the effort to retool all the codebase, and if that is worthwhile or even realistically possibly (we are saturated as it is). Also, one thing that may throw a spanner in the works here is the heavy use of python properties, I have not yet thought how well that meshes with a composition-focused approach.

Also, on the point *"Allowing to use an instrument of a certain class even though its specific implementation is not available"*: I currently don't see how you want to accomplish that, since typically every vendor has their own command set/preferences, even among their own instrument line/variants?

I appreciate the thought behind this. I just don't know if it's realistic to accomplish the painted vision, so probably we have to limit the scope -- e.g., making the API/property names of our instruments more consistent is more easily achievable seems scaleable and would yield more immediate benefits.

untzag, 2021-03-23:
@paulgoulain I'm not 100% sure what you're asking, but I think you might be thinking about what pymeasure calls "procedures". yaq has no official answer for this---we are trying to focus on building a lightweight self-describing hardware interface layer that can be used by whatever orchestration layer a particular user might prefer. Much of the time orchestration looks like a simple little script, but we're also trying to build interfaces to existing larger projects like bluesky [1, 2] and pymodaq (WIP). More automated "self-building" interfaces rely more on the traits system.

Sorry to muddy up this issue with yaq stuff---that wasn't my intention :sweat_smile:. Maybe someday I can write a pymeasure adapter for yaq.

[1] https://blueskyproject.io/
[2] https://github.com/bluesky/yaqc-bluesky

bilderbuchi, 2021-03-23:
FWIW, pymeasure is also two things in one package - the GUI/procedure stuff and the instrument drivers (the latter are largely independent from the former).

paulgoulain, 2021-03-26:
@untzag I understood that, you provide only the communication with the instruments, the user has to implement the procedure logic. The question was do you have a "stupid" script that connects to an instrument, and asks it something, returns a value, anything ^^

@bilderbuchi I agree it would be useful to make the instruments that do the same kind of measurement using a unify API, at least for the basic functionalities. For instance, I have an IV script that I use with 3 different Keithley model (2401, 2461 and 2636). I had to make a lot of tweaking to perform this IV measurement, independently of the SMU I connect.

untzag, 2021-03-26:
@paulgoulain https://python.yaq.fyi/installing-yaq/#communicate-with-your-daemon

msmttchr, 2021-03-28:
> OP: This is a good thought, and (in some ways) the thinking behind the IVI (Interchangeable Virtual Instrument) standard. Some of these ideas exist in https://github.com/python-ivi/python-ivi, but this is sadly unmaintained, and also crucially did not go as far re: instrument interchangeability that you mention (iirc). See also the related issue [python-ivi/python-ivi#50](https://github.com/python-ivi/python-ivi/issues/50).
> I recommend reading the long discussion we had on a similar topic a long time ago: #407
> 
> I think many of your points are worthwhile to discuss/investigate, I would also appreciate a more homogeneous API over instrument "classes". Probably, though, the devil is in the details and capability variations between instrument models. Maybe that is why IVI does not seem that prevalent/used much?
> Using the given generic `RFSignalGenerator` class as an example: would all signal generators have "automatic leveling control", and would they all use the query command `":SOURCE:POWER:ALC?;"` (afaik this is not scpi-standardised)? I suspect the answer to both is "no", and that the "common subset" for instrument classes is quite limited.

Well, as matter of fact ALC is present in the SCPI specification (section 5.1 Volume 1 for query form and  8.1.2.1 in Volume 4) and present in all the instruments we have in our lab. I do not pretend to have a  set representative of all RF signal generators, but they are from main manufacturer. On the other hand, I am old enough to not bet that all instruments follow specification in details.
Again, what's the problem here ? If we can find commonalities between instrument classes and exploit them saving a lot of code duplication why stopping because some instrument could not follow the rule ? In such case, the implementation specific for that instrument can patch the problem.

No matter what implementation strategy, we follow, I think it is useful to be pragmatic and face reality. I think all the RF generators will allow you to set the frequency, to set the power, to enable/disable output. The list presented in the example is very basic, but also very useful in general. It could be expanded with contribution of other people and by reading specifications.

I would also consider the digital modulation implementation as an interesting example on how different vendor specific commands can be exploited to achieve the same function.
> Re composition vs. inheritance: Yes, in general this seems to be the common wisdom. Pymeasure also uses this in places, e.g. in the way that some instruments define channels for instruments. I [...]

If the vendor follow SCPI specification, the generic class for that instrument should allow people to use it without creating a specific class. Of course, we cannot guarantee that it will work. Also, if user want to use a vendor specific feature of a certain instrument, then here we have no choice, the experiment setup should mandate to use a specific instrument and replacing it is not an option.

> I appreciate the thought behind this. I just don't know if it's realistic to accomplish the painted vision, so probably we have to limit the scope -- e.g., making the API/property names of our [...]

Well, I think we are saying the same thing here, what proposed is not a revolution, but just a factorization of what is already available, or did I misunderstand you ?
Nevertheless, I personally see a value in @untzag tag proposal, but I think this could require more effort.

bilderbuchi, 2021-04-27:
I am not opposed to the proposal on principle, and for sure reducing duplication would be very attractive. My assessment is driven by a couple of different experiences.

On ALC: Thanks, I was not aware that it's part of the specification (but then, I haven't read it). 
At my previous job we had a frequency generator where I'm pretty sure that it doesn't have ALC (granted, it's not a main-brand vendor). So, that immediately suggested to me that the practical amount of common functions could be quite limited. Is there a graceful way to deal with the situation when a device does not implement all the properties defined in the common class definition? We should not just delete a defined property, as that would violate all kinds of (API) expectations (e.g. [Liskov](https://en.wikipedia.org/wiki/Liskov_substitution_principle), I think). Do we just error out/throw?
I agree that composition or traits/mixins over inheritance could probably help with this, but I think it's a huge undertaking to retool pymeasure for this.

My point that device drivers based off of a generic class must be able to tweak the commands used (even if they are in SCPI) is grounded in my experience with a couple of different power supplies and source meters. With our current property factories, I don't see a good way to effect such a tweak.
I can't offer the concrete strings/models, but I remember instances where the same "basic" command (e.g. to read current) used different commands. With different vendors that can be expected (there are a couple of "interesting" command definitions out there), but it's a problem that the implementation of this issues _has to be able_ to deal with. In addition, this was even a problem (iirc) if they are all from the same vendor (e.g. Keithley) because the tree structure of command groupings is not consistent over all instruments, and it's reflected in the command string.

Overall, I am a bit concerned that we would bite off more than we can chew with this. IVI is a whole framework/foundation revolving around this afaict. Python-ivi tried to pull an implementation off and went not really anywhere. We are already resource constrained as is, and this is only one aspect of the project.
Incremental changes paving the way, e.g. making sure our command names are more homogeneous (and the old ones properly deprecated) across instrument implementations, are absolutely worthwhile as a first step in the right direction, though!

msmttchr, 2021-05-03:
> I am not opposed to the proposal on principle, and for sure reducing duplication would be very attractive. My assessment is driven by a couple of different experiences.
>
> On ALC: Thanks, I was not [...]

My opinion is that we should throw a "Not supported" exception. We could define a generic "not implemented" property to be used in such situations.

> I agree that composition or traits/mixins over inheritance could probably help with this, but I think it's a huge undertaking to retool pymeasure for this.
>
> My point that device drivers based off of [...]

Absolutely in agreement with You and I can give an example with Keysight and Rohde&Schwarz spectrum analyzers for the command to set/get the trace mode:

- Keysight: `":TRACe:MODE?;" `for getting and `":TRACe:MODE %s;"` for setting
- Rohde&Schwarz (some models): `"DISPLAY:TRACe:MODE?;" `for getting and `"DISPLAY:TRACe:MODE %s;"` for setting

Very minor, insignificant difference, but good enough to prevent an experiment to run successfully when a spectrum analyzer is swapped.

Here, my proposal is to push forward PR #376, so that Rohde&Schwarz implementation can be managed by using dynamic properties, i.e:
```
    trace_mode_get_command = "DISPLAY:TRACe:MODE?;"
    trace_mode_set_command = "DISPLAY:TRACe:MODE %s;"

```
> 
> Overall, I am a bit concerned that we would bite off more than we can chew with this. IVI is a whole framework/foundation revolving around this afaict. Python-ivi tried to pull an implementation off and went not really anywhere. We are already resource constrained as is, and this is only one aspect of the project.
> Incremental changes paving the way, e.g. making sure our command names are more homogeneous (and the old ones properly deprecated) across instrument implementations, are absolutely worthwhile as a first step in the right direction, though! A complete overhaul towards common device classes and/or traits/tags/composition are probably too big to pull off.

Again, absolutely in agreement with You and I think that a fundamental prerequisite to allow some form of incremental changes is to push PR #376 through.

bilderbuchi, 2021-06-22:
In #453, we had a detailed discussion [starting here](https://github.com/pymeasure/pymeasure/pull/453#discussion_r650675731_) about how to improve the abstraction regarding SCPI features with `includeSCPI`. 
As this is intricately connected to this issue concerning the introduction of instrument categories, we decided to solve both topics in the scope of this issue.

As a quite abbreviated summary, the two most promising approaches so far identified are:
* Add an `SCPIInstrument` class derived from `Instrument`, which contains all common SCPI commands. Base all instrument categories on this class. Non-SCPI instruments should derive from `Instrument` directly. Unresolved issue: how to obtain API consistency (which is actually independent from the various instruments being SCPI-conformant or not) for categorical instruments?
* Add SCPI commands, and potentially category information/command APIs as Mixins instead of plain base classes. Would speak more to the "trait-like" quality of these things. Downside: More complicated for contributors implementing instruments, as the order of specified Mixins, MRO, and correct handling of `super()` become important.

bilderbuchi, 2021-06-26:
https://hynek.me/articles/python-subclassing-redux/ : A long essay on subclassing techniques/ design patterns, worth the read to expand your horizon, or even to only put names to things you have seen before. To wit: InstrumentKits elegant approach to logging reads/write to/from an instrument (for all adapter types) uses the template method pattern.

bilderbuchi, 2022-01-29:
Further pointers, to a set of generic base classes in InstrumentKit: 
https://instrumentkit.readthedocs.io/en/latest/apiref/instrument.html#instrument-base-classes
https://instrumentkit.readthedocs.io/en/latest/apiref/generic_scpi.html

Could be useful as a source of inspiration, in my experience the IK are quite thorough regarding API design.

bilderbuchi, 2022-01-30:
It looks to me, also re: #580, that it would probably make sense to sublass `Instrument` to `SCPIInstrument` (for the SCPI-only properties), and then derive `GenericSCPIPowersupply` and other classes from that, which define a common base API/properties. This could happen at any time from now.

Then we could slowly adapt the existing instruments to the new scheme. As that would break backwards compatibility, that should happen after some new release (to give users a current-ish fallback in case they run into problems), but before 1.0 (because such a structural issue should be resolved by 1.0).

bilderbuchi, 2022-01-30:
One thing we can/should also resolve while we're at it is channel implementations: How do we account for channels in the generic instruments? 
Do they all have 1 channel in the background, and property access automatically gets delegated if there's only one channel (I think IK works that way nowadays) -- more channels then get addressed like `myinstrument.channel[0].voltage`.

bilderbuchi, 2022-01-30:
I agree that this issue is worth implementing. With #376 merged we are also in a much better place to deal with instrument differences than before.
However, I'm afraid I don't have enough bandwidth for dealing with this right now. 
@msmttchr I think by now you know what style/philosophy we are looking for in the codebase. Do you want to take the lead in resolving the questions, deciding on the best way forward, and implementing this? Probably makes sense to split it up into multiple PRs to limit their size.

I think the big questions to answer in here are
* [ ] Composition/traits (yaq-style) or inheritance for defining the instrument feature hierarchy? Consider also how much a contributor needs to know/account for to correctly implement an instrument
* [ ] How do we implement channels to cleanly allow for multichannel instruments?
* [ ] What is the instrument hierarchy gonna look like? SCPI/non-SCPI; if we have generic instrument classes are those only for SCPI or have a wider scope (where we at least can prescribe a standard set of property names), or both (like IK seems to have)

msmttchr, 2022-02-01:
@bilderbuchi, thanks for sharing your thoughts, let me share my opinion on this:
- I agree this is a potentially very huge work, so splitting it is very recommended
- I would favor composition over inheritance, especially after reading the fine essays you shared a while ago.
- Channel's implementation, I would include this in the scope of #567 and in particular after implementation of read/write/ask/values methods is clarified.
- Instrument hierarchy: we could have a composition approach here, my current thinking:
  * `class MySCPISpectrumAnalyzer(SCPIInstrument, SpectrumAnalyzer):` This instrument is an SCPI compliant spectrum analyzer
  * `class MyNonSCPISpectrumAnalyzer(SpectrumAnalyzer):` This instrument is a non SCPI compliant spectrum analyzer

`SCPIInstrument` is a class which includes a subset of the mandatory SCPI commands (i.e. the one currently implemented in the `Instrument` class)
`SpectrumAnalyzer` is a class which includes the base interface for a generic spectrum analyzer.

In addition to this, I would try to identify some implementation guidelines for base instrument classes:
- All the instrument features intended for interactive use does not need to be implemented in the base instrument class. (ie. controlling the brightness of the display or grid on/off on display, this list can be very long...). Rationale is that PyMeasure usage of the instrument is automatic and often remote.
- All the features that can be implemented in software, they should not go in the base implementation. This will increase compatibility between instruments. I could imagine an averaging function or similar done in python as opposed to use the specific hardware implementation. If a performance penalty is seen, the possible hardware implementation of the averaging can be implemented in the actual instrument derived from base instrument.
- Any advanced/fancy feature can be implemented in the actual instrument, if desired. As a last resort they can be used with basic read/write method's.

msmttchr, 2022-04-19:
@bilderbuchi, I have started a discussion #617 based on the information collected during the previous months.
For sure it is a significant work, but I would like a feedback from you and other people involved, before a coding can be started (possibly with more people involved :) ).

bilderbuchi, 2022-10-03:
Oscilloscopes might be a good early candidate for harmonization, the LeCroy T3DSO120 (#697), Keysight DSOX1102G and Tektronix AFG3152C `Channel` objects are quite similar/derived from each other. cc @LastStarDust

LastStarDust, 2022-10-04:
@bilderbuchi In the past I worked with a piece of software called [pyrame](https://llr.in2p3.fr/sites/pyrame/), that implemented a sort of abstraction layer for various devices. For example all power supplies would inherit from a common generic power supply ancestor.
I enjoyed using that part of the software and I think it lived up to its goal. However that team lost basically all developers and development stopped, leaving many features half-done or todo.
So my two cents are that an abstraction layer can be beneficial for this kind of use case ... but you already know that.

bilderbuchi, 2022-11-20:
See #751 for a discussion on if/how we could bring Modbus instruments into pymeasure.

bmoneke, 2022-11-20:
In #659 we also talked about separating SCPI from the `Instrument`.

I did not like the idea to create a `SCPIInstrument` as a subclass of Instrument, because you would have to use `SCPIInstrument.control` etc.

Therefore (as a rough idea), I preferred the option of a `SCPIMixin`, such that you subclass `Instrument` for the communication and `SCPIMixin` for the default properties/methods.

If we separate the communication (`Instrument`) from the properties/methods (`SCPI`, `Oscilloscope` or whatever Mixin) in a single mixin, it should work flawlessly.

bilderbuchi, 2022-11-20:
> in a single mixin

Why this restriction? I see Mixins kinda like/for "traits", why not include more? (E.g. an SMU would have all PSU-like things and more on top)

bmoneke, 2022-11-20:
For a single mixin, I think I can exclude conflicts, therefore, "flawlessly".
For more mixins, we have to be more careful lest there are conflicts, but I did not want to exclude them.

bilderbuchi, 2023-01-08:
One thought I had is that it might be most fruitful to mix mixins/traits and classes, and have generic instrument types composed of some common "characteristics" + specific stuff.

For example, there might be "traits" (I'll use that term for simplicity, but could be mixins or another mechanism) `has_output`, `senses_voltage`, `sets_voltage`, `sets_frequency`.
The `has_output` trait could also define a `shutdown()` method that automatically disables the output. IF that works as I think (e.g. hands control to the next `shutdown` method in the MRO), we could get an "automatically composed" shutdown that aggregates appropriate actions from all instrument qualities.

A `PowerSupply` would use `has_output`, `senses_voltage`, `sets_voltage`.
A `Multimeter` would use `senses_voltage`.
A `FrequencyGenerator` would use `has_output`, `sets_frequency`.
etc.

bmoneke, 2023-01-08:
Oh, good idea, that solves another question I had regarding mixins. The example #580 creates an instrument subclass, which can be used out of the box. An analyzer mixin class (trait) could not be used out of the box (as it lacks the Instrument parts).

If we have traits and classes implementing these traits, this problem is solved.

> The has_output trait could also define a shutdown() method that automatically disables the output. IF that works as I think (e.g. hands control to the next shutdown method in the MRO), we could get [...]

It should work as expected (calling all shutdown methods), but I see the difficulty to know, which properties are outputs (or do "set_voltage_trait" define the outputs in its shutdown method?).

bilderbuchi, 2023-01-08:
Let use continue discussions on the generic instrument _types_ and _characteristics_ we would like to use in #825 (technical implementation details remain in here). 
This should help keep the discussion herein focused.

bilderbuchi, 2023-01-08:
> I see the difficulty to know, which properties are outputs (or do "set_voltage_trait" define the outputs in its shutdown method?).

TBH, I was thinking primarily of having "an output" (like in a power supply). The trait `has_output` would define a `output_enabled` property and the `shutdown` logic pertaining to that. I'm not sure where the fact if it's a "voltage output" (or alternatively current output?) comes in?