pymeasure/pymeasure #419: Timeout expired before operation completed

Thermoelectric-CINTRA, 2021-03-24:
from pymeasure.instruments.keithley import Keithley2182A
import time
nano = Keithley2182A("GPIB0::15")
nano.measurement = "Voltage"
nano.channel = 1
data1 = nano.read_instrument
print("Last Instrument reading: ", data1)

data2 = nano.read
print("New reading: ", data2)

nano.buffer_size = 100
print("Buffer size: ", nano.buffer_size)

nano.buffer_source = "Sense"
# print("Buffer reading source: ", nano.buffer_source)

nano.buffer_control = "Start"
nano.initiate = "Enable"

while nano.reading_avaiable():
    print("Nanovolt meter triggering ...")
    time.sleep(0.1)
# time.sleep(2)
# print("Measurement even: ", nano.reading_avaiable())
print("Measurement done!")

datas = nano.get_data()
print(datas)
--------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "d:\PythonTestCode\test-keithley2812a.py", line 28, in <module>
    datas = nano.get_data()
  File "d:\cintra\pymeasure\pymeasure\instruments\keithley\keithley2182a.py", line 74, in get_data
    datas = self.values(":TRAC:DATA?")
  File "d:\cintra\pymeasure\pymeasure\instruments\instrument.py", line 111, in values
    return self.adapter.values(command, **kwargs)
  File "d:\cintra\pymeasure\pymeasure\adapters\adapter.py", line 81, in values
    results = str(self.ask(command)).strip()
  File "d:\cintra\pymeasure\pymeasure\adapters\visa.py", line 114, in ask
    return self.connection.query(command)
  File "C:\users\photonics_lab\appdata\local\programs\python\python39\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\users\photonics_lab\appdata\local\programs\python\python39\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\users\photonics_lab\appdata\local\programs\python\python39\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\users\photonics_lab\appdata\local\programs\python\python39\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\users\photonics_lab\appdata\local\programs\python\python39\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\users\photonics_lab\appdata\local\programs\python\python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

neal-kepler, 2021-03-26:
This issue is a little haphazardly written, but I think can offer a suggestion. It looks like when you go to read the data, the device doesn't finish responding in time.  That mean's that either the ":TRAC:DATA?" command was received but the instrument had nothing to produce, or so much data was produced that you reached the time out. Provided `nano.reading_avaiable`  is not a typo, it looks like that there is not check to see if there is actually data available, just a printing of the response. Do you get a sensible response for `nano.reading_available`?

bilderbuchi, 2021-04-18:
I added code block backticks( ```) to make this more readable.
Also note that this instrument does not seem to be from pymeasure itself - I guess it was added by the reporter?
I agree with the analysis above; although there is no other output given from the script, the fact that you don't run into a timeout earlier seems to suggest that the connection is working properly, only something with that one command is off.

bilderbuchi, 2021-04-18:
I'll move this into the discussion section.

closed 2021-04-18 by bilderbuchi