pymeasure/pymeasure #422: System Crash on Big Sur with Plotter

dph3dg, 2021-03-29:
I'm trying the tutorials and the moment I call plotter.start() in the example I get a segmentation fault.
The following is the errors thrown before the segmentation fault:
WARNING: QApplication was not created in the main() thread.
2021-03-29 14:21:59.858 python[23493:824564] WARNING: NSWindow drag regions should only be invalidated on the Main Thread! This will throw an exception in the future. Called from (
	0   AppKit                              0x00007fff22c8947f -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 352
	1   AppKit                              0x00007fff22c74121 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1375
	2   AppKit                              0x00007fff22c73bbb -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	3   AppKit                              0x00007fff22f7e1f4 -[NSWindow initWithContentRect:styleMask:backing:defer:screen:] + 52
	4   libqcocoa.dylib                     0x000000011db44df0 qt_plugin_instance + 20859
)
2021-03-29 14:21:59.881 python[23493:824564] WARNING: nextEventMatchingMask should only be called from the Main Thread! This will throw an exception in the future.

CasperSchippers, 2021-05-21:
Hi. Thanks for raising this issue. What I understand is that the latest macOS don't like it when you try to render something (in this case the plotter-window) from anything else than the Mean Thread. While I'm trying to find a solution, I want to note that it is rather difficult for me as I don't have access to macOS.

Could you tell us if this problem only arises in this plotter example, or also with other examples, such as the ManagedWindow?

dph3dg, 2021-05-24:
I used the plotter example as it was the 1st one, so I assumed it was the easiest. The managed window opens as the example and I get a moving plot al'a Labview. The progress bar doesn't seem to update, just shows if it's finished or not.

CasperSchippers, 2021-05-24:
Yes the plotter is supposed to be the easiest (or at least most stripped down) version, but apparently the way it is called gives some problems on the newest macOS.

That the progress bar does not show the progress in this example is correct. It seems the example omits the line responsible for updating the progress: `self.emit('progress', 100*i/iterations)`; in the examples folder on GitHub this line is included in the GUI example.

msmttchr, 2021-08-01:
I have actually discovered that also in `Ubuntu 20.04.2 LTS` with `python 3.8.10` and `PyQt5`, `script_plotter.py` does not work.
I see black screen and graph not shown. If I use `PySide2`, the script works as expected.
Unfortunately, this has probably uncovered something substantial on the `Plotter` class. Indeed Qt documentation (https://doc.qt.io/qt-5/thread-basics.html#gui-thread-and-worker-thread) says that:

> The Qt GUI must run in main thread. All widgets and several related classes, for example QPixmap, don't work in secondary threads

The class `Plotter` is designed to be run as a secondary thread inheriting from `StoppableThread` and so it is inherently violating the `Qt` constraints.

Possibly a redesign of `Plotter` class is needed.

CasperSchippers, 2021-08-12:
Good catch @msmttchr. Interesting to see that it doesn't work with PyQT by does with PySide. The fact that it is not allowed according to the QT documentation makes that I think we should indeed redesign the Plotter. As the problem also arises on Ubuntu it will also be a bit easier to check whether it works correctly.
I can have a look at redesigning the Plotter.

msmttchr, 2021-08-12:
To be accurate, PySide2 seems to work but throw some error message (I don't recall which one and I am not able to run it now) when you close the application.