pymeasure/pymeasure #426: instruments.shutdown() dangerous?

misterjay92, 2021-04-13:
Hey,

i can't find the definition of the instruments.isShutdown property and the method instrument.shutdown() does not even turn the output of my power supply Keithley2260B off. This might be dangerous for someone who trusts too blindly.

Best regards,
Jakob

bklebel, 2021-04-14:
Hi, I am happy someone has a use for this class! 
However, this instrument was implemented in a minimal way, due to my time constraints as a researcher. In this sense, I think it important not to trust any instrumentation code blindly - not always all possible API possibilities might be implemented.
 
The `isShutdown` property is defined (as I just found out it exists) in the `Instrument` class, separate from any concrete instrument implementation. As you found out, the `instrument.shutdown()` function does nothing special except setting this parameter to `True`, except if it was implemented directly with the respective instrument. 

I have proposed a quick fix in PR #427 - as you will see, the changes are small, and contributions are very welcome, so please consider writing PR's for similar issues you might find, this would help a lot, every small thing counts! 

I hope you will find this a useful addition :)

misterjay92, 2021-04-14:
Hello :)

i am very happy you already fixed this :)
I extended the class with some source-mode and slew-rate properties, but i
guess before one can have a proper merge on github, one has to put a lot
more work in there. Thanks for the work you put in there even next to your
research! :)
I will try to thank you and the community by contributing to your class and
maybe make a Keithley DMM6500 class.

Best regards,
Jakob

closed 2021-04-14 by misterjay92

bklebel, 2021-04-14:
Well, as of now it is not yet really fixed, I just made a PR with needs to be reviewed, which is only right and proper. 

If you have additions to the code from which others might benefit, why not go ahead and make a PR? in case you feel like you're not yet finish with the additions you want to make, you can mark it as "draft", so the maintainers will not jump on it immediately. Whether your PR will be merged will be up to the maintainers anyways, and it will always take some time to be reviewed. Additionally, if someone else needs this functionality and would be about to implement it themselves, they know you're already working on it, and don't have to do the work twice.  
So....don't hesitate to put something out there :)

bilderbuchi, 2021-04-18:
> so the maintainers will not jump on it immediately

Ha, no danger of that! The way our availability looks, nobody is jumping on anything, much less immediately :-P