pymeasure/pymeasure #428: tests fail because of QT

LongnoseRob, 2021-04-18:
Hello,
I was working on implementing a class for the HP3478A non-scpi multimeter.
After I had my implementation working, I tried the unit tests as recommended in the contribution section of the documentation.

with my system running on Archlinux I get the following errors, which seem to be related to some recent changes in the QT support modules:

`============================================================================================= short test summary info =============================================================================================
ERROR tests/display/test_inputs.py - AttributeError: module 'PySide2.QtCore' has no attribute 'pyqtSignal'
ERROR tests/display/test_plotter.py - AttributeError: module 'PySide2.QtCore' has no attribute 'pyqtSignal'
ERROR tests/display/test_windows.py - AttributeError: module 'PySide2.QtCore' has no attribute 'pyqtSignal'
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 3 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
========================================================================================== 2 skipped, 3 errors in 31.27s ==========================================================================================
`

full console log is attached

version infomation on my used python etc:
`platform linux -- Python 3.9.3, pytest-6.2.3, py-1.10.0, pluggy-0.13.1
PySide2 5.15.2 -- Qt runtime 5.15.2 -- Qt compiled 5.15.2
rootdir: /home/robby/builds/pymeasure
plugins: qt-3.3.0
`
[20210418_test_console_log.txt](https://github.com/pymeasure/pymeasure/files/6330342/20210418_test_console_log.txt)

msmttchr, 2021-04-18:
This is probably due to some conflict between PyQt5 and PySide2.
I could suggest to have a look at this link https://pyqtgraph.readthedocs.io/en/latest/how_to_use.html#pyqt-and-pyside.

msmttchr, 2021-04-19:
Checked on my PC and confirmed your problem:
```
Python 3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import PySide2
>>> from pyqtgraph.Qt import QtGui, QtCore, loadUiType
>>> QtCore.pyqtSignal
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'PySide2.QtCore' has no attribute 'pyqtSignal'
```
Looking at the `pyqtgraph ` code, I would say that is a bug in `pymeasure ` file `Qt.py` and the fix is to replace line:
`QtCore.QSignal = QtCore.pyqtSignal
` to
`QtCore.QSignal = QtCore.Signal
`

LongnoseRob, 2021-04-19:
@msmttchr Thank you for the confitmation.

WIth the changes applied as recommended, the tests run and some failures still occru especially  
`tests/display/test_inputs.py:48: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
pymeasure/display/inputs.py:156: in __init__
    if qt_min_version(5):
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

major = 5, minor = 0

    def qt_min_version(major, minor=0):
        """
        Check for a minimum Qt version. For example, to check for version 4.11
        or later, call ``check_qt_version(4, 11)``.
    
        :return bool: True if PyQt version >= min_version
        """
>       return (QtCore.QT_VERSION >= ((major << 16) + (minor << 8)))

E       AttributeError: module 'PySide2.QtCore' has no attribute 'QT_VERSION'` 

Looks like there is some more Qt stuff that needs to be fixed..

msmttchr, 2021-04-19:
Could You please try patching also the `qt_min_version`  function as follow ?

```
def qt_min_version(major, minor=0):
    """
    Check for a minimum Qt version. For example, to check for version 4.11
    or later, call ``check_qt_version(4, 11)``.

    :return bool: True if PyQt version >= min_version
    """
    version = QtCore.qVersion().split(".")
    version_major = int(version[0])
    version_minor = int(version[1])
    return (version_major > major) or ((version_major == major) and  (version_minor >= minor))

```

msmttchr, 2021-04-20:
@LongnoseRob, I think the problem is a little bit more deep than I initially thought.
In any case, it could make sense to raise a PR to fix the PySide2 support, but I am not clear whether this is actually supported by the project or not. In the requirements file https://github.com/pymeasure/pymeasure/blob/master/requirements/pymeasure.yml , I read:  `pyqt=5.12.3`

In any case, I have everything running properly on my side after updating the files attached.
Could You please give a go to the files and confirm that also on your side everything looks good?

[PySide2_fixes.zip](https://github.com/pymeasure/pymeasure/files/6342435/PySide2_fixes.zip)

LongnoseRob, 2021-04-20:
@msmttchr 

I just did the test with the files from your zip-file.

looks very fine for me now:
`
python setup.py test
running pytest
Searching for pyvisa-sim>=0.4.0
Best match: PyVISA-sim 0.4.0
Processing PyVISA_sim-0.4.0-py3.9.egg

Using /home/robby/builds/pymeasure/.eggs/PyVISA_sim-0.4.0-py3.9.egg
Searching for pytest-qt>=2.4.0
Best match: pytest-qt 3.3.0
Processing pytest_qt-3.3.0-py3.9.egg

Using /home/robby/builds/pymeasure/.eggs/pytest_qt-3.3.0-py3.9.egg
Searching for stringparser
Best match: stringparser 0.5
Processing stringparser-0.5-py3.9.egg

Using /home/robby/builds/pymeasure/.eggs/stringparser-0.5-py3.9.egg
running egg_info
writing PyMeasure.egg-info/PKG-INFO
writing dependency_links to PyMeasure.egg-info/dependency_links.txt
writing requirements to PyMeasure.egg-info/requires.txt
writing top-level names to PyMeasure.egg-info/top_level.txt
adding license file 'LICENSE.txt' (matched pattern 'LICEN[CS]E*')
adding license file 'AUTHORS.txt' (matched pattern 'AUTHORS*')
reading manifest file 'PyMeasure.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
no previously-included directories found matching 'docs/_build'
warning: no previously-included files matching '__pycache__' found under directory '*'
writing manifest file 'PyMeasure.egg-info/SOURCES.txt'
running build_ext
=============================================================================================== test session starts ===============================================================================================
platform linux -- Python 3.9.3, pytest-6.2.3, py-1.10.0, pluggy-0.13.1
PySide2 5.15.2 -- Qt runtime 5.15.2 -- Qt compiled 5.15.2
rootdir: /home/robby/builds/pymeasure
plugins: qt-3.3.0
collected 103 items / 2 skipped / 101 selected                                                                                                                                                                    

tests/test_log.py ...                                                                                                                                                                                       [  2%]
tests/test_process.py ..                                                                                                                                                                                    [  4%]
tests/test_thread.py ..                                                                                                                                                                                     [  6%]
tests/adapters/test_adapter.py ..                                                                                                                                                                           [  8%]
tests/adapters/test_prologix.py ....                                                                                                                                                                        [ 12%]
tests/adapters/test_serial.py ....                                                                                                                                                                          [ 16%]
tests/adapters/test_visa.py .....                                                                                                                                                                           [ 21%]
tests/display/test_inputs.py ................................                                                                                                                                               [ 52%]
tests/experiment/test_parameters.py ..........                                                                                                                                                              [ 62%]
tests/experiment/test_procedure.py ..                                                                                                                                                                       [ 64%]
tests/experiment/test_results.py .......                                                                                                                                                                    [ 70%]
tests/experiment/test_workers.py ......                                                                                                                                                                     [ 76%]
tests/instruments/test_instrument.py ...............                                                                                                                                                        [ 91%]
tests/instruments/test_validators.py ........                                                                                                                                                               [ 99%]
tests/instruments/keithley/test_keithley2750.py .                                                                                                                                                           [100%]

========================================================================================= 103 passed, 2 skipped in 36.15s =========================================================================================
`

thank you very much!

LongnoseRob, 2021-04-22:
@msmttchr  I will close this issue as you already have started #431

closed 2021-04-22 by LongnoseRob

bilderbuchi, 2021-04-23:
Thanks for the report! To confirm a couple of points:
Yes we don't care if you use pyside or pyqt, just like pyqtgraph. If we can easily enable both, that's good.
We only test one of those in CI, in this case pyqt.
Yes it's fine to remove support qt<=4.

Thanks for the PR!

msmttchr, 2021-04-25:
Is it possible to add `PySide2` in CI, in addition to `PyQt5` ?

bilderbuchi, 2021-04-26:
Possible, sure. It would make the setup more complex: another axis to the build matrix (so twice the builds), an additional environment file. Ideally I would like to keep the number of builds to a minimum, so maybe add only one test (latest python, ubuntu machine, pyside2) for this.