pymeasure/pymeasure #440: Keithley2000 instrument constructor error

martinjankoehler, 2021-05-05:
Hello, 

trying to construct `pymeasure.instruments.keithley.Keithley2000` results in an error:
```python
[MacPro]martin:~$ python3
Python 3.8.2 (default, Dec 21 2020, 15:06:04)
[Clang 12.0.0 (clang-1200.0.32.29)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pymeasure
>>> pymeasure.__version__
'0.9.0'
>>> from pymeasure.instruments.keithley import Keithley2000
>>> m = Keithley2000("ASRL4::INSTR")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/martin/Library/Python/3.8/lib/python/site-packages/pymeasure/instruments/keithley/keithley2000.py", line 439, in __init__
    self.adapter.config(
AttributeError: 'VISAAdapter' object has no attribute 'config'
```

LongnoseRob, 2021-05-06:
Looks like you have an VISA issue with your connection to the instrument.
Which connection are you using?
Which VISA backend are you using?

bklebel, 2021-05-07:
I don't think there is any problem with the visa backend. I think the function call for `config` here is problematic - I had the same problem once, and just commented the lines away, since I did not make the effort to see whether there exists an instrument which actually uses this, or somehow uses this. 
From a quick glance, I think this function call should be removed, and the functionality which it tries to achieve implemented correctly in the current version of pymeasure, however that might be.

bilderbuchi, 2021-07-24:
Closed by #443

closed 2021-07-24 by bilderbuchi