pymeasure/pymeasure #442: report code error found within Instrument.py, while Measuring with Keithley 6517B Electrometer

zlmturnout, 2021-05-08:
This is the basic code

`pAmeter=Keithley6517B('ASRL6::INSTR')
id=pAmeter.id
print(id)
#pAmeter.reset()
df=pAmeter.measure_current(0.1,1.05e-11)
sleep(0.1)
a=pAmeter.current
print(a)`

Type error see below:

> KEITHLEY INSTRUMENTS INC.,MODEL 6517B,4402161,A13/700x
Traceback (most recent call last):
  File "path\to\file/pAmeter_6517B_test.py", line 24, in <module>
    a=pAmeter.current
  File "path\to\python\lib\site-packages\pymeasure\instruments\instrument.py", line 233, in fget
    return get_process(vals)
TypeError: extract_value() missing 1 required positional argument: 'result'

Correction
###  in instrument.py line 233, 
the return of method fget() should be:
`return get_process(self,vals)`  instead of `return get_process(vals)`

I hope this issure will be fixed soon^OO^

CasperSchippers, 2021-05-21:
Thanks for reporting this issue.
I think that the solution you propose is not the best one: passing self will work in this case indeed, but for most other cases (e.g. when the `get_process` is provided as a lambda function) this will cause issues.
An easier solution I think would be to turn `extract_value` into a `staticmethod`, so it won't need passing `self`.

CasperSchippers, 2021-05-21:
I implemented it in a PR (#450). @zlmturnout, could you check whether this solves the issue for you.

zlmturnout, 2021-06-01:
Yeah, `staticmethod` is a better solution, agreed.

closed 2021-06-01 by zlmturnout

ventilator, 2022-01-25:
If I run the minimum example using the current version from yesterday from github, I get an "TypeError: 'staticmethod' object is not callable". Is this related to that fix?

Traceback (most recent call last):

  File "K6517B_simpe_demo.py", line 25, in <module>
    print(keithley.resistance)            # Prints the resistance in Ohms

  File "C:\WinPython\WPy64-3950\python-3.9.5.amd64\lib\site-packages\pymeasure\instruments\instrument.py", line 207, in __getattribute__
    return super().__getattribute__(name)

  File "C:\WinPython\WPy64-3950\python-3.9.5.amd64\lib\site-packages\pymeasure\instruments\instrument.py", line 377, in fget
    vals = get_process(vals)

TypeError: 'staticmethod' object is not callable

`# Import necessary packages
from pymeasure.instruments.keithley import Keithley6517B

keithley = Keithley6517B("GPIB::27")

keithley.apply_voltage()              # Sets up to source current
keithley.source_voltage_range = 200   # Sets the source voltage
                                      # range to 200 V
keithley.source_voltage = 20          # Sets the source voltage to 20 V
keithley.enable_source()              # Enables the source output

keithley.measure_resistance()         # Sets up to measure resistance

keithley.ramp_to_voltage(50)          # Ramps the voltage to 50 V
print(keithley.resistance)            # Prints the resistance in Ohms

keithley.shutdown()                   # Ramps the voltage to 0 V
                                      # and disables output`

PR #450 (2021-06-01): Fix bug in Keithley 6517B Electrometer
Fixes #442 by changing the `extract_values` method into a staticmethod.