pymeasure/pymeasure #458: Error ID code: -140 on Keithley 2400

lrennaIOS, 2021-06-15:
I am getting Error ID Code: -140 on the unit. Which I think is "Character data error".

Here is my code:
`from pymeasure.instruments.keithley import Keithley2400
from pymeasure.adapters import SerialAdapter
import numpy as np
import pandas as pd
from time import sleep
import matplotlib.pyplot as plt

adapter = SerialAdapter("COM1")
sourcemeter = Keithley2400(adapter)

# Set the input parameters
data_points = 5
averages = 1
max_current = 0.01
min_current = -max_current

# configure the instrument
sourcemeter.reset()
sourcemeter.use_front_terminals()
sourcemeter.measure_voltage()
sourcemeter.config_current_source()
sleep(0.1) # wait here to give the instrument time to react
sourcemeter.set_buffer(averages)

# Allocate arrays to store the measurement results
currents = np.linspace(min_current, max_current, num=data_points)
voltages = np.zeros_like(currents)
voltage_stds = np.zeros_like(currents)

# Loop through each current point, measure and record the voltage
for i in range(data_points):
    sourcemeter.current = currents[i]
    sourcemeter.reset_buffer()
    sleep(0.1)
    sourcemeter.start_buffer()
    sourcemeter.wait_for_buffer()

    # Record the average and standard deviation
    voltages[i] = sourcemeter.means
    voltage_stds[i] = sourcemeter.standard_devs

# Save the data columns in a CSV file
data = pd.DataFrame({
    'Current (A)': currents,
    'Voltage (V)': voltages,
    'Voltage Std (V)': voltage_stds,
})
data.to_csv('example.csv')

sourcemeter.shutdown()`

bilderbuchi, 2021-06-16:
Where does that error occur? Can you step through your code in a debugger or console.
When instantiating the instrument, can you pass `read_termination` and `write_termination` as appropriate for the settings on your device?

lrennaIOS, 2021-06-17:
The error occurs at the step: sourcemeter.measure_voltage()
I passed write_terminator = '\r\n', read_terminator = '\r\n', but it did not fix the error.

bilderbuchi, 2021-06-18:
I think you'll have to read up what that error means, exactly. What you can also do is manually execute [the things](https://github.com/pymeasure/pymeasure/blob/1a94492585d06ae6b55992f71f3213db638e7412/pymeasure/instruments/keithley/keithley2400.py#L390-L404) `measure_voltage` does to nail down which command exactly causes the error, and then try to fix that. 

Btw, are you on latest pymeasure 0.9?