pymeasure/pymeasure #465: Keithley2400_IV Sweep_VIA RS232 - ':FETCh?'의 크기를 늘리고 싶습니다.

kihyeon23, 2021-08-03:
I'm Kwon, an engineering student. I'm currently producing IV Sweep from keithley2400 products and from Python through rs232. While looking at the manual, I was trying to compensate for various errors and hit a dead end. I'm going to draw a graph with matplotlib, but the number of xvalues and yvalues is not correct. After several attempts, I found that 'yvalues' were fixed to a size of 5. (The graph came out well when each size was adjusted to 5.) The contents of the manual are as follows.
`"You can specify from one to all five elements."`
Please help me to increase the size of ':FETCh?' from 5 so that I can draw a graph that connects the steps I put in. Thank you for reading the long question.
` import sys
    startv = sys.argv[1]
    stopv = sys.argv[2]
    stepv = sys.argv[3]
    filename = sys.argv[4]
    startvprime = float(startv)
    stopvprime = float(stopv)
    stepvprime = float(stepv)
    steps = (stopvprime - startvprime) / stepvprime + 1

    # Import PyVisa and choose RS-232 as Drain-Source
    import pyvisa, time
    import serial
    rm = pyvisa.ResourceManager()
    rm.list_resources()
    with rm. open_resource('COM3') as Keithley:
        Keithley.port = 'COM3'
        Keithley.baudrate = 9600
        Keithley.timeout = 25000

    Keithley.open()
    Keithley.read_termination = '\r'
    Keithley.write_termination = '\r'
    Keithley.write("*RST")
    Keithley.write("*IDN?")

    Keithley.write(":SENS:FUNC:CONC OFF")                  
    Keithley.write(":SOUR:FUNC VOLT")                      
    Keithley.write(":SENS:FUNC 'CURR:DC' ")                 

    Keithley.write(":SOUR:VOLT:START ", startv)
    Keithley.write(":SOUR:VOLT:STOP ", stopv)               
    Keithley.write(":SOUR:VOLT:STEP ", stepv)              
    Keithley.write(":SOUR:SWE:RANG AUTO")                   

    Keithley.write(":SENS:CURR:PROT 0.1")                     
    Keithley.write(":SOUR:SWE:SPAC LIN")                    
    Keithley.write(":SOUR:SWE:POIN", str(int(steps)))      
    Keithley.write(":SOUR:SWE:DIR UP")
    Keithley.write(":TRIG:COUN", str(int(steps)))
    Keithley.write(":FORM:ELEM CURR")

    Keithley.write(":SOUR:VOLT:MODE SWE")                   
    Keithley.write(":OUTP ON")                             

    import numpy as np
    result = Keithley.query(":READ?")                       
    yvalues = Keithley.query_ascii_values(":FETCh?")
    Keithley.write(":OUTP OFF")
    Keithley.write(":SOUR:VOLT 0")

    import matplotlib.pyplot as plt
    from matplotlib.animation import FuncAnimation
    from scipy import stats
    xvalues = np.arange(startvprime, stopvprime+1, stepvprime)  
    plt.plot(xvalues, yvalues)
    plt.xlabel(' Drain-Source Voltage (V)')
    plt.ylabel(' Drain-Source Current (mA)')
    plt.title('IV Curve')
    plt.show()
    np.savetxt(filename, (xvalues,yvalues))`

kihyeon23, 2021-08-03:
ex) python name.py -10 10 1 filename
=> ValueError: x and y must have same first dimension, but have shapes (21,) and (5,)

bilderbuchi, 2021-08-03:
I'll convert this into a discussion, as this is not something related to pymeasure (you're not even using pymeasure functionality). Concerning your question, what does the user manual say about FETCH?

closed 2021-08-03 by bilderbuchi