pymeasure/pymeasure #473: How to define command to send value without read?

PeterPirog, 2021-09-02:
I would like to add zeroing function for  HP 34401 multimeter, but I'm not sure how to define it:

`zeroing=Instrument.control("CALC:FUNC NULL;:CALC:STAT ON", "Zeroing")`

I will be grateful for any suggestions.

Peter

LongnoseRob, 2021-09-02:
If you have a SCPI command that cannot be queried, I woudl recommend to use [Instrument.setting](https://pymeasure.readthedocs.io/en/latest/api/instruments/instruments.html#pymeasure.instruments.Instrument.setting)

Another idea would be to implement the CALC:FUNC with all possible values as one propery and add another propery for the CALC:STATE.

Do you also would like to set the null-value(offset) by remote?

PeterPirog, 2021-09-02:
Thank You fot the answer. Yes  I would like to use null value by remote to measure resistance change.

I try to understand how to define new instruments because I have some Fluke multifunction calibrators and RF power meters  which I would like to add.

bilderbuchi, 2021-09-02:
Indeed, a setting is the way to go here. I'll convert this into a discussion, where it is more appropriate, to keep the issue tracker "clean" 😁

closed 2021-09-02 by bilderbuchi