pymeasure/pymeasure #478: Run experiment in console/batch mode

msmttchr, 2021-09-12:
I think it can be extremely useful to be able to write script that can run both in GUI and console batch mode.
The example provided in pymeasure shows that the same Procedure class can be run both in GUI mode and in console mode, but in console mode all the parameters have to be provided inside the script.

My idea would be to create a sort of `ConsoleScript` class which is the console version of `ManagedWindow`. `ConsoleScript` it will take care of creating an `argparse` initialization to parse the users's input parameters via command line.

Does anyone thought about this ? Anyone who started such pull request ? Any other comment ?

msmttchr, 2021-09-22:
Indeed, my idea was to make the parameters available as command line options to be able to configure them. Obviously the graphic part will not be available, but log messages will be available in the console and the same results files (.csv) will be generated as the regular experiment in GUI mode.

CasperSchippers, 2021-10-06:
Sounds like it could be useful indeed. One thing to keep in mind is to ensure that interrupting the measurement (e.g Keyboard Interrupt) works as an abort, and hence also runs the shutdown method of the procedure.

LongnoseRob, 2021-10-06:
I like the idea.:+1:
These are two points which I would like o add to the discussion:
1. instead/in parallel to offering the parameters as command line options also offer the possibility to use a "config"-file[^1]
2. use something like [progressbar](https://progressbar-2.readthedocs.io/en/latest/) to visualize the progress (if neccessary/possible/wanted)

[^1]: some easy flavor

msmttchr, 2021-10-20:
@LongnoseRob, if you have time, could You please give a try to PR #500 ? It is still draft, but pretty much usable.
I would be interested in your preliminary feedback

CasperSchippers, 2021-11-02:
It looks very promising indeed; I tried some things (not yet all) and most works quite well and looks promising

> Have you tried some of the other command line options (e.g. to set parameter from log file)?

There is a small problem when using the --use-log-file option, see my remark in the PR.

I will continue testing and try to review asap

msmttchr, 2023-08-12:
Completed as PR #500 is merged. Thanks to all contributors and reviewers, especially @CasperSchippers, @LongnoseRob, @mcdo0486

closed 2023-08-12 by msmttchr