pymeasure/pymeasure #486: Can not perform basic math operations on Parameters

bicarlsen, 2021-10-05:
I am getting strange error when trying to perform math on Parameters.
+ When trying to do basic mathematical operations with a parameter and a built in Python data type I get a `TypeError: unsupported operand type(s)`.
+ When trying to use `parameter.value` in the math I get the error `AttributeError: 'float' object has no attribute 'value'`

```python
class My_Procedure( Procedure ):
    f_param  = FloatParameter( 'Float Parameter',   default = 0 )
    g_param = FloatParameter( 'Another Float Parameter',   default = 0 )

    def math( self ):
        sum = f_param + g_param  # no error

        res_1 = 1* self.f_param  # TypeError: unsupported operand type(s) for *: 'int' and 'FloatParameter'
        
        f_val = self.f_param.value  # no error
        res_2 = 1* self.f_param.value  # AttributeError: 'float' object has no attribute 'value'
```

msmttchr, 2021-10-05:
Strange problem, I would expect the following:
- line `sum = f_param + g_param` should actually give an error of variable not defined
- line `res_1 = 1* self.f_param` should be ok.
- line `f_val = self.f_param.value` and `res_2 = 1* self.f_param.value` both give errors

I have checked also my assumption using the `gui.py` basic example, adding your code and a call to `math` method.
Something weird going on, how do you trigger the problem ?

bicarlsen, 2021-10-06:
I think I've narrowed down the problem. It seems to be occurring because I am doing doing the arithmetic in an `@property` method. i.e.
```python

class My_Procedure( Procedure ):
    f_param = FloatParameter( 'Float Parameter',   default = 0 )
    g_param = FloatParameter( 'Another Float Parameter',   default = 0 )

     @property
     def my_property( self ):
        total = self.f_param + self.g_param  # no error

        res_1 = 1* self.f_param  # TypeError: unsupported operand type(s) for *: 'int' and 'FloatParameter'
        
        f_val = self.f_param.value  # no error
        res_2 = 1* self.f_param.value  # AttributeError: 'float' object has no attribute 'value'

    def my_method( self ):
        # everything works fine
        total = self.f_param + self.g_param 
        res_1 = 1* self.f_param  
        f_val = self.f_param.value 
        res_2 = 1* self.f_param.value
```

I forgot to include that detail in my original post. Apologies about that.

To answer your question `print (self.voltage_step, type(self.voltage_step))` outputs `0.01 <class 'float'>`.

CasperSchippers, 2021-10-06:
I think the problem lies in the fact that upon initialization of the Procedure class, the `update_parameters` method checks for every attribute of the class whether it is a type of `Parameter`; for this it gets every attribute, and hence also your `my_property` is executed. At this stage however, the Parameters are not yet converted into floats (or at least not per se all Parameters, as the initialization is in the process of doing so.

A simple (though I'm not sure if this would be the neatest) way out of this would be to put the contents of `my_property` in a `try: .... except TypeError: ....` statement.

bicarlsen, 2021-10-06:
Thanks @CasperSchippers , I agree that that is the case. In fact I've had to do similar in other `@property` methods when returning a private variable. e.g.
```python
class My_Procedure( Procedure ):
    
     @property
     def another_property( self ):
        try:
            return self._another_property

        except AttributeError as err:
            # elements not yet set
            return None

    def startup( self ):
        self._another_property = 'hi there'
```

@msmttchr , the output when in an `@property` method is `0.01 <class 'pymeasure.experiment.parameters.FloatParameter'>`.

msmttchr, 2021-10-06:
> I think the problem lies in the fact that upon initialization of the Procedure class, the `update_parameters` method checks for every attribute of the class whether it is a type of `Parameter`; for [...]

I think you are right and the problem happens in a non deterministic way, since when you execute `_update_parameters` you may have already processed the relevant parameter before processing the property and so this will succeed otherwise it will fail.

bilderbuchi, 2021-10-06:
Finally, I'm not very familiar with that part of the code base, but the way that update_parameters replaces the objects with their POF type, to mainly assemble a dictionary of parameters feels weird and could maybe be improved. I haven't dug deeper, though.

msmttchr, 2021-10-06:
> It would be interesting to get a test case together to demonstrate this.
> I'm not sure where nondeterminacy would come from, also because even dicts
> nowadays observe insertion order. Also, I'm not at a PC right now but my
> gut feeling is that this should be ordered because of the distinction
> between import and instantiation time.

I meant that order is fixed, but it depends whether the Parameter variables
are processed before property or not. I am not sure, but I think dir
returns names in alphabetical order, so processing is alphabetical. Now if
the property name is before the relevant parameter, you get the error,
otherwise everything looks good

msmttchr, 2021-10-06:
> It seems the issues comes from the [`#_update_procedures`](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/experiment/procedure.py#L99) method of `Procedure`. When [`getattr(self, item)`](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/experiment/procedure.py#L107) is called. I see two possible solutions:
> 
> 1. Overload mathemtaical operators for `Parameter`s, delegating the opertation to the Parameter's `.value` property.
> 2. In the `#_update_procedures` method, add a `try` block around the `getattr` section, and if there is an `AttributeError` delay the evaluation until the rest of the items have been initialized.
> 
> Personally, option 1 seems better to me. Are there any other ideas for a fix, or opinions on the solutions I proposed?

I was thinking another solution, that is loop through class attributes only and skip properties and methods, I wonder if this is possible with inspect

msmttchr, 2021-10-15:
Implemented fix in #489

closed 2021-10-15 by msmttchr