pymeasure/pymeasure #506: Thread safe communication

svihra, 2021-11-02:
Dear all,

I was trying to create a control tool on top of the Keithley2450 which would continuously take data while the user/code could apply changes. I created an update method which is providing the [readback/measured/status] values in one go. While testing, I ran into an issue where the internal methods are not written thread-safe, breaking the run. Debugging further (at least in case of this particular device) I was able to locally implement thread locks in the underlying Instrument class that provides the communication.

That said, I think the best place would be to modify the VISAAdapter class to ensure the thread-safe behaviour (which I think I should be able to do myself and create a merge request). Is there something that I miss and could cause major issues elsewhere?

bilderbuchi, 2021-11-02:
Note that pyvisa is [thread-safe](https://pyvisa.readthedocs.io/en/latest/faq/faq.html?highlight=thread#is-pyvisa-thread-safe), so the communication to the instrument should already be unproblematic. Where are you stumbling? I'd be interested in a repro procedure (ideally without needing a Keithley2450 :D).

svihra, 2021-11-04:
First of all, these are the library versions:
```
Python 3.7.11
pymeasure                 0.9              py37hf985489_2    conda-forge
pyvisa                    1.11.3           py37hf985489_1    conda-forge
pyvisa-py                 0.5.2              pyh44b312d_0    conda-forge
```

I have not tested with a dummy device but I think the issue is caused by the real lag in the communication.
The dummy code would be:

```
import threading, asyncio, time
from pymeasure.instruments.keithley import Keithley2450

class k2450(Keithley2450):
    def __init__(IP):
        super().__init__(f"TCPIP0::{IP}::inst0::INSTR")
        print(f"-> Connected {self.id}")
        self.loop = asyncio.get_event_loop()
        self._thread = threading.Thread(target=self.loop.run_forever).start()

    async def update(self):
        while True:
            self.ask('READ? "defbuffer1", SOUR, READ, SOURSTAT') # defbuffer1 is the default k2450 buffer name, this is the only way I found how to get back the source readback, measurement and source status values in one go
            await asyncio.sleep(0.25)

dev = k2450("1.1.1.1")
asyncio.run_coroutine_threadsafe(dev.update(), dev.loop) # placed in a separate loop to provide user access to the rest of the code

for _ in range(10):
    print(dev.source_voltage) # this should do an independent reading from the one in update, mimicking user access
    time.sleep(1)
```

Please ignore what this piece of code actually does, the main point is that there is an infinite loop constantly reading back from the device and another part that might be user access to the values. The error is then somehow caused internally by combination of `ask` and `source_voltage` functions (or any other Keithley2450 impelmented ones), basically the queries get misaligned. An example of the error:

```
>>> dev = iv.k2450("128.141.235.137")
-> Connected KEITHLEY INSTRUMENTS INC.,MODEL 2450,04041167,1.1.0s
>>> dev.source_voltage
ost recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pymeasure/instruments/instrument.py", line 148, in fget
    vals = self.values(get_command, **kwargs)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pymeasure/instruments/instrument.py", line 111, in values
    return self.adapter.values(command, **kwargs)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pymeasure/adapters/adapter.py", line 87, in values
    results = str(self.ask(command)).strip()
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pymeasure/adapters/visa.py", line 112, in ask
    return self.connection.query(command)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa/resources/messagebased.py", line 638, in query
    self.write(message)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa/resources/messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa/resources/messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa_py/highlevel.py", line 543, in write
    written, status_code = self.sessions[session].write(data)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa_py/tcpip.py", line 202, in write
    self.link, self._io_timeout, self.lock_timeout, flags, block
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa_py/protocols/vxi11.py", line 225, in device_write
    self.unpacker.unpack_device_write_resp,
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa_py/protocols/rpc.py", line 472, in make_call
    return super(RawTCPClient, self).make_call(proc, args, pack_func, unpack_func)
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa_py/protocols/rpc.py", line 238, in make_call
    self.do_call()
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa_py/protocols/rpc.py", line 501, in do_call
    xid, verf = u.unpack_replyheader()
  File "/opt/homebrew/Caskroom/miniconda/base/envs/clic_tools/lib/python3.7/site-packages/pyvisa_py/protocols/rpc.py", line 185, in unpack_replyheader
[... 20 lines cut ...]
class VISAAdapter(Adapter):
    ....

    def write(self, command):
        """ Writes a command to the instrument

        :param command: SCPI command string to be sent to the instrument
        """
        self._lock.acquire()
        self.connection.write(command)
        self._lock.release()

    def read(self):
        """ Reads until the buffer is empty and returns the resulting
        ASCII response

        :returns: String ASCII response of the instrument.
        """
        self._lock.acquire()
        tmp = self.connection.read()
        self._lock.release()
        return tmp
    ....
```
There might be definitely a better way, happy to see the ideas!

bilderbuchi, 2021-11-04:
Hm, that is strange. I was previously using 3 parallel threads (GUI interaction, regular polls, scripted changes) to talk to a number of instruments, and did not have any problems. I'm wondering why nobody has seen this before?
Btw have you tried with not the pyvisa-py backend, but a proper VISA backend (e.g. R&S, NI,...)?

On deeper investigation it seems that pyvisa's thread safety seems [a bit](https://github.com/pyvisa/pyvisa-py/issues/220#issuecomment-594718620) , well, [weaker than expected](https://github.com/pyvisa/pyvisa/issues/262#issuecomment-302613625).
At first glance your solution with a lock looks fine, although 
* it would be preferable to use the with statement form 
```
with _lock:
    bla
```
* you'd also have to lock the other uses of the connection (there are several that don't delegate to read/write)
* I'm thinking if we should not protect all adapters, but this will need some reorganisation and thought into the structure of `Adapter`.

svihra, 2021-11-05:
I only tried the pyvisa-py, I once tried to install VISA properly and failed miserably so I went with the path of the least resistance. Yes it really seems the issue is with my choice of the parallelisation (I can give a shot to your approach when I get to it).

Thanks, in hindsight this locking mechanism looks better indeed :-). Sorry for not being clear, I made sure I implemented it everywhere in my local VISAAdapter (not just the read write).

I briefly had a look at the other adapters and do not really have an idea how to implement this clearly and consistently for all of them. Just the fact that some of them rely on `ask` which consists of `write` and `read` could potentially cause issues in parallelisation (a sneaky command getting in between). Perhaps having two locks in those cases and checking their status? That said, it would be on the developer of the new adapter to make sure all commands are properly locked (so the base classes `ask` would have to also be virtual or both locks would already have to originate there).

Should I create a merge request on the VISAAdapter, or would you prefer to do it all in one go with other adapters included?

bilderbuchi, 2021-11-05:
If that's alright with you, create a merge request for the VISAAdapter, with the understanding that it might take quite some time to get it merged (we are low on manpower). We then have your contribution available and consider how to best integrate/rework it, or postpone the generic implementation.
You can work in your local branch until it becomes available upstream, so you are not blocked on this issue.

Yeah, a generic solution would need some restructuring. I have some ideas already, but yeah, I already realised that `ask` will be tricky.

scandey, 2023-01-15:
Summary of #833 in context of this discussion:
I was doing something similar to OP, with an infinite loop getting status from an instrument (one control at a time over two channels, which takes up an appreciable number of asks). The infinite loop was in a worker thread with a sleep of about 1 second between status checks. I'm also using the pyvisa-py backend, since that package makes it much easier to deploy a cross-platform system. 

I also was setting the controls elsewhere in a different thread (originally without any thread safety measures). This second thread was still referencing back to the same instrument instance (and by extension, the same adapter instance). The "write" commands were going through successfully but the instrument has check_set_errors turned on so each control setting requires at least one "ask" for the "SYST:ERR?" request. 

When the commands and the status checker collided in time, sometimes the results would get messed up and the check_set_errors function would receive the results intended for the status checker worker. check_set_errors would throw an error since the results aren't SCPI error messages and sometimes either check_set_error or the status check would time out. 

I eventually solved this by adding a basic threading lock in the same way as OP. In my case I didn't try to isolate the control setting vs the check_set_errors vs reading, I just turned off the status checker until the commanding script was finished. In the longer term, I'll probably do a pull request to turn off the check_set_errors for this instrument (T3AFG) for now since it doesn't add much information and makes this kind of scripting much harder.

bilderbuchi, 2023-01-15:
I think with the relatively recent reworking of the `Adapter` hierarchy, we should now be in a position to relatively easily add a lock inside `Adapter.ask()` (to ensure `write` and subsequent `read` stay synchronised), and possibly another one to make sure a solitary `write()` or `read()` (e.g. in a method) does not interfere with an in-progress `ask()`.

@bmoneke thoughts?

bilderbuchi, 2023-01-15:
Situation to avoid: the thieving ask()
```mermaid
sequenceDiagram
    Note over Thread1: ask(CMD1)
    activate Thread1
    Thread1->>connection: write CMD1
    Note over Thread2: ask(CMD2)
    activate Thread2
    Thread2->>connection: write CMD2
    connection-->>Thread2: REPLY1
    deactivate Thread2
    connection-->>Thread1: REPLY2
    deactivate Thread1
```
We should be able to deal with this with a lock around the write-read pair in `Adapter.ask`.

bilderbuchi, 2023-01-15:
Situation to avoid: _the thieving read()_
```mermaid
sequenceDiagram
    Note over Thread1: ask(CMD1)
    activate Thread1
    Thread1->>connection: write CMD1
    Note over Thread2: manual read and write
    Thread2->>connection: write CMD2
    connection-->>Thread2: read REPLY1
    connection-->>Thread1: read REPLY2
    deactivate Thread1
```
This can happen e.g. in method.

No good solution right now. 
Note that we also need to make sure that we don't create deadlocks or race conditions, which will be especially interesting as soon as we have multiple/nested locks!

bilderbuchi, 2023-01-15:
Situation to avoid: _the wrongly checked-get_

This is now a sequencing error, where we want to make sure that a property command and its `check_errors()` call stay together ("atomic" if you will)
This is what we want: 
```mermaid
sequenceDiagram
    Note over Thread1: get prop1
    activate Thread1
    Note over Thread1: calls values()->ask()
    activate Thread1
    Thread1->>connection: write GET_P1
    connection-->>Thread1: read REPLY_P1
    Note over connection: this created error state
    deactivate Thread1
    Note over Thread1: check errors
    Thread1->>connection: write ERRCHECK
    connection-->>Thread1: read ERRSTATE_P1
    deactivate Thread1
```

This might happen:

```mermaid
sequenceDiagram
    Note over Thread1: get prop1
    activate Thread1
    Note over Thread1: calls values()->ask()
    activate Thread1
    Thread1->>connection: write GET_P1
    connection-->>Thread1: read REPLY_P1
    Note over connection: this created error state
    Note over Thread2: get prop2
    activate Thread2
    Note over Thread2: calls values()->ask()
    activate Thread2
    Thread2->>connection: write GET_P2
    connection-->>Thread2: read REPLY_P2
    deactivate Thread2
    Note over Thread2: check errors for P2
    Thread2->>connection: write ERRCHECK
    connection-->>Thread2: read ERRSTATE_P1
    Note over Thread2: Gets P1's error!
    deactivate Thread2
    Note over Thread1: check errors
    Thread1->>connection: write ERRCHECK
    connection-->>Thread1: read ???
    deactivate Thread1
```

bilderbuchi, 2023-01-15:
Situation: _the stolen checked-get errorstate_
```mermaid
sequenceDiagram
    Note over Thread1: get prop1
    activate Thread1
    Note over Thread1: calls values()->ask()
    activate Thread1
    Thread1->>connection: write GET_P1
    connection-->>Thread1: read REPLY_P1
    Note over connection: this created error state
    Note over Thread1: check errors
    activate Thread1
    Thread1->>connection: write ERRCHECK
    Thread2->>connection: write CMD2
    connection-->>Thread2: read ERRSTATE_1
    Note over Thread2: This reply is garbage!
    connection-->>Thread1: read ???
    deactivate Thread1
    deactivate Thread1
```

bilderbuchi, 2023-01-15:
Equivalent situations that I have not explicitly sketched here should be possible when _setting_, not _getting_ properties (in `Instrument.control.fset`, not `Instrument.control.fget`), i.e. _the wrongly checked-set_ and _the stolen checked-set errorstate_. The main difference is that the main command sequence is not a `values()` callling `ask()`, but a plain `write()`.

bmoneke, 2023-01-15:
I see the problem of deadlocks.

Pyqt for example interrupts methods in order to react to another signal (in the same thread!).
With locks I repeatedly run in deadlocks.
As a solution I moved every instrument in its own thread and work through pipes.

Ensuring correct thread safety is complicated, I think and I'd propose to follow pyvisa: Just use a connection in a single thread.

bilderbuchi, 2023-01-16:
Yeah, it's not only pyqt, even the underlying OS can pause/resume threads more or less arbitrarily, that's why a mechanism is needed for handling access.
How did you implement the locking that did not work correctly?

bmoneke, 2023-01-16:
~~~python
@pyqtSlot()
def method1(self):
    self.lock.lock()
    self.device.ask("abc")
    self.lock.unlock()

@pyqtSlot()
def method2(self):
    self.lock.lock()
    value = self.device.ask("def")
    self.lock.unlock()
    return value
~~~

The problem was, that with one button click, it started `method1`. In the meantime, a timer started `method2` interrumping `method1`. Therefore `method2` gets stuck, because the lock is still locked from `method1`. As `method2` never terminates, pyqt never returns to `method1` to unlock the lock.
Later I used `tryLock()`, but that made the code even more complicated, as you have to repeat the method until you achieve a lock...

bilderbuchi, 2023-01-16:
Hm, interesting. Was that all on one thread? Otherwise, I would have thought that the thread with `method1` would resume as soon as the thread with `method2` starts waiting for the lock to release....

In other news, I've been thinking a bit, and one reentrant [`threading.RLock`](https://docs.python.org/3.11/library/threading.html#rlock-objects) in `Adapter` looks like it could be a viable approach to deal with the lock recursion, as the owning thread could recurse into the lock as required (e.g. a property `fget` calling `values` which calls `ask` which calls `read`, all of which need to lock to ensure no other thread interferes and steals replies/disturbs sequencing).

bmoneke, 2023-01-16:
> Was that all on one thread? 

Yes, both methods were in the same class (subclass of main window). No (manual) threading, only the event loop stuff necessary for Pyqt gui.

bilderbuchi, 2023-01-18:
This is weird, I would _not_ have thought that a firing timer can interrupt the execution of another running method/slot in the same thread, and afaik a Qtimer would (typically? always?) live/be started in the main thread. Maybe I can reproduce this.

bilderbuchi, 2023-01-18:
I guess we won't solve this here, this still smells weird/impossible to me (might be another problem like `method1` calling `method2` through several layers of indirection?), let's get a minimal reproducible example that we can debug together, elsewhere (if you want).

bmoneke, 2023-01-19:
I cannot reproduce it. But it killed my program regularly (controlling lasers...), therefore I circumvented the problem by moving the actors into their threads communicating via pipes...

bilderbuchi, 2023-03-14:
Short update: I have a in-progress solution for thread-safe instrument communications. It's not yet complete, and still rough around the edges, but I managed to write tests that reliably test multi-threading behaviour. 
It will need feature-completion and some polish, then I'll open a PR with a branch for people to test with instruments.

bmoneke, 2023-05-17:
I'd like to link to PR #608, such that that PR could be closed, once this issue is resolved.