pymeasure/pymeasure #507: Wrong right errorbar

maederan201, 2021-11-03:
More by accident I stumbled upon a bug/typo/copy-paste-error in ```update_data``` of the ```ResultsCurve``` class when looking for something unrelated. The following code-block sets errorbar options if enabled:

```
if hasattr(self, '_errorBars'):
            self._errorBars.setOpts(
                x=data[self.x],
                y=data[self.y],
                top=data[self.yerr],
                bottom=data[self.yerr],
                left=data[self.xerr],
                right=data[self.yerr],
                beam=max(data[self.xerr], data[self.yerr])
            )
```
I think the line ```right=data[self.yerr]``` should be changed to ```right=data[self.xerr]```. Can someone confirm my suspicion? I have never worked with errorbars and I don't have an example to test it ready.

maederan201, 2021-11-04:
I checked further, this feature does not seems necessary at this point in pymeasure as the only point where a ResultsCurve is created is in ```PlotterWindow```. There the xerr and yerr keywords are not given, so this error is without consequence in the behaviour of pymeasure.

CasperSchippers, 2021-11-04:
You're indeed right that the `self.yerr` for right should've been `self.xerr` (though when diving into this it seems that using the `height` and `width` arguments also would've been nicer than the `top`, `bottom`, `left`, and `right` arguments).

I also noted that, at this moment, the entire feature does not work as far as I can tell (though I have never tried it before, so I can't tell if that is due to a recent change somewhere). If you get the xerr and yerr defined (you need to override the new_curve method for this i think), you get an error concerning the beam parameter and (when omitting the beam parameter) the error-bars are not added to the graph.

maederan201, 2021-12-14:
I looked in to this a bit more. I came to the same conclusion(s) as you did. I think the `beam` argument at least for the further development can be set to 0.5

I managed to have an example now on a branch on my fork which at least works. One thing that I found missing was the fact that the errorbar was never added to the `PlotWidget` which I for now solved by appending the constructor of the `PlotterWindow` part

```
self.curve = ResultsCurve(plotter.results, columns[0], columns[1],
                                  pen=pg.mkPen(color=pg.intColor(0), width=linewidth), antialias=False,xerr=xerr,yerr=yerr)
self.plot.addItem(self.curve)
if hasattr(self.curve, '_errorBars'):
    self.plot.addItem(self.curve._errorBars)
```

Is it worth following this path? If so one should probably find some scenarios where this could be useful to understand how the errorbar values are obtained in an experiment and where (on which level) they should be implemented.

CasperSchippers, 2021-12-26:
I think it is indeed worth following this up, since it'd be a bit strange to have this feature partially implemented, but not fully working; so either we'd have to fix it or have to remove it.

The first step would be (I think) to fix the parts that are incorrect (in the way you suggested). The next step would be to see how errorbar values are obtained in experiments and if/how the should be implemented more conveniently.

Do you use the error-bars in your experiments, or did you just accidentally happen upon the feature?

CasperSchippers, 2022-01-06:
Ahh, check.

I would agree that I don't see much use in having error-bars in live plots; the only real use case I can come up with would involve instruments that can return an error-value directly, but even in that case I would think that is only relevant for post-processing, and not really useful in a live plot.

As such I would vote for removing the error-bar feature; unless of course someone has a use case for them, but then it needs more than simply fixing that part of the code.

bilderbuchi, 2022-01-06:
Yeah, I agree, remove it.

PR #569 (2022-01-26): Remove errorbars
PR to remove the unused and not-working errorbar feature.
Resolves #507
closed 2022-01-26 by CasperSchippers