pymeasure/pymeasure #513: Should Instrument.shutdown() do more?

bilderbuchi, 2021-11-05:
Right now, we have

I want to open a discussion if there is an expectation that `Instrument.shutdown()` do more. Right now it's ~ a no-op (setting a boolean and emitting a log message), and it's fully up to the instrument developer to make it do more.

Should it at least close the connection to the hardware? 
*If so: Extract an `Adapter.close()` from `Adapter.__del__()` and call that. TBC Do all Adapters observe that?*

Should it log/warn if not `self.complete`?

Any other things?

LongnoseRob, 2021-11-06:
From my point of view,  `Instrument.shutdown()` should be the place where the instrument is brought to a safe condition before the connection is closed.
With safe condition I mean conditions like this:
- output off for RF-generators and power suppliies
- motors stopped for motor drivers.

But it could also mean that certain settings in the instrument could be recorded before the connection is terminated.

Ideally `Instrument.shutdown()` would be called for all open instruments in a procedure/experiment, when the procedure/experiment is completed.

Also I would like to see it called if there is an interruption by the user (Ctrl+C) or failure exception stopping the execution of the python script, I think this could be doable with defining sth like `Procedure.__exit__`

I know that SCPI (and lower protoclls like GPIB or USB-TMC) has less focus on safety then e.g. industrial field busses like Profibus, DeviceNet or EtherCat. 
Nevertheless we should try to get everything into a safe-state in such cases.

msmttchr, 2021-11-06:
I would vote for closing the connection and leaving instrument's specific action to subclasses

bilderbuchi, 2021-11-06:
All good points! 
Yes, device-specific measured would be implemented in the concrete subclass, and we make sure that that is automatically called by the `shutdown()` method. 
Exit/interruption-safe handling could be happen in `__del__` and/or a finally clause in code handling this, as appropriate.