pymeasure/pymeasure #514: VISAAdapter initialization only allows a subset of key word arguments for the pyvisa open_resource() method

samcondon4, 2021-11-05:
https://github.com/pymeasure/pymeasure/blob/7580edd2cc5dfaafb9c0d6462d6f95cd87a3d468/pymeasure/adapters/visa.py#L60

In the line I linked above, there is a list of allowed keywords for the open_resource() method. Is this check helpful/necessary? I just ran into an issue where I have an instrument with a configurable baud rate and I'd like to be able to specify the baud rate when I instantiate the instrument using a keyword argument. With this check the baud_rate=<desired baud value> keyword is removed. For this case it is a simple fix to just add 'baud_rate' to the safeKeywords list, though I wonder if this check is necessary? There are many keyword arguments in the open_resource() method which a user may find useful but will not have access to with this check in place.

msmttchr, 2021-11-06:
See also #334

bilderbuchi, 2021-11-06:
Indeed, closing as duplicate, let's continue over there.

closed 2021-11-06 by bilderbuchi