pymeasure/pymeasure #523: USB to RS-232 adapter support

mfikes, 2021-11-13:
It is possible to control a Keithley 2400 via its RS-232 port, using PyMeasure via a USB / RS-232 adapter cable by introducing a new adapter which does some minor newline processing.

This issue is essentially a question on whether it might make sense to add support for such USB / RS-232 cables to PyMeasure (or perhaps whether it is already possible and perhaps I've missed something).

The specific hardware USB / RS-232 adapter cables I'm referring to have USB (type A or C) on one end, and DB9 on the other end, and they typically have FTDI or Prolific USB to UART chipsets embedded directly in the cable. They are an inexpensive alternative to GPIB offerings if you want to control a single instrument (they are often USD 15 or less).

I found that it is not quite possible to use PyMeasures's `SerialAdapter` directly with one of these cables because, for example the command passed to `ask` needs a newline appended at the end. It appears that one way to make an adapter that works is to modify a copy of the `PrologixAdapter` and remove all of the Prologix-specific aspects in it, leaving only the newline processing logic. The code that remains after doing this is included below as an example, and use of this adapter is much like any other adapter with PyMeasure:

```python
adapter = UsbSerialAdapter('/dev/cu.usbserial-FTCGVYZA', baudrate=9600)
smu = Keithley2400(adapter)
```

Below is the sort of code that could be used to form a new adapter (I hereby offer it as falling under the PyMeasure license, as it was directly derived from the Prologix adapter by simply deleting code from it). Alternatively, you could perhaps envision some of the Prologix adapter being factored out into an intermediate class that lies between the Prologix class and the base serial class.

If it turns out such an adapter might make sense, I'd be happy to contribute, perhaps with a little guidance from PyMeasure's maintainers on naming, structure, etc. Or, if I'm missing some aspect on how this could easily be done without adding a new adapter, perhaps I could contribute to some new documentation.

```python
import time

import serial

from .serial import SerialAdapter

class UsbSerialAdapter(SerialAdapter):
    """ Encapsulates the additional commands necessary
    to communicate over a USB to Serial Adapter,
    using the SerialAdapter.

    """

    def __init__(self, port, rw_delay=None, serial_timeout=0.5,
                 preprocess_reply=None, **kwargs):
        super().__init__(port, timeout=serial_timeout,
                         preprocess_reply=preprocess_reply, **kwargs)
        self.rw_delay = rw_delay

    def ask(self, command):
        """ Ask the adapter, include a forced delay for some instruments.

        :param command: SCPI command string to be sent to instrument
        """

        self.write(command)
        if self.rw_delay is not None:
            time.sleep(self.rw_delay)
        return self.read()

    def write(self, command):
        """ Writes the command to the adapter

        :param command: SCPI command string to be sent to the instrument
        """
        command += "\n"
        self.connection.write(command.encode())

    def write_binary_values(self, command, values, **kwargs):
        """ Write binary data to the instrument, e.g. waveform for signal generators.

        values are encoded in a binary format according to
        IEEE 488.2 Definite Length Arbitrary Block Response Data block.

        :param command: SCPI command to be sent to the instrument
        :param values: iterable representing the binary values
        :param kwargs: Key-word arguments to pass onto :meth:`._format_binary_values`
        :returns: number of bytes written
        """
        super().write_binary_values(command, values, **kwargs)
        self.connection.write('\n'.encode())

    def read(self):
        """ Reads the response of the instrument until timeout

        :returns: String ASCII response of the instrument
        """
        return b"\n".join(self.connection.readlines()).decode()

    def __repr__(self):
            return "<UsbSerialAdapter(port='%s')>" % self.connection.port
```

LongnoseRob, 2021-11-13:
> I found that it is not quite possible to use PyMeasures's SerialAdapter directly with one of these cables because, for example, the command passed to ask needs a newline appended at the end. It [...]

Have you tried using pyvisa (with pyvisa-py as a backend  (assuming you are running on a linux-system)) to use the [`ASLR INST` ressource](https://pyvisa.readthedocs.io/projects/pyvisa-py/en/latest/installation.html#serial-resources-asrl-instr)?

Example from [pyvisa docs](https://pyvisa.readthedocs.io/en/latest/introduction/communication.html#an-example) adopted to serial connection
```import pyvisa
rm = pyvisa.ResourceManager()
rm.list_resources()
('ASRL1::INSTR', 'ASRL2::INSTR', 'GPIB0::14::INSTR')
#  my_instrument = rm.open_resource('GPIB0::14::INSTR') 
# assuming the USB-RS232 adapter is ASLR1
my_instrument = rm.open_resource('ASLR1::INSTR') 
print(my_instrument.query('*IDN?'))
```

With the use of pyvisa, it is also easy to set the proper termination characters or query delays if required.

mfikes, 2021-11-13:
@LongnoseRob Thank you! Indeed PyVISA works properly for this use case and I've confirmed that the Keithley 2400 can be easily driven from PyMeasure 

```python
smu = Keithley2400('ASRL/dev/cu.usbserial-FTCGVYZA::INSTR',
                   baud_rate=9600, write_termination='\r', read_termination='\r')
```

So, there is definitely no new adapter needed. 

And insofar as PyMeasure documentation is concerned, I'm leaning towards this issue simply being a lack of familiarity on my part with PyVISA.  (In other words PyMeasure need not document things that are in the world of PyVISA, and it already has support for an documentation pointers to PyVISA.)

closed 2021-11-13 by mfikes

bilderbuchi, 2021-11-13:
Yeah, nowadays it make most sense to use the visaadapter, and if your port name is a string that's what gets used by default anyway. Also, depending on which visa backend you use, it even recognizes simpler strings like "COM2" without prepended  ASLR... (NI's backend does, for example)

LongnoseRob, 2021-11-13:
> @LongnoseRob Thank you! Indeed PyVISA works properly for this use case and I've confirmed that the Keithley 2400 can be easily driven from PyMeasure using ..

You're welcome :grin:

mpikmwm, 2022-08-25:
> ```python
> smu = Keithley2400('ASRL/dev/cu.usbserial-FTCGVYZA::INSTR',
>                    baud_rate=9600, write_termination='\r', read_termination='\r')
> ```
> 
> So, there is definitely no new adapter needed.

I've just found this and it helped me to use the existing Keithley2000 instrument with my RS232 to USB adapter.
It'll be super usefull to mention this in the documentation, I've struggeld a while until I found this here.

bilderbuchi, 2022-08-25:
> It'll be super usefull to mention this in the documentation

With "this" you mean the connection string format that PyVISA can work with?
Could you make a PR that adds a paragraph before "For instruments with standard SCPI commands," in [this section](https://pymeasure.readthedocs.io/en/latest/tutorial/connecting.html#connecting-to-an-instrument) in the docs? This could give one example of a serial connection string (`ASRL12::INSTR` in your case) and a link to https://pyvisa.readthedocs.io/en/latest/introduction/names.html#intro-resource-names or https://pyvisa.readthedocs.io/en/latest/introduction/communication.html, for example.

mpikmwm, 2022-08-25:
With "this" you understood me right. I'm thinking that maybe it could be sufficent to point out that every pyvisa resource can be used and make an example with the "COM12" or what ever.

What do you mean with PR ?

bilderbuchi, 2022-08-25:
Ah, I meant "pull request", see https://pymeasure.readthedocs.io/en/latest/dev/contribute.html#making-a-pull-request
If that's too far outside your comfort zone, maybe someone else wants to pick this up (it's not a big task, basically editing one paragraph in https://github.com/pymeasure/pymeasure/blob/master/docs/tutorial/connecting.rst)

mpikmwm, 2022-08-25:
Ah okay, I'll try :)

mpikmwm, 2022-08-25:
So I've written something. hope it is understandable. I'm not sure about the formating, never hat to edit an .rst