pymeasure/pymeasure #532: For connecting to instruments, reccomnd using b"address"

ggggggggg, 2021-11-29:
I spent about an hour trying and failing to connect to a Keithley 33500B with `a = pymeasure.instruments.agilent.Agilent33500("USB0::0x0957::0x2C07::MY57801982::0::INSTR")` before I decided to try `a = pymeasure.instruments.agilent.Agilent33500(b"USB0::0x0957::0x2C07::MY57801982::0::INSTR")` which worked. I have no idea how often `b""` is needed, but since it probably will never break something, maybe every example should use it?

ggggggggg, 2021-11-29:
Nevermind it doesn't actually work, it just fails in a different way.

closed 2021-11-29 by ggggggggg

dphaas, 2021-12-22:
335xxB is currently manufactured by Keysight (formerly Agilent) not Keithley.  This works for me:
`from pymeasure.instruments.agilent import Agilent33500
awg = Agilent33500("USB0::0x0957::0x2707::MY52303253::INSTR")
awg.reset()`