pymeasure/pymeasure #542: Serial connection end of line termination

laserlab, 2021-12-10:
I'm trying to use a SR830 through the serial port

``` python 
from pymeasure.instruments.srs import SR830
from pymeasure.adapters import SerialAdapter
import serial
ser = SerialAdapter(port='/dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0',
                       baudrate=9600,
                       parity=serial.PARITY_NONE,
                       stopbits=serial.STOPBITS_ONE,
                       bytesize=serial.EIGHTBITS,
                       timeout=1)
lockin = SR830(ser,write_termination="\r",read_termination="\r")
```

I'm having trouble with the end of line termination which is \r for this instrument.

if I try:
```
ser.ask("*IDN?\r")
```
I get the correct response:
```
'Stanford_Research_Systems,SR830,s/n88888,ver1.03 \r'
```
However, for the implemented commands to work I need to have the EOL character added automatically, which I thought `write_termination="\r"` should do.

But that doesn't work:
```
lockin.ask("*IDN?")
```
times out.

However:
```
lockin.ask("*IDN?\r")
```
works.

There is issue #160 which seems similar, but any advise would be appreciated.

bilderbuchi, 2021-12-10:
>` SR830(ser,write_termination="\r",read_termination="\r")`

Those kwargs are only used by the VISAAdapter that gets used when you pass a resource ID as a string. If you are using a separately-constructed adapter, kwargs are ignored. If you can afford to wait a little, #539 fixes an important issue around customizing connection settings, probably you don't need to use a `SerialAdapter` anymore. 

Aside of that, neither pyserial nor the SerialAdapter support read/write_termination attributes, yet (I think there's an issue about that)

msmttchr, 2021-12-10:
As highlighted by @bilderbuchi this is being addressed and a temporary workaround could be the following:
- use a VISA adapter string address for the instrument adapter parameter, e.g. something like `"ASRL1::INSTR"`.
- Pass your read/write_termination parameters
- In this case you need to pay attention since baudrate and other serial port related parameters are ignored and so you need to do something like this:
`lockin.adapter.connection.baud_rate = 9600` if the default baud rate is not ok for you.

A bit tricky, bit it should work.

laserlab, 2021-12-12:
Thanks for the quick responses.

@msmttchr At least using a VISA resource name is not a quick workaround since no resources are available

neither
```
rm = pyvisa.ResourceManager('@py')
rm.list_resources()
```
nor
```
rm = pyvisa.ResourceManager()
rm.list_resources()
``` 
return anything.

However I might have to investigate this further since everything is running in a docker container anyway which might complicate things for PyVISA.

msmttchr, 2021-12-12:
I think you need to make sure that you are able to see the serial port.
On my linux PC, I have something like this:
```
micheles@q4os-desktop:~/Desktop$ dmesg | grep ACM0
[  303.055289] cdc_acm 3-1:1.2: ttyACM0: USB ACM device
micheles@q4os-desktop:~/Desktop$ ls -la /dev/ttyACM0
crw-rw-rw- 1 root dialout 166, 0 Dec 12 18:29 /dev/ttyACM0
```
and then this is the output of the basic test:
```
Python 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyvisa
>>> pyvisa.ResourceManager('@py').list_resources()
('ASRL/dev/ttyACM0::INSTR',)
>>> pyvisa.ResourceManager().list_resources()
('ASRL/dev/ttyACM0::INSTR',)
>>> pyvisa.__version__
'1.11.3'

```

msmttchr, 2021-12-13:
> So I got that working by linking the serial port to a serial port name
> PyVISA likes.
>
> ln -s /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0 /dev/ttyACM0
>
> generates a symbolic link to the serial port named "ttyACM0", which PyVISA
> recognizes:
>
> rm = ***@***.***')
> print(rm.list_resources())
>
> gives me now: ('ASRL/dev/ttyACM0::INSTR',)
>
> and such a:
>
> lockin = SR830("ASRL/dev/ttyACM0::INSTR",write_termination="\r",read_termination="\r")
> print(lockin.id)
>
> is successful.
>
> As I mentioned before I run this all in docker and so there is change to
> add that to the Dockerfile. I can post that here once I get it working.
> Might proof useful. I would keep the issue open till then, but feel free to
> close it beforehand.
>
> Thanks so much for the help.

I think you are using a usb to rs232 cable, in such case to get the symlink
done automatically, you probably need to play with udev rules.
Hope this helps

bilderbuchi, 2021-12-14:
You can also check the output of `lsusb` and `dmesg` after you connect the usb device. 
In my case, with a USB to serial adapter, I see at the end of dmesg:
```
[185785.755580] usb 2-4: new full-speed USB device number 50 using xhci_hcd
[185785.904319] usb 2-4: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 3.00
[185785.904325] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[185785.904329] usb 2-4: Product: USB 2.0 To COM Device
[185785.904331] usb 2-4: Manufacturer: Prolific Technology Inc.
[185786.109083] usbcore: registered new interface driver usbserial_generic
[185786.109096] usbserial: USB Serial support registered for generic
[185786.111042] usbcore: registered new interface driver pl2303
[185786.111056] usbserial: USB Serial support registered for pl2303
[185786.111084] pl2303 2-4:1.0: pl2303 converter detected
[185786.112626] usb 2-4: pl2303 converter now attached to ttyUSB0
```
and there's a corresponding entry in lsusb:
```
Bus 002 Device 050: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
```
Using the vendor and product ID from above, you'll probably be able to turn this into a proper USB resource name for pyvisa (I'm not sure about the serial number in this case)

> USB::0x1234::125::A22-5::INSTR | A USB Test & Measurement class device with manufacturer ID 0x1234, model code 125, and serial number A22-5. This uses the device’s first available USBTMC interface. This is usually number 0.

(In my case this appears with list_resources, though, so maybe it's really Docker throwing a spanner in the works, here)

laserlab, 2021-12-17:
So it turns out that the udev service is not available in docker:

https://stackoverflow.com/questions/62060604/why-udev-init-script-default-disable-container-support-while-in-fact-it-works

However as described in the link above you can manually enable it and it does in fact work. However this is pretty cumbersome, and I rather go with the symbolic link solution posted above.

For a more permanent solution I will monitor #536 so that I don't need PyVISA at the first place.

closed 2021-12-17 by laserlab