pymeasure/pymeasure #546: Testing aborted running tests/display/test_inputs.py

bicarlsen, 2021-12-15:
Trying to run the test suite on an Ubuntu 21.04 I get the error below. However, when running on Ubuntu 20.04 everything runs fine.

```bash
$ python setup.py test
running pytest
Searching for pyvisa-sim>=0.4.0
Best match: PyVISA-sim 0.4.0
Processing PyVISA_sim-0.4.0-py3.8.egg

Using /home/projects/pymeasure/.eggs/PyVISA_sim-0.4.0-py3.8.egg
Searching for pytest-qt>=2.4.0
Best match: pytest-qt 4.0.2
Processing pytest_qt-4.0.2-py3.8.egg

Using /home/projects/pymeasure/.eggs/pytest_qt-4.0.2-py3.8.egg
Searching for stringparser
Best match: stringparser 0.5
Processing stringparser-0.5-py3.8.egg

Using /home/projects/pymeasure/.eggs/stringparser-0.5-py3.8.egg
running egg_info
writing PyMeasure.egg-info/PKG-INFO
writing dependency_links to PyMeasure.egg-info/dependency_links.txt
writing requirements to PyMeasure.egg-info/requires.txt
writing top-level names to PyMeasure.egg-info/top_level.txt
reading manifest file 'PyMeasure.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no previously-included files matching '__pycache__' found under directory '*'
adding license file 'LICENSE.txt'
adding license file 'AUTHORS.txt'
writing manifest file 'PyMeasure.egg-info/SOURCES.txt'
running build_ext
============================= test session starts ==============================
platform linux -- Python 3.8.8, pytest-6.2.5, py-1.10.0, pluggy-1.0.0
PyQt5 5.12.3 -- Qt runtime 5.12.9 -- Qt compiled 5.12.9
rootdir: /home/projects/pymeasure
plugins: qt-4.0.2, anyio-3.3.3
collected 108 items / 2 skipped / 106 selected                                                                                                                     

tests/test_log.py ...                                                                                                                                        [  2%]
tests/test_process.py ..                                                                                                                                     [  4%]
tests/test_thread.py ..                                                                                                                                      [  6%]
tests/adapters/test_adapter.py ..                                                                                                                            [  8%]
tests/adapters/test_prologix.py ....                                                                                                                         [ 12%]
tests/adapters/test_serial.py ....                                                                                                                           [ 15%]
tests/adapters/test_visa.py .....                                                                                                                            [ 20%]
tests/display/test_inputs.py Fatal Python error: Aborted

Current thread 0x00007f902948f740 (most recent call first):
  File "/home/projects/pymeasure/.eggs/pytest_qt-4.0.2-py3.8.egg/pytestqt/plugin.py", line 41 in qapp
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/fixtures.py", line 932 in call_fixture_func
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/fixtures.py", line 1126 in pytest_fixture_setup
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_callers.py", line 39 in _multicall
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_manager.py", line 80 in _hookexec
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_hooks.py", line 265 in __call__
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/fixtures.py", line 1072 in execute
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/fixtures.py", line 687 in _compute_fixture_value
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/fixtures.py", line 601 in _get_active_fixturedef
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/fixtures.py", line 581 in getfixturevalue
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/fixtures.py", line 568 in _fillfixtures
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/python.py", line 1647 in setup
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/runner.py", line 449 in prepare
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/runner.py", line 150 in pytest_runtest_setup
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_callers.py", line 39 in _multicall
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_manager.py", line 80 in _hookexec
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_hooks.py", line 265 in __call__
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/runner.py", line 255 in <lambda>
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/runner.py", line 311 in from_call
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/runner.py", line 254 in call_runtest_hook
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/runner.py", line 215 in call_and_report
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/runner.py", line 120 in runtestprotocol
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/runner.py", line 109 in pytest_runtest_protocol
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_callers.py", line 39 in _multicall
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_manager.py", line 80 in _hookexec
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_hooks.py", line 265 in __call__
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/main.py", line 348 in pytest_runtestloop
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_callers.py", line 39 in _multicall
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_manager.py", line 80 in _hookexec
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_hooks.py", line 265 in __call__
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/main.py", line 323 in _main
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/main.py", line 269 in wrap_session
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/main.py", line 316 in pytest_cmdline_main
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_callers.py", line 39 in _multicall
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_manager.py", line 80 in _hookexec
  File "/home/user/anaconda3/lib/python3.8/site-packages/pluggy/_hooks.py", line 265 in __call__
  File "/home/user/anaconda3/lib/python3.8/site-packages/_pytest/config/__init__.py", line 162 in main
  File "/home/projects/pymeasure/.eggs/pytest_runner-5.3.1-py3.8.egg/ptr.py", line 220 in run_tests
  File "/home/projects/pymeasure/.eggs/pytest_runner-5.3.1-py3.8.egg/ptr.py", line 209 in run
  File "/home/user/anaconda3/lib/python3.8/distutils/dist.py", line 985 in run_command
  File "/home/user/anaconda3/lib/python3.8/distutils/dist.py", line 966 in run_commands
  File "/home/user/anaconda3/lib/python3.8/distutils/core.py", line 148 in setup
  File "/home/user/anaconda3/lib/python3.8/site-packages/setuptools/__init__.py", line 153 in setup
  File "setup.py", line 28 in <module>
Aborted (core dumped)
```

CasperSchippers, 2021-12-17:
From this error message I can't tell what's going wrong; unfortunately I don't have access to an Ubuntu PC at this moment to test it out.

bilderbuchi, 2021-12-17:
Does this also happen if you run plain `pytest`, i.e. without going through `python setup.py test`?

bicarlsen, 2021-12-18:
When running just `pytest` I get the error `fixture 'qtbot' not found`.

bilderbuchi, 2021-12-18:
Weird. Where/when you execute that, is pytest-qt available/installed?
Do you see a plugins "qt" entry at the beginning of the pytest output?

bicarlsen, 2021-12-18:
Indeed `pytest-qt` was not installed, however, after installing it the original error is given when running either `python setup.py test` or `pytest`.

bilderbuchi, 2021-12-18:
Ok, so setuptools is not involved. Can you isolate which test this occurs in, and if it also occurs if only that test runs? 
Try "pytest -v", "pytest -v -k test_inputs", "pytest -v -k a-test-name"

bicarlsen, 2021-12-18:
It occurs in all of the tests in `display/test_inputs.py` because they all use `qtbot`.
e.g. If I add comment out all the test in the file then add a test that doesn't use `qtbot`, such as
```python
def test_me( self, default_value ):
        pass
```
everything works. However if I modify the test to include `qtbot`, I get the original error posted.
```python
def test_me( self, qtbot, default_value ):
        pass
```

bilderbuchi, 2021-12-18:
This is great repro! It indicates that the problem does not seem to come from the pymeasure side, though. 
You can check that your pytest-qt and pytest are on the latest version, and that it occurs with just that `test_me` file in a folder separate from pymeasure (you can just run `pytest` there and it should pick it up fine). 
If that does not fix things, please investigate in the [pytest-qt issue tracker](https://github.com/pytest-dev/pytest-qt/issues) and maybe also the pytest issue tracker.

I'll close this. Feel free to reply if you have more questions, or reopen if it unexpectedly turns out that this is something we cause.

closed 2021-12-18 by bilderbuchi