pymeasure/pymeasure #548: joined_validators() does not appear to work as intended

dphaas, 2021-12-18:
**validate() should only require value parm and not values also.  Very possibly I've made a mistake but the modified function and test snippet is below.**

`def strict_discrete_set(value, values):
    """ Provides a validator function that returns the value
    if it is in the discrete set. Otherwise it raises a ValueError.

    :param value: A value to test
    :param values: A set of values that are valid
    :raises: ValueError if the value is not in the set
    """
    print('Running strict_discrete_set')
    if value in values:
        return value
    else:
        raise ValueError('Value of {} is not in the discrete set {}'.format(
            value, values
        ))

def strict_range(value, values):
    """ Provides a validator function that returns the value
    if its value is less than the maximum and greater than the
    minimum of the range. Otherwise it raises a ValueError.

    :param value: A value to test
    :param values: A range of values (range, list, etc.)
    :raises: ValueError if the value is out of the range
    """
    print('Running strict_range')
    if min(values) <= value <= max(values):
        return value
    else:
        raise ValueError('Value of {:g} is not in range [{:g},{:g}]'.format(
            value, min(values), max(values)
        ))

def joined_validators(validators):
    """ Join a list of validators together as a single.
    Expects a list of validator functions and values.
    [(func1, vals1_list), (func2, vals2_list), ...]

    :param validators: an iterable of other validators
    """
    def validate(value):
        for validator, vals in validators:
            try:
                return validator(value, vals)
            except (ValueError, TypeError):
                pass
        raise ValueError(f'Value of {value} not in chained validator set')

    return validate

validators_to_join = [(strict_discrete_set, ['MAX','MIN']),
                      (strict_range, [0, 100])]
v_list = joined_validators(validators_to_join)
print(v_list(99))
print(v_list('MIN'))
print(v_list('MAX'))
print(v_list(101))  # ValueError
`

bilderbuchi, 2021-12-22:
The way this works is this:
```
    dphaas_validator = joined_validators(strict_discrete_set, strict_range)
    dphaas_values = [['MAX', 'MIN'], [0, 100]]  # matching the input validators above
    assert dphaas_validator(99, dphaas_values) == 99
    assert dphaas_validator('MIN', dphaas_values) == 'MIN'
    assert dphaas_validator('MAX', dphaas_values) == 'MAX'
    dphaas_validator(101, dphaas_values)  # ValueError
```

I agree that the docstrings could be improved -- I pushed a small update in d367e17.

If this does not actually resolve your problem, feel free to reopen this.

closed 2021-12-22 by bilderbuchi

dphaas, 2021-12-22:
Ahhh, one must pass the joined values every time a value is evaluated.  This was not obvious to me.

As a highly esteemed member in good standing of the Idiots-R-Us club I feel I must lodge a complaint that this is not the most intuitive approach and puts my fellow members at a distinct disadvantage to those more brilliant.

I suppose it would take a committee to change this methodology since it has been out in the wild for >5 years and no others seem to have a problem with it as is.  I yield to your issue closure.

Thanks for taking the time to educate me.
Thanks for taking the time to improve the docstrings.  Nicely done!
Thanks for making the test code more readable.
I hope Santa brings you something nice!

### One last nit picker item.
if its value is less than or equal **than** the maximum and << change 'than' to 'to'
    greater than or equal **than** the minimum of ``values``.  << change 'than' to 'to'
    Otherwise it raises a ValueError.

bilderbuchi, 2021-12-23:
> Ahhh, one must pass the joined values every time a value is evaluated. This was not obvious to me.
> not the most intuitive approach 

Yeah, this is because the validators are intended for use in the Instrument property factories (control|setting|measurement). If you want to avoid that in another use case you can e.g. use `functools.partial` to fill that argument one time.

You're welcome, glad I could help. :-)

@nit pick: Thx, fixed.