pymeasure/pymeasure #562: SR830 issue with returning instrument parameters

FlyingCurryMonster, 2022-01-04:
I've just started using the pymeasure library and hoping to build a script revolving around the SR830 and eventually an AFG waveform generator.  I run into strange issues when simply trying to return instrument settings in jupyter notebooks.

`import pymeasure`
`from pymeasure.instruments.srs import SR830 `

`lockin = SR830('GPIB2::25::INSTR')`
`print(lockin.id)`
`print(lockin.sine_voltage)`
`print(lockin.x)`
`print(lockin.frequency)`
`print(lockin.phase)`

The output I get is below.  It has a weird lag in the parameters I print.  My sine voltage is set to 1.3V and frequency is 1000Hz

0
['Stanford_Research_Systems', 'SR830', 's/n43345', 'ver1.07']
1.3
-0.712894
1000.01

I've tried to use Instrument.clear() to maybe clear and commands that might be hanging but I'm pretty lost as to why there's a lag.  Is there something i need to initialize when I call the SR830 class for the first time to avoid this lag issue?

msmttchr, 2022-01-05:
Does the same problem happens when you run the sequence out of a jupyter notebook?

FlyingCurryMonster, 2022-01-05:
The problem went away after I queried the instrument on NI Max.  I defined a instrument object with the SR830 with pyvisa earlier in the notebook and commented out that code, and the lag disappeared.  Thanks for the help!

closed 2022-01-05 by msmttchr