pymeasure/pymeasure #571: Automatic docstring for Instrument properties

msmttchr, 2022-01-16:
PyMeasure comes with some convenience functions for making properties for `Instrument` sub-classes.
The aim of these properties is to provide basic instrument control and it has an associated docstring.
The docstring often include validity range for parameter or whether a property can be set or not.
All these information are available in the implementation, since the `validator` and `values` parameters specify the applicable range, so the docstring is in part a human readable version of the meaning of these parameters.

We could envisage two problems:

1) Duplication creates extra work
2) Inconsistency may happen if the code is changed but docstring isn't.

@CasperSchippers highlighed some problem of docstring incosistency in PR #376 and
@bilderbuchi hinted the possibility of enhancing the user docstring provided by users during property creation as part of https://github.com/pymeasure/pymeasure/pull/376#issuecomment-1012399933

It could be nice, if we can implement this to remove the burden to mention validity range or whether the property can be set or not as a part of the doc parameter. Also we could have a sort of standardization in providing this information to users.

Some challenge could be that validator may be complex to decode (eg. joined validators).

bmoneke, 2022-05-27:
As you have to write manually, what the property is about, you can write the validation as well.

The set/get/values can be added easily and reduce the amount of needed writing.

bilderbuchi, 2022-10-31:
A preliminary format that I have used when updating the documentation in #712 is appending some info to the first/summary line of a property docstring:
`(<type> [if strict validator: "strictly"] [if set: "in <values>", else if range: "from <value0> to <value1>"] [if dynamic: ", dynamic"])`.

It looks like this could be nicely automated.

bmoneke, 2022-11-03:
One question for discussion: Should we prepend the property type (measurement, setting, control), according to the coding style in the new documentation (measure, set, control) or should the user write it?

Pro: Users won't forget it and maintainers don't have to remind.
Con: User won't know it and the result won't match: "Set" + "Configure the turning rate" does not sound great.

It's an idea I want to raise, even though I'm slightly against it.

LongnoseRob, 2022-11-03:
> One question for discussion: Should we prepend the property type (measurement, setting, control), according to the coding style in the new documentation (measure, set, control) or should the user write it?

> One question for discussion: Should we prepend the property type (measurement, setting, control), according to the coding style in the new documentation (measure, set, control) or should the user [...]

I understand your idea behind it, but beside the pro&cons you mentioned it also would make the source less comprehensible if the docstring is reduced to `"""the turning rate"""`.

On the other hand this could also go well with a unit-based suffix (if there ever will be unit support)

bilderbuchi, 2022-11-05:
I would like to get/keep contributors in the habit of writing PEP257-conforming docstrings, so I'm also for the user-supplied docstring including the verb. Appending something autogenerated to the summary line before the period I'm ok with.

This could btw. make for a good automated check -- go through all instruments, look at all properties, cross-check verb usage (allow both "Read" and "Measure" for a measurement) with presence of `fget`, `fset`, complain if not fulfilled. :grin:

bmoneke, 2023-02-13:
For https://github.com/pymeasure/leco-protocol/issues/45 we need a machine readable docstring.

Ideas for automatic creation:
- The property creators may check, that the first word of the dosctring is appropriate for the property (get/measure/set/control).
- We could add the property type in front of the variable type: `(get float strictly from...)`. That way the parenthesis in the end contains all machine readable information.
- We could use the "cast" parameter in order to automatically read the variable type. A `setting` could infer it from the set command. Additionally, we should add a "type" parameter (name tbd), in order to override the automatic detection.

bilderbuchi, 2023-02-13:
We could also put CI mechanisms in place to check docstring compliance (maybe via [ruff](https://github.com/charliermarsh/ruff)), just like we lint other things.

bilderbuchi, 2023-02-13:
Alternatively/in addition, if we just want to serve https://github.com/pymeasure/leco-protocol/issues/45, we could autogenerate a condensed description of the relevant information and append that to the docstring. The source would remain readable, we don't have to be precise/naggy about docstring format, and the autogenerated slug will be out of the way. E.g.
```
""" Control the source voltage (in volt).

type info: RWF0.0..1e3
"""
```
denotes a read-write float property, with valid values from 0.0 to 1e3.

bmoneke, 2023-02-13:
The most difficult part, in my opinion, is the variable type to get right. Especially bool, we can barely recognize, unless the users help.

bmoneke, 2023-05-10:
Side note: This automation could be a solution for #543 , at least, it the automated docstrings should follow correct docstring style.