pymeasure/pymeasure #573: command_process parameter

msmttchr, 2022-01-19:
During review of PR #376, some doubts were raised about the usefulness of parameter `command_process` in property creation helper methods (https://github.com/pymeasure/pymeasure/pull/376#discussion_r536805762)

The original idea of this parameter is to allow users to process command string before sending to the instrument.

I think, specific use cases needs to be identified and possibly mentioned in user documentation or make a deprecation and eventually remove this parameter

bmoneke, 2023-03-07:
A quick check reveals, that we just use it for a deprecation warning.

I think it might have been useful for channels before we implemented `Channel`.

If we do not find any use case, I'm for deprecating this option.

bilderbuchi, 2023-03-11:
I don't have an opinion on this, either way is fine by me.

bmoneke, 2023-03-14:
Do you, @msmttchr or @CasperSchippers see any use specific use case?

If not, I'd deprecate the option.

msmttchr, 2023-03-14:
Ok for me to go with deprecation.

PR #878 (2023-03-27): Parameter `command_process` of property creators is deprecated.
Closes #573
closed 2023-03-27 by bmoneke