pymeasure/pymeasure #583: Warning in documentation build SwissArmyFake not found

msmttchr, 2022-01-29:
When building hml documentation an error is reported as follow:
```
Traceback (most recent call last):
  File "/home/micheles/.local/lib/python3.8/site-packages/sphinx/util/inspect.py", line 412, in safe_getattr
    return getattr(obj, name, *defargs)
AttributeError: module 'pymeasure.instruments' has no attribute 'SwissArmyFake'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/micheles/.local/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 110, in import_object
    obj = attrgetter(obj, mangled_name)
  File "/home/micheles/.local/lib/python3.8/site-packages/sphinx/ext/autodoc/__init__.py", line 318, in get_attr
    return autodoc_attrgetter(self.env.app, obj, name, *defargs)
  File "/home/micheles/.local/lib/python3.8/site-packages/sphinx/ext/autodoc/__init__.py", line 2606, in autodoc_attrgetter
    return safe_getattr(obj, name, *defargs)
  File "/home/micheles/.local/lib/python3.8/site-packages/sphinx/util/inspect.py", line 428, in safe_getattr
    raise AttributeError(name) from exc
AttributeError: SwissArmyFake

```
I believe this has been inadvertently introduced by PR #552, but I am not sure why it was not detected by automatic checks.
@bilderbuchi, any idea ?

bilderbuchi, 2022-01-29:
Agree on both points. Don't know how this was missed, maybe the squash merge was not built?

The reason is mist probably that we removed the fakes from the instruments namespace late in the PR, although I see a green checkmark after that.

bilderbuchi, 2022-01-29:
In any case I agree, we need to find out why this was not discovered automatically,  and fix that.

msmttchr, 2022-01-29:
Actually, looking at the build logs, the warning is there, but possibly is not considered a failure condition.

bilderbuchi, 2022-01-29:
Yeah, sphinx considers [this a warning](https://github.com/pymeasure/pymeasure/runs/4942688875?check_suite_focus=true#step:10:154). The thing is, sphinx warnings should not stop the build, I'll see if I can tune to make exceptions an error, instead.

bilderbuchi, 2022-01-29:
So, it's possible with a -W flag to have sphinx treat warnings as errors (then we could fail the build to detect this) with `make html SPHINXOPTS="-W --keep-going"`. 
However, this also catches the syntax warnings like "underline too short" etc, which we only want to parse into hints in the diff view, but not fail the build. 
Apparently (https://github.com/sphinx-doc/sphinx/issues/3926) in this case it's only possible to _ignore_ the autodoc-related warnings (i.e. not have them turned into error). We want the other way round: treat everything as warnings, _except_ autodoc import errors. This does not seem to be possible with Sphinx. Someone else requested the same thing in https://github.com/sphinx-doc/sphinx/issues/8516, but apparently this is not desired.

bilderbuchi, 2022-01-29:
What we could do is extend the problem matcher in .github/sphinx.json to match on the autodoc warning, so we at least get a bubble (although the warning message does not contain a lot of info to point us to the exact place it occurs.)

bilderbuchi, 2022-01-30:
Closed by #584

closed 2022-01-30 by bilderbuchi