pymeasure/pymeasure #592: Managed Window example plotter Xaxis scale is not correct

HoWeBrz, 2022-02-17:
I am running the MaganedWindows example from the website with todays latest git-version.
The measurement points are not correctly adjusted on the x-axis.
Please see in the attached picture - the iteration 0 is not at x=0  and the same for the others.
I consider this a bug - or maybe I did something wrong ?

[image]

HoWeBrz, 2022-02-17:
I have found the same behaviour in the pyMeasure version 0.9 (installed via pip) !

HoWeBrz, 2022-02-17:
For you it is working ?
Would it be possible to quickly confirm, by copying the code from here: [https://pymeasure.readthedocs.io/en/latest/tutorial/graphical.html#using-the-managedwindow](url)

HoWeBrz, 2022-02-17:
I am usng this versio of pyqtgraph:
......> pip install pyqtgraph --upgrade --force-reinstall
Collecting pyqtgraph
  Using cached pyqtgraph-0.12.3-py3-none-any.whl (992 kB)
Collecting numpy>=1.17.0
  Using cached numpy-1.22.2-cp39-cp39-win_amd64.whl (14.7 MB)
Installing collected packages: numpy, pyqtgraph
  Attempting uninstall: numpy
    Found existing installation: numpy 1.22.2
    Uninstalling numpy-1.22.2:
      Successfully uninstalled numpy-1.22.2
Successfully installed numpy-1.22.2 pyqtgraph-0.12.3

bilderbuchi, 2022-02-17:
Ping @CasperSchippers do you know more about this GUI issue?

HoWeBrz, 2022-02-17:
Can anyone confirm this was working properly before ? 
Or did it break just recently ?

Since the overall features of pymeasure sound very promising, I am planning to use pyMeasure in our new lab.
As life plotting is one very basic feature, I am wondering how stable pyMeasure is and if it can be used reliably in labs for evaluation/verification/qualification.

CasperSchippers, 2022-02-17:
Also, actually, I don't have any grid-lines on by default; did you manually turn them on, or where they on by default?

Regarding stability; I've been using it in our lab for the past few years now and, small bugs aside, had never any real problems with it (at least nothing that can be attested to pymeasure).

mcdo0486, 2022-02-17:
[image]
It looks like you are using Windows, can you make sure the image scaling is set to 100%, especially if you are using two monitors? 

I had a researcher that had monitor 1 at 125% (or w/e) and monitor 2 at 100% and the axis would get messed up on the second monitor. The axis label large font size hints that may be the issue.

HoWeBrz, 2022-02-18:
Next obvious question is, if the plots can be made aware of the display-settings and scale correctly, even if another value than 100% is used ?

bilderbuchi, 2022-02-18:
To answer this question, I recommend heading over to the pyqtgraph issue tracker, a very quick (and definitely not exhaustive) search yields https://github.com/pyqtgraph/pyqtgraph/pull/1509 and https://github.com/pyqtgraph/pyqtgraph/issues/756#issuecomment-1023182391

HoWeBrz, 2022-02-18:
Thanks for the links - it seems they are working on it, or even did implements a solution ?

I am not skilled to properly interprete the different #xyz numbers from the discussions and relate them to any git version or branch (I am not a developer - just a user ;-) ). Can I see from those discussions, if a fix is already implemented, and if yes, how can I get it (e.g. whic git version do I need to download)  ?

xMesa-git, 2023-03-08:
Having same issue on:

pyqtgraph                  0.13.1
PySide6                      6.4.2

mcdo0486, 2023-03-08:
> Having same issue on:
> 
> pyqtgraph 0.13.1 PySide6 6.4.2

Can you provide a screenshot of the issue?

Does the workaround of setting "Scale and Layout" display to 100% work?

xMesa-git, 2023-03-08:
i edited my original comment