pymeasure/pymeasure #597: keithley 2450 voltage measurement with option 4 wires

ced091, 2022-02-23:
Hello,
I find a problem when I want to use this command : 
k2450.wires = 4

Indeed, when I want to make a voltage measurement in 4-point mode and I use this command line, nothing happens.
I found in the source code of the keitley 2450 this:
wires = Instrument.control(
        ":SENS:RES:RSENSE?", ":SENS:RES:RSENSE %d",
        """ An integer property that controls the number of wires in
        use for resistance measurements, which can take the value of
        2 or 4.
        """,
        validator=strict_discrete_set,
        values={4: 1, 2: 0},
        map_values=True
    )
But I want to use the 4 wires in voltage measurement.
So I replaced this:
":SENS:RES:RSENSE?", ":SENS:RES:RSENSE %d",
by:
":SENS:VOLT:RSENSE?", ":SENS:VOLT:RSENSE %d",
This is not a good way to do things I understand, but it works.

ced091

msmttchr, 2022-02-23:
Apparently, this function is not supported in the instrument driver.
A possible workaround that will not require to edit the file is:
```
k2450.write(":SENS:VOLT:RSENSE 4") # To set
k2450.ask(":SENS:VOLT:RSENSE?") # To query
```
If you want, you could also add this properly by having a look at documentation (https://pymeasure.readthedocs.io/en/latest/dev/contribute.html) and submitting a PR

achadha0111, 2022-07-14:
That makes sense, the program seems to execute without error (and I see MEASURE VOLTAGE 4-WIRE) show up on the screen of the instrument too. However, when I use the ask function, I seem to be getting the same voltage reading no matter the current applied even when I change the resistance of the devices under test.

```
def startup(self):
        log.info("Connecting and setting up instrument")
        self.sourcemeter = Keithley2450("USB0::0x05E6::0x2450::04365678::INSTR")
        self.sourcemeter.reset()
        self.sourcemeter.use_front_terminals()
        # https://github.com/pymeasure/pymeasure/issues/597
        self.sourcemeter.apply_current()
        self.sourcemeter.enable_source()
        self.sourcemeter.measure_voltage()
        self.sourcemeter.write(":SENS:VOLT:RSENSE 4")
        time.sleep(2)
        
    def execute(self):
        currents = np.linspace(
        self.min_current,
        self.max_current,
        num=self.data_points)
        
        for current in currents:
            log.info("Current is %g A", current)
            self.sourcemeter.source_current = current
#             self.sourcemeter.reset_buffer()
#             time.sleep(0.1)
#             self.sourcemeter.start_buffer()
            
            voltage = self.sourcemeter.ask(":SENS:VOLT:RSENSE?")
#             self.sourcemeter.wait_for_buffer()
            self.emit('results', {
                'Current (A)': current,
                'Voltage (V)': voltage
#                 'Voltage (V) std': self.sourcemeter.standard_devs,
            });
            
            time.sleep(1)
            if self.should_stop():
                log.info("Procedure aborted")
                break
    
    def shutdown(self):
        self.sourcemeter.shutdown()
        log.info("Measurement finished")```

achadha0111, 2022-07-18:
Sorry about the delay in getting back to you. I was able to make it work by editing the source code and adding the property. I shall open a pull request for it once I have completed my measurements lest I break something with the installation on my computer.