pymeasure/pymeasure #599: 'USBInstrument' object has no attribute 'ask'

HoWeBrz, 2022-03-10:
I do have some instruments connected - one is a USB connected Rigol Digital Oscilloscoe.
When trying to list the available instruments to find out the VISA-address, I get:

```
>>>import pymeasure.instruments as pmi
>>>pi.list_resources()
Traceback (most recent call last): 
  File "D:\LabTestPy\venvPy39\lib\site-packages\pymeasure\instruments\resources.py", line 50, in list_resources
    idn = res.ask('*idn?')[:-1]
AttributeError: 'USBInstrument' object has no attribute 'ask'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "D:\LabTestPy\venvPy39\lib\site-packages\pymeasure\instruments\resources.py", line 55, in list_resources
    print(n, ":", instr, ":", idn)
UnboundLocalError: local variable 'idn' referenced before assignment
```
Apparently, the USB instrument does not have the "ask()" method implemented.

When trying the same directly with the pyVisa module (wothout the PyMeasure-wrapper) it works:
```
>>> import pyvisa as pv
>>> rm = pv.ResourceManager()
>>> instrs = rm.list_resources()
>>> print(instrs)
('USB0::0x1AB1::0x0588::DS1ET161654496::INSTR', 'ASRL3::INSTR', 'ASRL4::INSTR', 'ASRL5::INSTR')
```

I am able to properly communicate to the USB device with SCPI commands, once I know the VISA adress.

1. Is this "ask()" method missing intentionally for USB instruments, or is this a bug ?
2. Can someone confirm this ?
3. How can I bug-fix this quickly - any recommendations ?

msmttchr, 2022-03-10:
`PyVISA` replaced `ask` method with `query` method, the current code base already contains the fix (https://github.com/pymeasure/pymeasure/blob/135711821d3a4c5f8de709161411303f301c4a16/pymeasure/instruments/resources.py )

msmttchr, 2022-04-10:
With v0.10 out, this should be fixed now. @HoWeBrz are you OK to close this issue ?

HoWeBrz, 2022-04-19:
I confirm the issue is fixed in v0.10 - please close the issue !

closed 2022-04-19 by msmttchr