pymeasure/pymeasure #602: FloatParameter input in PyMeasure GUI removes decimal

En1kay, 2022-03-21:
Hello,
I'm currently running the managed window example from the tutorial page. Whenever I select the delay field or change the value it changes from e.g. 0.02 to 2 or 0.11 to 11. I reproduced the problem on both windows and Mac with the latest pip package 0.9.0 on python 3.8.9 and 3.9.3.

Thanks in advance.

msmttchr, 2022-03-21:
I think this is linked to the regional settings in your computer. Is your PC settings using , (comma) as decimal separator ?

En1kay, 2022-03-21:
Yes my PC is using a comma as decimal separator. I also suspected this to be the problem however:
* entering e.g. 0,02 or 11,2 and hitting enter results in the default value 0.2 being displayed
* entering e.g. 2e-2 and hitting enter results in the correct input 0.02 being displayed for as long as I don't change the focus (select any other buttons or input fields). As soon as I change the focus the value changes to 2.
* while writing this and testing my claims I found out that entering 2e-2 and changing focus to another input without hitting enter does result in the value being accepted

msmttchr, 2022-03-21:
I think this is a known limitation (see #310), but I am not sure if it is related to PyQt or to pymeasure itself.
The only workaround I know is to set '.' as decimal separator.

En1kay, 2022-03-21:
Thank you very much. I will have a look around the PyQt5 documentation. Maybe I can find a way to override the localization setting for the decimal separator.

En1kay, 2022-03-21:
By importing Locale like `from PyQt5.QtCore import QLocale` right in the tutorial code and adding `    QLocale.setDefault(QLocale(QLocale.English, QLocale.UnitedStates))` right after `app = QtGui.QApplication(sys.argv)` I was able to override the locale settings. Now everything works like expected with the scientific and "." notations.

msmttchr, 2022-03-21:
I would reopen this, so eventually can be properly fixed.

bilderbuchi, 2022-03-22:
We could find out if pyqtgraph inputs work correctly with this, and if yes maybe transition our inputs to those instead of plain Qt.

pyZerrenner, 2024-03-21:
The issue is, that `ScientificInput` uses `QDoubleValidator`, which in turn "[uses its `locale()` to interpret the number](https://www.riverbankcomputing.com/static/Docs/PyQt5/api/qtgui/qdoublevalidator.html)". On German machines, this ignores the dot as decimal separator (`"30.5"` becomes `"305"`). The dot can still be typed in the input field, since it is used for grouping (`"1.000,5"` means `1000.5`).

At the same time, `ScientificInput.valueFromText` simply uses `float` to convert the input text into a number (after removing any units). So typing a comma as decimal separator results in a `ValueError` in which case the default value is restored. 

The other way around, `ScientificInput.valueFromText` converts numbers to a string with `f"{value:g}"`, i.e. it always uses a point as decimal separator. This displays well, as it appears to circumvent the validator. Also using the up/down arrows increments the value correctly.

My current solution is to subclass `ScientificInput` and use the `toDouble` and `toString` methods of the validator's `locale()` for the conversion. Then I initialise my parameters accordingly with `FloatParameter(..., ui_class=ScientificInput_locale)`. These inputs then properly handle the current locale's format. It also works with switching the locale back to English (as suggested above by @En1kay )

```python
import re
from pymeasure.display.inputs import ScientificInput

class ScientificInput_locale(ScientificInput):
    """
    Locale-sensitive spinner input box for floating-point values, connected to
    a :class:`FloatParameter`. This box will display and accept values in
    scientific notation when appropriate.

    .. seealso::
        Class :class:`~.FloatInput`
            For a non-scientific floating-point input box.
    """

    def toDouble(self, string):
        value, success = self.validator.locale().toDouble(string)

        if not success:
            raise ValueError('String could not be converted to a double')
        else:
            return value

    def toString(self, value, fmt='g', prec=6):
        return self.validator.locale().toString(value, fmt, prec)

    def valueFromText(self, text):
        try:
            text = str(text)
            if self._parameter.units:
                text = text[:-(len(self._parameter.units) + 1)]
            return self.toDouble(text)

        except ValueError:
            return self._parameter.default

    def textFromValue(self, value):
        string = self.toString(value).replace("e+", "e")
        string = re.sub(r"e(-?)0*(\d+)", r"e\1\2", string)
        return string
```

pyZerrenner, 2024-03-21:
It would be consistent with the standard behaviour of `QDoubleSpinBox` from pyQt, so I think it makes sense to update `ScientificInput` directly. Feel free to use the code. (In the hope, that it does not introduce any new error sources.)

`IntegerInput` is already locale-sensitive, as it does not re-implement the methods `valueFromText` and `textFromValue`. Here, the locale seems to only concern the grouping character ("," in English and "." in German). Though a `QSpinBox` does [not display them by default](https://doc.qt.io/qtforpython-5/PySide2/QtWidgets/QSpinBox.html#PySide2.QtWidgets.PySide2.QtWidgets.QSpinBox.textFromValue).

BenediktBurger, 2024-03-21:
Would you mind opening a pull request with these changes?

pyZerrenner, 2024-03-21:
I have never before contributed to someone else's project, but I would like to try. Do I have to fork the repository or just clone it, e.g. to my local computer, before making a new branch?

BenediktBurger, 2024-03-21:
> Do I have to fork the repository or just clone it, e.g. to my local computer, before making a new branch?

You have to fork it, because we have to merge the changes from some repository sitting on github and you cannot push to pymeasure directly.

In the most simple case, you can fork and do the changes online in github, but it is better to clone your fork (not pymeasure directly).

Once you have a new branch in your fork, you will see a yellow notice on the pymeasure github page (or the one of your fork) above the "code" section, which invites you to open a PR.

We have a short introduction on our [contribute](https://pymeasure.readthedocs.io/en/latest/dev/contribute.html) page. If you have questions, feel free to ask.

closed 2024-03-25 by CasperSchippers