pymeasure/pymeasure #609: Creating complex systems?

cherkasovn, 2022-04-08:
I am looking for a Python platform to automate chemical experiments.
The key is to control multiple instruments and (most frequently) perform sequences with some interactions between the instruments.

Adding new instruments to PyMeasure is simple and works well. Yet, the challenge is that various instruments have various data rates and some asynchronous communication would be great.

For example, I have several liquid pumps doing something and a measuring device looking at the signals. The pumps need only occasional inputs, while the measurements work frequently. Sending any command to the pump "blocks" the programme and limits usefulness of the measurement data,

Is there any way to do instrument interactions in an asynchronous way?

neal-kepler, 2022-04-08:
This problem and a consequence of trying to fix it is discussed in #506 . You might find that a good place to start for ideas on how to implement it (either asyncio or in the suggestion by @bilderbuchi of the way it was proposed for the anaheim motion controller). 

Otherwise I defer to others more versed in async programming for more thorough suggestions

bilderbuchi, 2022-04-09:
Exactly, you could use separate threads or concurrent.futures to separate instrument communication.
E.g. something like this sketch:
```
with ThreadPoolExecutor(max_workers=len(my_instruments)) as executor:
    futures = {}  # future-name coupling for the logging below (optional)
    for instr in my_instruments:
        futures[executor.submit(instr.do_update)] = instr.name  # schedule whatever function call you want to do here
    for fut in as_completed(futures):
        LOGGER.debug(f'Finished comms with {futures[fut]}')
```
This parallelizes communication with all instruments, I was using something similar for fetching updated state. If you want to do fully asychronous communication at different rates, you need to adapt this of course. E.g. in the as_completed loop immediately submit a new update for the instrument that just finished. That should work, but there are surely a number of resources online dealing with your specific use case, including proper termination etc.

bilderbuchi, 2022-04-09:
As this is more of a usage question, I'll convert this into a discussion.