pymeasure/pymeasure #610: Go even further with breaking out widgets?

neal-kepler, 2022-04-08:
PR #594 broke out the widgets into separate files for readability. I'm wondering if it's worth going even further and complement moving all of the browser logic to the browser_widget.py file from the windows.py file in the same way that sequencer_widget.py does. This would make Managed window really just a layout organizer. It would also require changing the way the other widgets access the manager instance, but that would not be too onerous.

I ask because I would like to add a widget that fires off analyses in a separate queue after the browser reports and experiment is finished and I basically reimplemented a stripped down version of the browser and invoking it like the Sequencer and Estimator widgets. 

```       if self.use_analyzer:
            self.analysis_browser_widget = AnalysisBrowserWidget(
                self.procedure_class,
                self.displays,
                [],  # This value will be patched by subclasses, if needed
                parent=self
            )
```

[image]

bilderbuchi, 2022-04-09:
I'll let @CasperSchippers handle this as I'm not too familiar with the GUI code, but in principle refactoring is good if it improves the structure and in the end yields an "improved" situation. Separating "concerns" seems like an improvement, but I don't have a good grasp on the implications and side effects.

CasperSchippers, 2022-04-19:
I'm also fine with refactoring the way you suggest, I can see how decoupling the browser and the GUI can be useful and improve the structure.

PS sorry for the late response; I might also be a bit slow with responding to/reviewing a PR (wrapping up a PhD is taking a bit more of my time and concentration than I'd hoped 😅)

msmttchr, 2022-04-20:
@CasperSchippers finger crossed for your PhD, just to let you know that we miss a lot your expertise. I hope you will have more time soon :).