pymeasure/pymeasure #616: Use of dynamic properties in instrument "channel"

msmttchr, 2022-04-18:
Just discovered that dynamic properties are not supported in some implementation of instrument "channel". For example, if we consider the following class: https://github.com/pymeasure/pymeasure/blob/d7547d6cf79b85d8bee96958067537095c116df9/pymeasure/instruments/newport/esp300.py#L130
The properties declared inside the class can be defined dynamic by adding the appropriate parameter, but they will not behave as dynamic properties and the original value cannot be overridden.
I think this needs to be taken into account when some guidelines for Channels implementation are provided (see https://github.com/pymeasure/pymeasure/issues/416#issuecomment-1025090693)

bilderbuchi, 2022-04-18:
Good catch! This could be because it's missing something that is coming from `Instrument`, which we normally inherit from. Maybe we should see if we can extract the common/needed bits into a superclass for `Instrument`, and implement the `BaseChannel` that comment mentions, which then also inherits from that class to get this all working as intended?

bilderbuchi, 2022-04-18:
Does it fail "visibly" or does it just not work as expected?

msmttchr, 2022-04-18:
> Good catch! This could be because it's missing something that is coming from `Instrument`, which we normally inherit from. Maybe we should see if we can extract the common/needed bits into a [...]

I agree, think this could be a good approach.

msmttchr, 2022-04-18:
> Does it fail "visibly" or does it just not work as expected?

It just doesn't work, but there is no visible failure.
I was logging it, just to take into account this aspect for the restructuring of `Instrument` class.

msmttchr, 2022-07-19:
See also #649 for a possible solution to this problem

bilderbuchi, 2022-08-02:
> See also #649 for a possible solution to this problem

I have not reviewed this in detail, so I might not know what you refer to exactly, but I think #649 shows a workaround (that will probably bite us later), not a solution (see https://github.com/pymeasure/pymeasure/pull/649#issuecomment-1202467447).

I think implementing a `BaseChannel`/`ChannelBase` as outlined above could resolve this, too, so we should make sure not to forget to

* [ ] fix the AWG-401x workaround for dynamic properties in channels.

Now that #634 is merged (finally! :sweat_smile: ), it should be easier to put some protocol tests in place to nail down the exchanged messages, including dynamicness etc, and then refactor all those classes inheriting from `Instrument`, without having the device present. :-)

PR #718 (2022-11-19): Channel implementation added.
Fixes #616 which complains about missing Channel implementation.

I'm done with the basic implementation:
- Creation of some `Base` class (name is open for suggestions), which contains properties and methods which work inside the class itself (values, ask...). 
- Instrument is based on this Base and calls the adapter. 
- The `Channel` (maybe a different name, I made it similar to Instrument) calls the instrument's methods (instead of adapter) and `write` uses `str.format` to replace '{ch}' in a command with the channel name. I figured, this is the most common use  [... cut]