pymeasure/pymeasure #627: Test adapter for unit tests of devices

bmoneke, 2022-05-20:
What is the recommended way for writing tests on the software itself, without using an actual device?

I know there is the `Fake_Adapter`, but querying it would not work, as it would read back its written message.

If there were some adapter, which could be given a list of responses, which are popped for each readout, and could store a list of sent messages, Tests could verify, for example according to manual examples, that the communication is right.
InstrumentKit uses this kind of adapter to write an `expected_protocol` test routine, which makes it very easy to write tests for devices.

Obviously this depends on the definition of adapter, talked about in #567 about adapters and instruments.

bilderbuchi, 2022-05-20:
I agree that testing without the hardware is a sore point for pymeasure. 
This has been on my wish list for a long time (see #249), sadly still TODO as I never had the free capacity.
In that issue we have discussed `expected_protocol` a little with the IK maintainer -- I know about, it's awesome, and I think we should basically follow that (proven) design.

We have [a few tests](https://github.com/pymeasure/pymeasure/blob/master/tests/adapters/test_visa.py) using pyvisa-sim, but that never really caught on and is not that great to use nor flexible.

In fact, triggered by your PR #621, and the easy tests you could make over in the Ik version of the driver ;-),  I just yesterday thought about this again and resolved that that should be one of the next "bigger" goals I should aim at :D. 

Ultimately, I want to achieve proper support for frame-based devices (#296), but for _that_ work `expected_protocol` would be very useful, if not essential.

First, we'll have to check how `expected_protocol` works internally
I imagine that, yes, we'll need some `ProtocolTestAdapter` that we attach to an instrument to take care of the message exchange.
If we have a good separation of instrument vs. adapter concerns (cf #567) this should work out nicely. Where not I imagine that we will discover some edge cases that we can deal with.

Do you want to help out? 😃

bmoneke, 2022-05-20:
I appreciate that offer and am willing to help out. The next weeks I'm going to be busy with a paper, but I'll be able to it every now and then, and I think it is good to habe some time in between to think about those design ideas.

Tests help a lot writing drivers. In fact, I adapted your suggestions for the hcp ovens to the ik version, ran the tests there and then I pushed them here.

bilderbuchi, 2022-05-20:
Great! No pressure, we'll take it slow.

The wax IK does logging, iirc (I'dhave to check again), is that they have one central read/write method that the adapter-specific read/write call into, and that one does the logging no matter what adapter is used.

bilderbuchi, 2022-05-22:
I took a quick look through the IK codebase, and have a rough idea how this could be implemented in pymeasure. I've started (very rudimentary) work (`expected_protocol` and `ProtocolAdapter` plus tests) at https://github.com/pymeasure/pymeasure/tree/protocol-tests. I'll iterate on this TDD-style -- as soon as this is doing something interesting, I'll open a WIP PR so that we can discuss/review/iterate. Exceptionally, I put the branch in the central repo so it's easier for you to contribute/send PRs against this branch.

For logging, AFAICT IK does not log terminators and other ancillaries, as the universal logger triggers on commands before all that has been added. I would like to log the final commands being sent and received over the wire (including terminators), that should make it much easier to troubleshoot communication issues and e.g. compare with HTerm logging.
Here, too, I have an idea how to do that, hopefully without too much code duplication, but I'll postpone that work until after ` expected_protocol` has matured a bit.

bmoneke, 2022-05-22:
Thank you very much for that first implementation. I like your test driven approach, as that is exactly the reason for this issue. I had planned to do it as well.

For the protocol adapter, you chose to have message pairs, while IK has a send and a receive list. Your approach has the advantage of being easier to understand the communication timeline, but may be more difficult to implement than just pop or push a message at read or send respectively. Or we just convert the list of pairs to a pair of lists.

Regarding the termchar for the protocol test adapter: if the adapter used the termchar, some instruments' tests could just follow the manual. Others (scpi) do not include them in the manual and may have different term chars for different connection types. Maybe we could use a parameter telling the adapter which type of connection (gpib, serial...) it should emulate and whose config dict it should take, another option could enable/disable termchar. 

Regarding the logging of the whole communication: Right now pyvisa, for example, adds the Termchar, so we would need either a visalogadapter which uses read raw and write raw in order to capture everything and then does the work of pyvisa, or we change the visa adapter itself.

For the logging, I find two options, one stores the command, parameters, communication and return value in some kind of file (json, pickled dict...). The other one writes corresponding tests, which could be adapted manually afterwards.

bilderbuchi, 2022-05-23:
> For the protocol adapter, you chose to have message pairs, while IK has a send and a receive list. Your approach has the advantage of being easier to understand the communication timeline, but may [...]

My thinking was that ease of use by the code writer is most important. I browsed some of the uses of `expected_protocol` in IK and sometimes had difficulty identifying which (sent) commands are associated with which (received) messages. 
I think message pairs are clearer and thus easier to come up with. Of course, send and receive lists correspond more closely to the stream-like nature of the device communication flow. If needed, we can always convert from one to the other internally.

bilderbuchi, 2022-05-23:
> Regarding the termchar for the protocol test adapter: if the adapter used the termchar, some instruments' tests could just follow the manual. Others (scpi) do not include them in the manual and may [...]

*different term chars for different connection types* is exactly the problem I thought of, and the indication that this is an adapter-specific thing. I don't want to have different tests for different adapters, whose messages essentially just differ in the term char, if it can be avoided. 
I thought about any more connection parameters influencing the messages, and only Prologix has (I think) additional messages over the wire depending on config settings. @msmttchr do you have input on this?

Including the termination chars would indeed be "closer to the wire" and would enable more comprehensive testing (i.e., we would also confirm that the line endings are correctly configured). OTOH, it's also highly redundant because the terminator is typically the same for all messages/tests for a given instrument.
I have not yet identified a good way to achieve that -- we would probably have to mock out the connection objects of all our adapters, sometimes probably touching 3rd-party (visa, serial) internals. A specific `ProtocolAdapter` is so much cleaner and less complex, but if we find a better way I'm open to ideas.
Maybe we can separate those two concerns -- have one set of tests for the protocol, and one test (could even be semi-automated?) asserting that the terminations are correct.

In manuals I have often seen a generic placeholder like `<MT>` in communication examples. Our list-based definition already naturally accounts for that since the messages are already separate elements, so maybe that is already good enough?

bmoneke, 2022-05-23:
> My thinking was that ease of use by the code writer is most important. I browsed some of the uses of `expected_protocol` in IK and sometimes had difficulty identifying which (sent) commands are [...]

I looked at IKs implementation, they convert the lists internally to a bytes object. We could follow this approach (with our different style of writing, which is clearer to understand and write), or we could have to buffers (send/read) and populate it at each sending with one element of our lists. This would allow to check for timing as well, as the answer may come only after the request has been sent.

bmoneke, 2022-05-23:
For the termchars, I like your proposed combination of tests for the commands themselves and a test for the termchar (one for reading, one for sending)

We have to document well, though, that the tests will exclude the termchar. Or we could allow the option, that the expected protocol strips the term-chars of the expected protocol. That way you can write, if desired, tests with strings from the manual (if it includes term-chars), but you do not test the correct working of the termchar itself. The option could be ":param str strip_termchar: Removes this terminating character from the test messages."

bilderbuchi, 2022-05-23:
> Regarding the logging of the whole communication: Right now pyvisa, for example, adds the Termchar, so we would need either a visalogadapter which uses read raw and write raw in order to capture everything and then does the work of pyvisa, or we change the visa adapter itself.

Yeah after I made my commits I realised that we will maybe have to reach deeper into `visa` than initially expected. I'll have to check, maybe there is a way from inside `visa` (callback?), or something else that enables us to get the actual message sent (preferably in string form, though).

> For the logging, I find two options, one stores the command, parameters, communication and return value in some kind of file (json, pickled dict...). The other one writes corresponding tests, which could be adapted manually afterwards.

Primarily, I think we should stay with the Python `logging` framework. We can have expressive prefixes and log at different levels, which can then be easily extracted from the logging stream if you want to compose tests recording the as-is state of the instrument.
For example, log commands and responses at high level (`Instrument`? `fget/fset`?) with `CMND`/`RESP`, and writes/reads at adapter level (as close to the wire as we can) with `SEND`/`RECV`. TBD at which point of message processing to place the high level, maybe we could neatly enable troubleshooting of the various processing our properties do.
```
CMND MODE?
SEND CH1:MODE?\n
RECV CH1:MODE 3\r\n
RESP current_control
```

bilderbuchi, 2022-05-23:
> That way you can write, if desired, tests with strings from the manual (if it includes term-chars),

I agree, being able to (nearly) copy-paste examples from the manual is important to write tests easily. As you say, though, if the examples include termchars or not is far from given/homogeneous, and I think if we document this well we can assume that our contributors are able to strip them from the examples when writing tests.

bilderbuchi, 2022-05-23:
> The bytes object approach offers the possibility to read single bytes or groups of bytes (as my TC038D does), so that it might be a good idea to follow IK in that respect as well.

Yeah, we'll also need a good story for how to deal with binary or frame-based communication, which is why I already planned to have tests for that. :-) I'll have to study that a bit more.
Sure, the binary stream approach is quite elegant, and also the possibility for the user to supply replies on-demand! I don't know yet what I think about converting to bytes _that early_, I find it easier to debug and reason about code when dealing with strings compared to raw bytes.

bmoneke, 2022-05-23:
I think, it should be possible to keep the list of message pairs.
A send or read method loads the first message pair in the test buffer, if it is empty. Once the written and read bytes are equal to the buffer, the test buffer is cleared, such that in the next round, the next buffer is loaded, repeating the same procedure.
That allows even for several sent or read messages while the other one is just a single one (e.g. `((1, 4), (4), (7))`, or `(1,4), (None, 4), (None, 7))`). The only problem would be, if both read and writes happen without any structure.

bmoneke, 2022-05-24:
In #628 I made an example implementation of the ProtocolAdapter, which does most of the checks. Expected protocol checks, whether all expected communication has been done (no message pairs left, nothing left in the buffers).

As an example I used the tests for the hcp TC038D (for which I had to add read_bytes to the visaAdapter).

Regarding communication written to stdout / read from stdin, we could make another Adapter (most simple) or incoroporate it here. At IK they had not used that function in a long time. I think it is better to propose writing tests as a better alternative to console testing, as the written tests stay around, while the console communication during development is lost afterwards.

bilderbuchi, 2022-05-24:
> Regarding communication written to stdout / read from stdin, we could make another Adapter (most simple) or incoroporate it here. At IK they had not used that function in a long time. I think it is [...]

Good point, agreed!
(I plan to review #628 soon -- tomorrow evening?)

bmoneke, 2022-06-02:
I looked into pyvisa's logging: pyvisa logs whatever it's backend does, not what it does itself. Therefore the logged commands differ depending on the used backend (pyvisa-py vs. NI).
Therefore we cannot rely on pyvisa-logging in order to get the real communication (unless someone wants to write log parser for every backend...).

I guess we're back at logging at Adapterlevel.

bilderbuchi, 2022-06-02:
> I guess we're back at logging at Adapterlevel.

Thanks for checking! I agree, that's probably the best we can do.

PR #634 (2022-08-02): Add feature: Protocol tests without hardware present
This will be useful to test pymeasure instrument implementations without having the hardware present (see #627).

@bmoneke I have now gone through my notes for now and opened this PR so we can iterate further and knock out the remaining `TODO` items together. If you're up to it we can to PRs against this branch again, I think this worked well for collaborative design/implementation.

* [x] ~~What I still have noted down to do is "Debug/investigate and comment on what ProtocolAdapter.read() is doing with those indices" because it was not clea [... cut]
closed 2022-08-02 by bilderbuchi