pymeasure/pymeasure #632: Command line option to pass resource address when running tests that require an instrument

bleykauf, 2022-06-02:
While working on some tests for a new instrument (https://github.com/pymeasure/pymeasure/pull/582) I did not find an easy way to pass resource addresses to pytest. Other instruments with tests use a `RESOURCE`  constant at the top of the test file but this will of course only work for one specific device.

I looked into this a little bit I suggest adding a custom marker (see https://jwodder.github.io/kbits/posts/pytest-mark-off/) by adding to `conftest.py`:

```python
def pytest_addoption(parser):
    parser.addoption(
        "--resource-name",
        action="store",
        default=None,
        dest="resource_name",
        help="""Pass a resource name for an instrument needed for a test.
                See also --instrument-present."""
    )

def pytest_configure(config):
    config.addinivalue_line(
        "markers", """needs_instrument:
                        mark test that needs an instrument to be present."""
    )

def pytest_collection_modifyitems(config, items):
    if not config.getoption("--resource-name"):
        skipper = pytest.mark.skip(
            reason="Only run when a resource name is provided via --resource-name."
            )
        for item in items:
            if "needs_instrument" in item.keywords:
                item.add_marker(skipper)
```

Tests that require an instrument can then be marked by `@pytest.mark.needs_instrument` and will be skipped by default. If the resource address is passed via the command line option `--resource-name` the tests will run.

Within the tests, the instrument can be implemented as a fixture, for example

```python
@pytest.fixture(scope="session")
def hmp4040(pytestconfig):
    """
    Return a HMP4040 instrument. The resource name can be specified via the
    --resource-name command line option.
    """
    resource_name = pytestconfig.getoption("resource_name")
    hmp4040 = HMP4040(resource_name)
    hmp4040.reset()
    return hmp4040
```

I think this would encourage contributors to write these kinds of tests and would make it easier to others to actually run them (without changing the code).

If you like the idea I could start a PR with these changes.

msmttchr, 2022-06-02:
At first glance, it looks very nice idea. I have a doubt, `pymeasure` has a large set of instruments and potentially a test for each instrument, so when you pass the option `--resource-name` to identify the instrument, how do you know to which test this instrument is applicable?

bleykauf, 2022-06-02:
Good point. I did only run the testfile directly, `pytest tests/instruments/.../test_x.py`. I guess that would be the most common use case unless somebody has several supported instruments connected to a computer and wants to run a bunch of tests at once. 

Maybe there is a way to raise an exception in the case that `--resource-name` is passed but no test file is explicitly specified.

bilderbuchi, 2022-06-02:
We currently have a branch in progress (see #627) that add protocol tests without a device, so there will be a much nicer way to test stuff! 
PR will be up soon, progress added peacemeal, I currently have a kitchen renovation ongoing that means June will be busy.

bmoneke, 2022-06-02:
Good luck with your kitchen.

@bleykauf great that you want to write tests while writing the driver itself. 
As the testing environment is currently under development, you may look into it and give feedback from your perspective. The current state is in the branch ["protocol-tests"](https://github.com/pymeasure/pymeasure/tree/protocol-tests).

bmoneke, 2022-06-02:
Testing with the device can be handy, you're right.

But it is difficult to assure, that the responses are, what you expect.

I think "real device" tests should be in their own file, marked to be skipped, only if explicitly called.

The parser code could be in the docs, such that everybody may copy it for their "real device" tests.

bleykauf, 2022-06-03:
Implemented the proposed changes and a dummy test file in this branch in case somebody wants to try it out: https://github.com/bleykauf/pymeasure/tree/dev/instrument-tests

@bmoneke 
> I think "real device" tests should be in their own file, marked to be skipped, only if explicitly called.

I don't think the proposed changes contradict this: they could still be in a separate file, would automatically be skipped if the resource-name is not passes. But there is no need to change any files.

bmoneke, 2022-06-03:
Renaming the file "instrument_testing.py" or "Testing_instrument.py" already prevents auto-detection by pytest, but a manual invocation is still possible. That way we don't need any extra code, which is imported but not executed. That way, the special marker could be dropped, as the user has to explicitly call the test file.

The idea of the parser itself is good, such that you can tell the test, how to connect to the device.

How would you use the parsed resource_name? In this manner?
`instr = Instrument(adapter=resource_name)`

bleykauf, 2022-06-03:
Ah, I see. Did not know that there was such a mechanism, where is this implemented?

Yes, that is pretty much how I would use it, except for in a fixture, see the origin post for an example.

bilderbuchi, 2022-06-03:
> Did not know that there was such a mechanism, where is this implemented?

https://docs.pytest.org/en/7.1.x/explanation/goodpractices.html#conventions-for-python-test-discovery

bmoneke, 2022-12-05:
Seeing, that several pull requests (the latest #773) with changes to device tests, change the address as well. This command line parameter seems a good idea.
Do I understand correctly, @bleykauf , that the following code in `conftest.py` would be sufficient
~~~python
def pytest_addoption(parser):
    parser.addoption(
        "--resource-name",
        action="store",
        default=None,
        dest="resource_name",
        help="""Pass a resource name for an instrument needed for a test.
                See also --instrument-present."""
    )
~~~
such that you can get the `resource_name` (or `adapter`) with: `resource_name = pytestconfig.getoption("resource_name")`?

bmoneke, 2023-07-19:
Closed with #789

closed 2023-07-19 by bmoneke