pymeasure/pymeasure #636: timeout error using Rohde&Schwarz HMP4040 to control HMP2030

RaykHoefemann, 2022-06-09:
Hello,
I tried the beta of Rohde&Schwarz HMP4040 to control my Rohde&Schwarz HMP2030. 

`from pymeasure.instruments.rohdeschwarz import HMP4040
instr = HMP4040("TCPIP0::172.27.0.48::5025::SOCKET", read_termination = '\n', write_termination = '\n', timeout = 5000 )
print(instr.id)
instr.control='LOC'
instr.shutdown()
`
This very short code works only 1 times. After that the code produces only timeout errors. Output console:
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62

Traceback (most recent call last):
  File "C:\Users\hfm\Documents\PYTHON\Demos\pymeasure_RohdeSchwarz_HMP2030.py", line 11, in <module>
    print(instr.id)
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pymeasure\instruments\instrument.py", line 209, in __getattribute__
    return super().__getattribute__(name)
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pymeasure\instruments\instrument.py", line 242, in id
    return self.ask("*IDN?").strip()
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pymeasure\instruments\instrument.py", line 253, in ask
    return self.adapter.ask(command)
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pymeasure\adapters\visa.py", line 142, in ask
    return self.connection.query(command)
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\hfm\Anaconda3\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

The same functionality I tested with pyvisa:
`
import pyvisa
rm=pyvisa.ResourceManager()
instr = rm.open_resource("TCPIP0::172.27.0.48::5025::SOCKET", read_termination = '\n', write_termination = '\n', timeout = 5000 )
print(instr.query('*IDN?'))
instr.write='SYST:LOC'
instr.close()
`
This works very well without any issues. Output console:
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62

Any ideas?

RaykHoefemann, 2022-06-09:
same issue if I replace some functions:
`from pymeasure.instruments.rohdeschwarz import HMP4040
instr = HMP4040("TCPIP0::172.27.0.48::5025::SOCKET", read_termination = '\n', write_termination = '\n', timeout = 5000 )
print(instr.ask('*IDN?'))
instr.write('SYST:LOC')
instr.shutdown()
`

RaykHoefemann, 2022-06-09:
After some testings and studying the pymeasure code I found the solution. Pymeasure doesn't close the handle to pyvisa and this cause in a connection problem with the HMP2030. If I add the following command at the end of my code, then all repeated runs are OK:
`instr.adapter.connection.close()`

Output console:
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62
ROHDE&SCHWARZ,HMP2030,101545,HW50010003/SW2.62

RaykHoefemann, 2022-06-09:
I run this code only several times in a row in the IDE, without restart spyder. The code works only again if I restarted the instrument...restarting IDE I didn't tested.

Other instruments like Keysight 34465A or Yokogawa GS200 didn't show this issue...so I think it's a special problem in combination with the HMP2030.

RaykHoefemann, 2022-06-09:
I guess you are right that python isn't restarted because only during testing some hacks I got also a message like "restarting python interpreter". But again, with the Keysight 34465A and the Yokogawa GS200 I didn't get this issue and all 3 instruments connected via ethernet to the computer.

I'll check this next week.

bilderbuchi, 2022-06-10:
Why reconnecting on an existing connection is not a problem for these other instruments is not clear to me. Maybe you can reproduce that with pure pyvisa as you did above, only _without_ calling close(). If yes, maybe ask at pyvisa. If no, there must be something different within pymeasure. 
Be aware though that this is something you should not do in the first place! You can either use a new python interpreter, del the instrument object explicitly at the end, close the connection explicitly as you did, or put some logic in the beginning that does not recreate the instrument instance should it already exist.

bilderbuchi, 2022-06-10:
I'll close this as a duplicate of the issue about shutdown(), as I think your usage question is diagnosed and answered. We can easily reopen if needed. Mind that for usage questions, the github Discussions forum here is a better place than the issue tracker. 😀

closed 2022-06-10 by bilderbuchi