pymeasure/pymeasure #657: ManagedWindow not working

RahulVedula, 2022-07-27:
Hi, im currently working on a code that takes in data from a Keithley 2400. However, while using the ManagedWindows and Plotter packages, I kept getting the error: 

`**_pyqtgraph requires Qt version >=5.12 (Mine is 5.9.7)._** 
`
However, I'm wasn't able to find a way to update the Qt version, so I tried downloading pyqtgraph, now im getting the error: 
cannot import name 'exec_' from 'pyqtgraph.Qt'

For reference, I'm using a windows operating system and Spyder as my IDE.

neal-kepler, 2022-07-27:
I don't know what the official position is on this, but the recipe that we use is:
``
pip install pyqt5==5.12.3
``
``
pip install pyqtgraph==0.11.1
``

RahulVedula, 2022-07-28:
Thanks for the help. It seems like I've gotten past the version issues, but now im getting the error:
the PyQt5.QtCore module failed to register with the sip module.

Im assuming it may be related to the error I got while updating pyqt5. 

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
spyder 5.1.5 requires pyqtwebengine<5.13, which is not installed.

Sorry for this, but does anyone know what steps I should take from here? Thanks

LongnoseRob, 2022-07-28:
You do not mention which python distribution you are using  under Windows.

You could try [WinPuython](https://winpython.github.io/) they usually have a good base on which one  can easily add `pyvisa & pymeasure`

and it comes with spyder 5.4..

RahulVedula, 2022-07-28:
Hi, sorry for excluding this crucial detail, but I'm currently using Anaconda as my python distribution. Does changing my python distribution make a difference?

LongnoseRob, 2022-07-28:
I dont know much about Anaconda, but it could be easier to install an up-to-date distribution than updating the existing one.

RahulVedula, 2022-07-28:
The error still stays: `the PyQt5.QtCore module failed to register with the sip module.`

Don't know what to do anymore.

msmttchr, 2022-07-30:
I wonder whether the basic GUI example in https://github.com/pymeasure/pymeasure/blob/7a07a85dcc8f7d1782919fdc556d996066f6187c/examples/Basic/gui.py works for you.
I am not familiar with anaconda, but it's supported from what I understand. Could you please check?
Finally, could you report on which OS you are running?

bilderbuchi, 2022-08-03:
I can help with Anaconda (less with Spyder as I am not using that). 

It seems from your error messages that a version that is compatible with both spyder and pyqtgraph seems to be 5.12.

If you can work in Spyder using a separate environment, that would be easiest, as you could isolate your pymeasure packages from the packages spyder needs.

If you don't know what that means, try first upgrading Spyder (the current version is 5.3.2) `conda update spyder`. 
After that, or if that fails, please show the output of `conda list` .. I suspect you have a mix of packages installed with pip and with conda, which sometimes leads to problems.

bilderbuchi, 2022-08-03:
Also, I think this is not a pymeasure problem, so I'll convert this into a discussion.