pymeasure/pymeasure #662: DAQmx is not an Instrument

bilderbuchi, 2022-08-03:
The DAQmx [class](https://github.com/pymeasure/pymeasure/blob/8086835d8a211268a26451c28a054770b3907d40/pymeasure/instruments/ni/daqmx.py#L60) is not an Instrument subclass, but a plain `class`.
We should investigate why that is, and subclass from Instrument if possible, as all our instruments are/should be `Instrument`s.

NitramC, 2023-12-14:
@bilderbuchi how do you feel about also refactoring the code to take advantage of the NI-authored package NI-DAQmx?
https://nidaqmx-python.readthedocs.io/en/latest/

bilderbuchi, 2023-12-14:
I'm not particularly keen on adding yet another dependency, but the whole `ni` folder is a mixed bag of different solutions/dependencies that could do with some cleanup and straightening out and fitting into our `Instrument` hierarchy. I don't have the time to guide a redesign, though. I don't think many people use the `ni` stuff, tbh.

NitramC, 2023-12-14:
Understand you perspective on all points. A few thoughts from my side:

1) I suppose ctypes is already a dependency, but it would take a lot of effort to re-wrap the NI-DAQmx C API.
2) I will make a pull request that offers one possible way to have the DAQmx class inherit from the Instrument class and use an Adapter to connect to the device.
3) A lack of a well developed NI DAQ driver was one reason I initially used the Instrumental package. IMO, the more universal PyMeasure is, the more collective contributions it will gain, and I think contributions grow nonlinearly!

BenediktBurger, 2023-12-15:
A few comments from my side:

First regarding our current code base:
- The `VirtualBench_Direct` class also needs another python package `pyvirtualbench`, which is not a general pymeasure dependency. It is a large driver
- The `NIDAQ` instrument is already an `Instrument` subclass, but requires the `instrumental` package. It is very short driver
- The `DAQmx` instrument requires `ctypes` package **and** the DAQmx driver installed locally. It is a short driver

Now regarding the NI drivers in pymeasure:
- pymeasure's drivers use all bytes communication (or strings etc.) and are independent of manufacturer drivers, except for the NI ones (see above)
- The pymeasure `Instrument` with its property creators is an aid to facilitate that bytes/string communication
- Wrappers for manufacturer drivers do not profit from that property creators and are difficult to test

Consequently I propose to deprecate the NI drivers in pymeasure (maybe hinting at `nidaqmx` package) as they do not fit into the overall picture.
Except for the `VirtualBench`, which is a large (over 1000 lines) driver, the loss is very little.

BenediktBurger, 2023-12-30:
The NI drivers seem to make problems with Python 3.12, see #1008

BenediktBurger, 2024-02-21:
> Consequently I propose to deprecate the NI drivers in pymeasure (maybe hinting at nidaqmx package) as they do not fit into the overall picture.
Except for the VirtualBench, which is a large (over 1000 lines) driver, the loss is very little.

What do you think about deprecating the NI _instruments_ @bilderbuchi , @msmttchr, @CasperSchippers ?

bilderbuchi, 2024-02-21:
I've no opinion on this. Sad to see functionality go away, but if it cannot be made to fit in then why not. I think we should try to gauge how much usage these drivers see with pymeasure, i.e. how many users would be impacted by deprecating/removing it?

BenediktBurger, 2024-02-21:
We could move the ni drivers to their own package, as they do not use any functionality of pymeasure.

It's strange: They use different external dependencies, which pymeasure does not use, but they do not use pymeasure itself

We could keep them where they are right now, just catching the error (such that tests succeed) and adding notes in the documentation.

CasperSchippers, 2024-02-21:
I also do not have a strong opinion on this.

If they do not fit into PyMeasure, I understand deprecating them. Your suggestion to move it to a separate package (maybe marked unmaintained) might be okay indeed.