pymeasure/pymeasure #666: Extend instrument's property factories to unitful quantities.

bmoneke, 2022-08-05:
Instrument's property factories do not take into account unitful quantities (pint).

`set_process` and `get_process` can be used as a stop-gap measure, but a proper implementation would be nice.
For example validators cannot work with quantities yet.

bilderbuchi, 2022-08-05:
Most relevant notes from #248:
> We should first find a good, working design before implementing it. Some thoughts about potential approaches that I had in the back of my head were to
> 
>     * Add a `unit` attribute to the property factories to indicate which unit the property is using. Now that we have dynamic properties, we can even change that at runtime! Very useful e.g. when you can change the unit in the instrument.
> 
>     * Have a global switch (`pymeasure.USE_UNITFUL_QUANTITIES`?) to disable the behaviour by default (at least in a transition period), or find another good way to on-ramp users while eliminating/limiting breakage of existing code.
> 
>     * If `USE_UNITFUL_QUANTITIES` is True _and_ a property has a `unit` defined, interaction (get/set) happens with pint Quantities.
> 
>     * I think it's possible for properties to always _accept_ float, and then assume they are in the property's `unit`.
> 
> 
> > I think it is good to be unit aware, but would not enforce it (but recommend it) for new instruments.
> 
> Agreed!

> Having an option to deactivate units, might be good, because someone might not want to use units (simple program). Maybe a missing pint library switches to that mode? It's only necessary to change the getter, as the setter is always able to handle numbers (see answer below)
> 
> > I think it's possible for properties to always accept float, and then assume they are in the property's unit.
> 
> Yes, that should be done. I already improved IK's `assume_units` method, which takes the input (number or string or quantity) and returns a quantity, to be used by the driver.

bmoneke, 2023-07-11:
In #929 an instrument with unitful quantities is added, which should be revisited after this issue is resolved.