pymeasure/pymeasure #667: check_errors usage inconsistent

bilderbuchi, 2022-08-06:
Our `Instrument.check_errors()` [implementation](https://github.com/pymeasure/pymeasure/blob/ebf6df97e229c0e916192a495ade65053b234ac3/pymeasure/instruments/instrument.py#L514) returns a list of received errors (as its docstring says).

Interestingly though, where we use them, e.g. in control's fget, https://github.com/pymeasure/pymeasure/blob/ebf6df97e229c0e916192a495ade65053b234ac3/pymeasure/instruments/instrument.py#L361, we simply discard those return values without doing anything!

I think where we use it we should do something with the errors, most probably log them.

msmttchr, 2022-11-02:
Actually default `check_errors` provided in `PyMeasure` logs errors. If you agree, this issue can be closed.

msmttchr, 2022-11-10:
I am closing this as per explanation above, @bilderbuchi feel free to reopen if you disagree

bilderbuchi, 2022-11-10:
No, thanks, it's fine!

bilderbuchi, 2023-01-18:
I think we should reopen this, even if only to adjust the docstring of `Instrument.check_errors` -- as it is now, the usage does not reflect the docstring, return values are not used (some instrument docstring examples, print them, though!).

We should probably 
* review usage in the codebase
* consider changing the method to return nothing (as we don't use the return values in the core, right now)
* adjust/correct the docstring to advise the correct behaviour (logging errors). In overridden methods we don't have influence on what they will do, and implementors will need guidance on that.

bmoneke, 2023-04-18:
In #883 the question arose, how to handle errors.

`check_get/set_errors` could
- return a single (maybe aggregated from several errors) error
- return a list of errors
- raise an error.

My proposal:
`check_set_errors` shall raise an error (which is **not** caught by the property, therefore shall be caught in the application) for critical errors. This is at the discretion of the instrument implementor.
For less critical errors it shall return a single error or list of errors (TBD), and the property shall create a corresponding log entry.

bilderbuchi, 2023-04-28:
Regarding logging: 
I think we should log as close to the source as possible, so in the default `check_errors`. 
This does not have to be the only way to "handle" an error, so I don't think "log and don't return anything" is a good approach. For example, a user of `check_errors` might want to do a retry or something when an error occurs (e.g. in an instrument method) -- we should enable such a workflow without having to intercept/parse logging streams to find out _if_ an error was encountered.

Note that it's still possible to dump/ignore those returned values, as we do in the properties currently.
Note also that even a property does not know its "name", so doing the logging in the fget/fset instead of check_errors does not bring the benefit of more information in that respect (or am I missing something?)

bmoneke, 2023-04-28:
> Note also that even a property does not know its "name", so doing the logging in the fget/fset instead of check_errors does not bring the benefit of more information in that respect (or am I missing something?)

Yes, it does not know its name, but it knows the command sent to the instrument, which may lead to the right property name.

Regarding my proposal above:
- I specified the return value of the **abstract** `check_set_errors`: List of non-critical errors.
- It may raise an exception for certain errors.
- What errors raise an exception and what not, depends on the implementation of `check_set_errors` chosen by the implementor (or overriden by the user).
- Properties do not do anything, if a list of errors is returned (as it is currently done)
- If `check_set_errors` raises an Exception, the properties do not catch the exception, such that a user might handle it.

This allows a user to choose, which errors stop the properties from working (raising an exception) and which not.

Now the question is, what our default `check_set_errors` does:
For backward compatibility (no behaviour changes), they should return a list.

>  I don't think "log and don't return anything" is a good approach

What should the properties do, in your opinion, if they encounter an error?

bilderbuchi, 2023-04-28:
> Yes, it does not know its name, but it knows the command sent to the instrument, which may lead to the right property name.

Yeah, that's a good idea to leave more "breadcrumbs". In that light, I agree it might be useful if the properties log errors.

bilderbuchi, 2023-04-28:
_(the following is a bit train-of-thought style, I don't have well-ordered thoughts on this apparently)_

> > I don't think "log and don't return anything" is a good approach
> 
> What should the properties do, in your opinion, if they encounter an error?

Let me clarify: I meant specifically the error checking functions.

I'd prefer if there is API homogeneity between `check_errors`, `check_get_errors`, `check_set_errors` (at least the defaults that we provide), i.e. that they all return lists of errors. We don't control what people implement/override, but we can (and should) give guidance via a defined interface.
The _callers_ should "properly" deal any received errors. Callers will probably mostly be properties' fget/fset, and instrument methods. 

What the properties should do is a good question. I don't have an obviously-preferable answer right now :shrug: I guess that what we do right now (only log) is good enough.

Re Exceptions: 
I'm not clear on: are you proposing that we _recommend_ instruments override check_errors to define a set of errors that raise instead of just logging/returning? And if that does not happen, we don't raise for errors? Or that we just _allow_ that by defining it in the docstring (`raises: XXX`) What kind of Exception do you want to raise? A custom generic `InstrumentError` or somesuch?

bmoneke, 2023-04-28:
> Re Exceptions:
> I'm not clear on: are you proposing that we recommend instruments override check_errors to define a set of errors that raise instead of just logging/returning? And if that does not [...]

I thought about **allowing** it, hinting at the possibility to use it that way.

> What kind of Exception do you want to raise? A custom generic InstrumentError or somesuch?

That should be open to the developer, I'd say, as it depends on the error itself (wrong value or wrong type or connection etc.).

bilderbuchi, 2023-04-28:
What do you think about a slight duplication of logging?

We could retain the current behaviour of default `check_errors`, i.e. 
* log errors
* return list of errors

`check_get/set_errors` document that they (if overridden) shall
* log received errors
* return a list of encountered errors
* Optionally _may_ raise an Exception if a critical error is encountered

Property fget/fset, if they receive an error during checking, log that error + useful context (command). So we might get
> ..error: KeithleyXYZ:  235, baz is borked (from check_errors)
> ..error: Error received after writing "POW 1": "235, baz is borked" (from within property fget/fset)

This way, we ensure that
* we get logging on error by default, even if the error does not happen within a property
* we get additional useful context if this happens in a property
* we document that an instrument may raise an Exception if deemed appropriate

Thoughts?

bilderbuchi, 2023-04-28:
One thing I'm not sure on: What kind of error instruments emit, and if they (all? some?) make the returned values meaningless?
Out of range should be intercepted by validators already, connection errors will probably be raised by the Adapters. What else occurs "typically" (if that can be said)?

Because, if we let check_x_errors raise an Exception, that means that any further processing (within the property) does not happen anymore, and the user does not get their values.

bmoneke, 2023-05-08:
Trying to fix an instrument, I found a different path, which reduces the number of logging events, but breaks backward compatibility:
- the `check_errors` methods (all three) accept a "context" parameter, which will be included in the log message / exception message.
- `control`'s getter and setter method give the command (and whether a getter or setter) as context to the `check_errors` method, which does the logging, including this vital information.

Advantage of this approach is, that you can include the failed command into an exception message, if `check_get/set_errors` raises an exception. Without that information, the Exception itself does not offer the command, which caused the error.

bmoneke, 2023-05-08:
Regarding breaking backward compatibility: Only if someone had implemented its own `check_errors` method **and** used `check_set/get_errors` in a property, the instrument will break.

We could mitigate this, if `control` does the following:
~~~python
if check_get/set_errors:
    try:
        self.check_get/set_errors(command_sent)
    except TypeError:
        self.check_get/set_errors()
        warn("Deprecated to call check_get/set_errors without a `context` parameter", FutureWarning)
~~~

P.S: I pushed this suggestion (without that try except clause).

bilderbuchi, 2023-05-13:
I have to admit I'm not too keen on forwarding this `context`, and breaking backwards compat for this. I'd prefer two separate logging messages, especially as they're coming from separate "layers" in the call stack. However, I recognize the use in adding some state/context to a stateless function, so I think I'm hesitantly OK with this.

Similar feeling with adding `context` to `read`, too -- should not the error checkers that _call_ `read` add that context if read discovers and error? It's the [responsibility](https://en.wikipedia.org/wiki/Single-responsibility_principle) of the error checkers to react appropriately to errors, not the `read` method's - that one should just, well, _read_. This should be refactored if possible, imo.

In general: I think we have to be careful (in general) with passing "state"/context objects through our call layers, this will bloat our signatures and more closely entwine our methods. 
However, it's only for a couple instruments, so the blast radius is small. I'd rather not encourage this as a general pattern, though.

bilderbuchi, 2023-05-13:
The proposal to handle backwards compat (which turns the backwards break into a deprecation) looks sound to me.

bilderbuchi, 2023-05-13:
The idea with the SRP is that we separate our building blocks into isolated parts as much as feasible, so that changes for refactorings are needed in few(er) places. On the other hand, if many methods "know" about many things, and our call stacks are deep, there are many places that we need to change when restructuring/improving -- they are "entwined" a lot, which makes changing things cumbersome/effortful.

bmoneke, 2023-05-15:
I see the problem with the connections between the methods. (btw, what do you mean with "SRP"?)

So, do I understand, that we have two viable courses of action:
1. no context at all: less entanglement, but `check_errors` cannot tell which property failed to get/set
2. context for `check_errors` only: reasonable entanglement
3. ~context for `check_errors` **and** `read`~: scrapped due to reasons mentioned above (`read` should just read...)

I'm unsure, whether the entanglement of option 2 is worth the additional information. On the other hand, if your program dies with the message "some property could not be set", and you do not know, which property, you might be frustrated. (in REPL you see the last command, but a program running happily in the background might not.)

bilderbuchi, 2023-05-16:
> (btw, what do you mean with "SRP"?)

Sorry, I was too terse; the Single Responsibility Principle I linked above.

> EDIT: Another idea: The control catches exceptions raised in the check_errors methods, log them and reraises them. That way we can add the context to the log, but do not have to give context for every method call.
I implement this last one.

:+1:

bmoneke, 2023-05-18:
Done in #883

closed 2023-05-18 by bmoneke