pymeasure/pymeasure #672: Define a deprecation policy

bilderbuchi, 2022-08-07:
We should come up with and document a general deprecation policy/timeline, to set expectations for feature deprecations/removal.
When we deprecate features, we should define a reasonable timeframe (wall time or number of versions) for how long deprecation warnings stay in.
For all deprecated features, we need to (should) keep the old code in place, with a `FutureWarning` (preferred as it's not filtered out by default) or a `DeprecationWarning`.

Currently, we should remove all deprecated features before releasing v1.0. 
Later, we will follow semver and need a major release for backwards-incompatible changes.

bmoneke, 2022-11-14:
@bilderbuchi proposed a "@deprecated" decorator in #717 .

I have some ideas:
That decorator could contain to parameters: Version and the warning message. It could also add the deprecation note (with the warning message as content) to the docstring. How to include it in Sphinx, I do not know.

bmoneke, 2023-01-03:
A deprecation warning should indicate, what is deprecated and how to fix it.

bmoneke, 2023-01-05:
Some example deprecation code:
~~~python
from warnings import warn

def deprecated_feature():
    """Do something deprecated.

    .. deprecated:: {next_version}
        Use {replacement} instead.
    """
    warn("{feature} is deprecated, use {replacement} instead", FutureWarning)
~~~

- The Sphinx documentation should contain the next version (from which on it will be deprecated) and should indicate what to do instead. As it is in the documentation of the deprecated feature, the feature does not have to be named.
- Raise a warning with **FutureWarning** as DeprecationWarnings are ignored unless debug options are activated. The warning should contain the deprecated feature as you receive the warning without knowing where it originates from.

bmoneke, 2023-01-09:
For new Python features (or software version etc.), which can not yet be used, due to support for older python versions,
it is good to mention the last not supported python version.
For example feature X is introduced in Python 3.8, therefore it should be put into a comment `works for python>3.7` or similar, such that it can be found, when python support 3.7 is dropped. See also #673 .

bilderbuchi, 2023-01-18:
Researching available libraries, https://pypi.org/project/Deprecated/ seems to be a good candidate for an already implemented decorator. It is [possible](https://deprecated.readthedocs.io/en/latest/tutorial.html#deprecation-warning-classes) to change it to emit `FutureWarning` (as we want). It can also automatically add a Sphinx note to the docstring.
TBC how we best decorate attributes/properties with this.

BenediktBurger, 2024-01-22:
In upcoming python 3.13, there is a `deprecated` decorator: https://docs.python.org/3.13/library/warnings.html#warnings.deprecated which might serve as an inspiration.

bilderbuchi, 2024-02-01:
Note https://peps.python.org/pep-0702/#backwards-compatibility: 
> [...] the @deprecated decorator will be added to the typing_extensions module, enabling its use in older versions of Python.

So it seems we could use this even in older python versions.