pymeasure/pymeasure #673: Drop Python 3.7 support

bilderbuchi, 2022-08-07:
When Python 3.7 support ends in June 2023, we will drop support, too.
At this point, we can also start using Python 3.8 features. Please collect any changes we should make in this issue.
* [ ] use the `@final` decorator for a stronger hint than a docstring that some methods should not be overridden: https://stackoverflow.com/a/58939735/599884 (Some candidates will probably be in Adapter)

bmoneke, 2022-08-08:
I got something else for Python>3.8 (not yet for Python3.8!).
Maybe we can mark such parts with "# Python>versionToDrop code", e.g. "# Python>3.7 @final", or in my case "# Python>3.8 str.remove_suffix()".
That way you can search the codebase for this string and find all the places to change.

bilderbuchi, 2022-08-08:
Agreed, I have also been doing that in the past.

BenediktBurger, 2024-10-25:
Python 3.7 support is already removed.

closed 2024-10-25 by BenediktBurger