pymeasure/pymeasure #675: PyQt6 Compatibility

bicarlsen, 2022-08-10:
Are there plans to update the package for PyQt6 / PySide6?

bilderbuchi, 2022-08-10:
Not at this time. I _think_ a pull request would be welcome, but @CasperSchippers is the point person for GUI-related issues.

Minor point: We'd need to confirm that releases are available for Python 3.7 as we still support that, but I'd think that's a given. :-)

I don't know if an upgrade puts any users in a bind because they can't update their Qt version because of version constraints of other installed packages. Would be good to find out if nowadays a codebase compatible with both 5 and 6 is feasible.

CasperSchippers, 2022-08-26:
In my opinion, a PR for this would indeed be welcome
.
I tried quickly and it seems to me that the change would be rather simple: we rely on pyqtgraph for handling the import of Qt; they already support importing pyqt6, so on that side there is nothing that we have to change in our import section.

As to the rest of the code; it seems to me at this moment that the most of the changes come down the changing the shorthand name of the enum-flags by the full name (i.e. changing `QtCore.Qt.AlignRight` into `QtCore.Qt.AlignmentFlag.AlignRight`); if that is indeed everything, I think there is no issues in keeping the code compatible with both pyqt5 and pyqt6. But then again, chances are there will be something else that can cause some troubles.

msmttchr, 2022-08-26:
@CasperSchippers, would such a PR be also a good chance to add PySide6, in your opinion?

CasperSchippers, 2022-08-26:
That would indeed seem a logical addition to me, provided that the differences between pyqt6 and pyside6 are not too large (what I remember from working with pyside2 and pyqt5, is that if the code is compatible with pyside2, it often also works with pyqt5, so in that expect not much issues).

CasperSchippers, 2022-08-26:
When adding support for pyqt6 (and maybe also pyside6), it might be good to also setup tests (I recall we discussed about this before, b to ensure nothing breaks and is compatible with both pyqt5 and pyqt6 (and possibly pyside2 and pyside6).

bilderbuchi, 2022-09-03:
Yes, pyside compatibility would be great to obtain. if there is too much work necessary to account for the slight api differences, wasn't there this unified compatibility layer by the spyder folks? Qtpy? Is that still maintained/necessary?

CasperSchippers, 2022-09-10:
PyQt seems to be maintained still, so that is an option; I'm not sure if it in necessary though: I just did some quick testing, PySide6 seems to work right out of the box (you get some deprecation warnings, but that's the worst I've seen so far). For PyQt6, a bit more work is required, but it still seems to be manageable.

CasperSchippers, 2022-09-10:
While checking some things, I did most of the legwork already, so I decided to open a PR for this (#668); what I've seen so far is that all of them appear to be compatible* with each other, as long as you make sure that you adjusted the code to Qt6.

\* The only exception is the `exec` method of `QtWidgets.QApplication`: in the new versions `exec()` is recommended (in stead of `exec_()`; some implementations (PySide6 and PyQt5) support both `exec()`  and `exec_()` (with a deprecation warning), but PyQt6 only provided `exec()`, and PySide2 only provides `exec()`. I've solved this now by shifting to `exec()`, and adding `exec()` to `QApplication` whenever it is not present.

closed 2022-10-08 by bilderbuchi