pymeasure/pymeasure #689: resourceName error

dhunganaanish, 2022-09-16:
Traceback (most recent call last):
  File "\\home.mines.edu\home\adhungana\Documents\Python 3.10\test1.py", line 3, in <module>
    x=SR830_kb.device()
TypeError: device.__init__() missing 1 required positional argument: 'resourceName'
[Finished in 4.262s]

neal-kepler, 2022-09-16:
I'm not sure what the ``_kb`` is from in ``SR830`` is and I don't understand this attribute pattern, but it looks you haven't passed a resource identifier to device. This would be something like what is describe in pyvisa's resource name [overview ](https://pyvisa.readthedocs.io/en/1.8/names.html). 

If the instrument is detected in something like NI Max [and you have NI Max installed), you can grab the identifier string there. Otherwise you should be able to find out how to get it from the instruments manual. If you need more help please post the code that imports or defines ``SR830_kb`` because that is not in the main pymeasure repository, it seems.

dhunganaanish, 2022-09-20:
This is the code!
**import time
import numpy as np

from pymeasure.instruments.srs import SR830
lockin = SR830("GPIB0::8::INSTR")

lockin.frequency(130)**

Error as below:

  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pymeasure\instruments\instrument.py", line 209, in __getattribute__
    return super().__getattribute__(name)
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pymeasure\instruments\instrument.py", line 359, in fget
    vals = self.values(command_process(get_command), **kwargs)
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pymeasure\instruments\instrument.py", line 272, in values
    return self.adapter.values(command, **kwargs)
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pymeasure\adapters\adapter.py", line 88, in values
    results = str(self.ask(command)).strip()
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pymeasure\adapters\visa.py", line 142, in ask
    return self.connection.query(command)
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\resources\messagebased.py", line 648, in query
    return self.read()
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 226, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\adhungana\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
[Finished in 10.561s]

bilderbuchi, 2022-09-23:
This seems like a connection problem that needs to be troubleshooted -- it's more appropriate in the discussions area. :-)

OP: can you talk to the instrument in any other way? What does running these commands yield?
```
import pyvisa
rm = pyvisa.ResourceManager()
rm.list_resources()
```