pymeasure/pymeasure #694: Pyqtgraph 0.13 Compatibility problem

maederan201, 2022-09-29:
Hello,

I stumbled upon a compatibility problem with the newest pyqtgraph release - or at least I think it is. To reproduce it I just type
```
from pymeasure.display.windows import ManagedWindow
```
into a python console and the following error is thrown:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/maederan/Documents/00 Repositories/pymeasure/pymeasure/display/__init__.py", line 31, in <module>
    from .plotter import Plotter
  File "/home/maederan/Documents/00 Repositories/pymeasure/pymeasure/display/plotter.py", line 31, in <module>
    from .windows import PlotterWindow
  File "/home/maederan/Documents/00 Repositories/pymeasure/pymeasure/display/windows.py", line 34, in <module>
    from .browser import BrowserItem
  File "/home/maederan/Documents/00 Repositories/pymeasure/pymeasure/display/browser.py", line 36, in <module>
    class BrowserItem(QtGui.QTreeWidgetItem):
AttributeError: module 'pyqtgraph.Qt.QtGui' has no attribute 'QTreeWidgetItem'
```
My PyQt knowledge is too little but shouldnt QTreeWidgetItem be in pyqt5.Qt.QtGui or something like this? 

I can reproduce this bug both in python 3.9.10 and 3.8.12. Here's the relevant pip output of all relevant packages:
```
PyQt5==5.15.7
pyqt5-plugins==5.15.4.2.2
PyQt5-Qt5==5.15.2
PyQt5-sip==12.11.0
pyqt5-tools==5.15.4.3.2
pyqtgraph==0.13.1
```

I can avoid this bug by downgrading to pyqtgraph 0.12.4 on one of the machines where I use the pymeasure version deployed via PyPi. On another machine where I use the source repo (installed via `pip install -e .`) a downgrade does not help, but this might be a different problem.

I hope someone can resolve what's going on. For now I'm using pyqtgraph 0.12.4 on the live machine and its working.

bilderbuchi, 2022-09-29:
So, from a quick check, there seems to have been a [deprecation warning](https://github.com/pyqtgraph/pyqtgraph/commit/bc6c7212a7ed1ebd4c9eab65929c920aecb01407#diff-2ad3fe8652945b78e4f24890e68f21cfdf1288c771d2d019e055d773a6b2daffR316-R317) that we have never encountered, that seems to recommend to use `class BrowserItem(QtWidgets.QTreeWidgetItem):` instead. Can you check if that fixes the problem? You'll also have to add a QTWidgets import [here](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/Qt.py#L27), I think.

maederan201, 2022-09-29:
Indeed, this is the culprit and changing `QtGui` to `QtWidgets` in various class definitions solves the problem. Import is now working, but the examples not yet. I will work on a PR fixing this in all places required. Then people with more Qt-expertise can point out other places where changes are required.

msmttchr, 2022-10-02:
Fixed via #695

closed 2022-10-02 by msmttchr