pymeasure/pymeasure #708: problem with reading prologix adapter

ElaheParham, 2022-10-15:
Hello, i'm using a GPIB prologix adapter with a lock-in amplifier (Stanford research system 830 -  SR830).

i'm trying the following code:

```
prologix = adapter.PrologixAdapter(port = port, address=address)
lockin = instrument.SR830(prologix)
```

and then i tried:
```
prologix.write("OUTR?1")
outPutString = prologix.read()
```

but the output is not a complete string like it should be. for example, i expect to get a value like [b'0.00123\n'] but then i get:
[b'0.']
[b'0']
[b'']
[b'123\n']

It gets only a part of the output every time I write and then read the value. it seems like the baud rate is not 9600 or maybe the termination is not what it should be (\n). do you have any idea how it can be fixed?

bilderbuchi, 2022-10-15:
What pymeasure version are you using? Can you try with current master branch?

bilderbuchi, 2022-10-15:
Wait, are you trying to communicate with the instrument or the adapter? Should that not be 
``` 
lockin.write("OUTR?1")
outPutString = lockin.read()
```

ElaheParham, 2022-10-15:
> What pymeasure version are you using? Can you try with current master branch?

i'm using version 0.10.0 which is the most recent one. i can try with the master branch but as far as i looked, they are the same.

> Wait, are you trying to communicate with the instrument or the adapter? Should that not be
> 
> ```
> lockin.write("OUTR?1")
> outPutString = lockin.read()
> ```

at first, i was communicating with the instrument and it gave me the same output. then i went through the package and i saw the `read` function of the instrument uses the` read` function of the adapter so they are doing the same thing.

bilderbuchi, 2022-10-15:
> i can try with the master branch 

yes, please do.
> as far as i looked, they are the same.

Didn't look far enough, we recently merged a large amount of adapter and testability improvements. ;-)

This is the message protocol we would expect to be exchanged via the prologix adapter. I'm not 100% sure this would catch the issue you're having, though.
```
from pymeasure.adapters import PrologixAdapter
from pymeasure.test import expected_protocol

def test_sr830_with_prologixadapter():
    with expected_protocol(
            PrologixAdapter,
            [("++auto 0", None),
             ("++eoi 1", None),
             ("++eos 2", None),
             ("++addr 5", None),
             ("OUTR?1", None),
             ("++read eoi", "0.00123"),  # what the instrument would return
             ],
            address=5,
    ) as adapter:
        adapter.write("OUTR?1")
        reply = adapter.read()
        assert reply == "0.00123"
``` 

> it seems like the baud rate is not 9600

did you check if the adapter is configured to that?

bilderbuchi, 2022-10-15:
You can also do
```
import logging
logging.basicConfig(level=logging.DEBUG)
```
before you create the adapter -- this should show some info about what's going on in the background.

msmttchr, 2022-10-15:
> > it seems like the baud rate is not 9600
> 
> did you check if the adapter is configured to that?

I think the baudrate parameter is irrelevant since the communication is USB-to-GPIB and there's no real serial port behind it. Prologix presents a virtual serial port as user interface on the PC side.

msmttchr, 2022-10-15:
> but the output is not a complete string like it should be. for example, i expect to get a value like [b'0.00123\n'] but then i get: [b'0.'] [b'0'] [b''] [b'123\n']
> 
> It gets only a part of the output every time I write and then read the value.

I think this may be caused by a timeout too small.
You could try to:
- `print (adapter.connection.timeout)` to know the timeout (in seconds if you're using v0.10)
- `adapter.connection.timeout = <value>` to increase the timeout

BTW, I would expect `read` to return a str object and not a list (in your screen dump, it seems that value is a list, as it enclosed in square brackets)

msmttchr, 2022-10-18:
Possibly defaults (I haven't found them in manual though) Prologix configuration has been changed by using `++savecfg` in the past.
Would it be possible to query `++mode` and `++eot_enable` after power on and report the value? In general, it could be considered to add these settings explicitly on the Prologix `set_defaults`.

bilderbuchi, 2022-10-18:
> Thanks for the hints but i had previously tried several timeouts and baudrates and everything. it didnt work though.

In general, it is easier for us to help you if you give complete and detailed information up front about what you have tried and what the specific (unexpected) results were. Then we don't end up suggesting troubleshooting steps you have already tried. 😉