pymeasure/pymeasure #726: Parameter values not showing up in bottom frame of the GUI

LastStarDust, 2022-10-30:
Hello!
For some reason the value of the parameters is not showing up in the bottom frame.
However the plot is correct and the output CSV file contains all values.
How do I solve this problem? Are there any additional steps to be taken to make the values show up there?

[image]

msmttchr, 2022-10-30:
The parameters you would like to show are defined in the `inputs` parameter of `ManagedWindow`. Is the parameter you want to show in `inputs` parameter ?

LastStarDust, 2022-10-31:
@msmttchr Thank you for the hint.
I noticed that the parameters defined in the `inputs` variable are displayed in the left panel and the parameters defined in the `displays` variable are displayed in the bottom panel, as you suggested.

These parameters are treated and displayed as fixed input so they are not refreshed during the execution. But in my case they are output parameters, so for me the solution was to just remove those parameters altogether from the `inputs` and `displays` variables.

Thank you again.
Giorgio

closed 2022-10-31 by LastStarDust