pymeasure/pymeasure #730: 0.11 release preparation

bilderbuchi, 2022-11-01:
After having already merged #660, I would like to wait to
* [ ] integrate #718 and 
* [ ] integrate #712 

and then cut the next release, 0.11. I think we have a good number of new big features, and it would be good to get feedback on those.

Any other feature issues/PRs you would like to **urgently** see included in the next release? Mind that every additional one will delay the release further. 
Instrument additions will be reviewed continuously (and go out with the release that happens next), but unfortunately that takes a lot of time and effort.

bmoneke, 2022-11-01:
The PRs you mentioned, together with #660 and the protocol tests, are a great improvement, which should be published.

Instruments are not urgent, in my opinion, as those who need an instrument may always copy the instrument file and are ready to go.

CasperSchippers, 2022-11-02:
I think that that when the PRs mentioned by @bilderbuchi are merged 0.11 is good to go.

bilderbuchi, 2022-11-02:
OK then, let's do it like this, thanks for the feedback!

bmoneke, 2022-11-16:
@bilderbuchi suggested to move #718 to another release, in order to have more time to nail down the final implementation (changing the instruments might change the channel implementation).

As we already have enough good content to deploy and both #503 and #701 are merged, I'm for releasing now (without #718).

What do you think @CasperSchippers , @msmttchr ?

msmttchr, 2022-11-16:
Ok for me.

CasperSchippers, 2022-11-16:
Fine with me

bilderbuchi, 2022-11-16:
> I'll branch off a separate branch so that the PR review/merging work can continue unhindered.

I'll use `develop` as that will get CI testing without creating a PR to master (which could hinder PR merges). The tag will then maybe happen on that branch (if we have merged the channel PR to master in the meantime), and merged into master immediately afterwards.

bilderbuchi, 2022-11-19:
It's done, v0.11.0 tagged and released on pypi and conda-forge! :partying_face: 
Thank you everyone for your contributions!

closed 2022-11-19 by bilderbuchi

msmttchr, 2022-11-19:
Congratulations 👏👏👏

CasperSchippers, 2022-11-19:
Nice, thanks, Good job!!

LongnoseRob, 2022-11-19:
_Very nice_, thank you!

LongnoseRob, 2022-11-19:
Github shows me still V0.10 as latest release..
[image]

bilderbuchi, 2022-11-19:
Thanks! I don't know exactly what the deal is with "Releases" vs. Git tags, but I manually did this now.

LongnoseRob, 2022-11-19:
> Thanks! I don't know exactly what the deal is with "Releases" vs. Git tags (annoying), but I manually did this now.

No worries, now it shows v0.11  correctly