pymeasure/pymeasure #733: Make properties return enums

bmoneke, 2022-11-05:
Enumerations (in `enum` module) offer autocompletion for selecting different values (e.g.  Sine or triangle voltage curve) and decoding of a bitwise combined value (e.g. Status byte).

Right now, pymeasure's property factories can accept IntEnum, StrEnum, IntFlag for the setter out of the box, as they are interpreted as Str or int.
The getter, however, cannot natively use an enum to map a return value. You can use `get_process=lambda v: SomeEnum(v)`, but that is cumbersome.

I see different implementations in the getter of control:
1. Leave as is (using get_process). 
2. set the enum as 'values' parameter (we could add an enum validator as well) and add an instance check to the "map_values" routine (right now lists and dictionaries), which is only a small change. This requires to set map_values to True.
4. add an enum parameter which will work like suggestion 2, but without specifying map_values again. Maybe it could do the validator check as well. This requires the most changes to `control`, but gives the smallest invokation (see examples) 

Examples:
```
Status1 = Instrument.measurement("stb?", get_process= lambda v: SomeEnum(v)) 

Status2 = Instrument.measurement("stb?",
    values=SomeEnum, map_values=True, 
    validator=enum_validator) 

Status3 = Instrument.measurement("stb?", enum=SomeEnum) 
```

My favorite is version 2, as it does not introduce a new logic, but uses already known mechanics. Version 3 is the most neat one, but I do not think, that it is worth the new logic with more complexity, especially, as it won't be used that much (and most of the time you have to define several parameters anyway). 

What do you think?

bmoneke, 2022-11-05:
Implementation for version 2 (extract of control getter and the newly inserted lines) :
```
            if len(vals) == 1:
                value = get_process(vals[0])
                if not map_values:
                    return value
                elif isinstance(values, (list, tuple, range)):
                    return values[int(value)]
                elif isinstance(values, dict):
                    for k, v in values.items():
                        if v == value:
                            return k
                    raise KeyError(f"Value {value} not found in mapped values")
# begin new part
                elif isinstance(values, Enum):
                    return values(value)
# end new part
                else:
                    raise ValueError(
                        'Values of type `{}` are not allowed '
                        'for Instrument.control'.format(type(values))
                    )
```

bilderbuchi, 2022-11-05:
I think the values/mapping code is already quite complex and hard to reason about (cf. that one instrument PR that wants to add multi-map capability), I'm hesitant to make that even more complex.
I'd prefer option 1, `get_process=lambda v: SomeEnum(v)` is not that complicated, and it's clear enough.

Should we want to improve on this, without having spent a lot of thought on it, my tentative vote would be for option 3, an `enum=SomeEnum` argument could work both ways (getting and setting), and it's very clear for the user. as you say, it will be tricky to get a good (and clear/understandable) interaction with the map_values and validator options -- maybe we can/have to disallow some argument combinations.

bmoneke, 2022-11-05:
I did not think, that option 1 would get any vote. But now I see, that it has as many parameters as option 1, does not require any code rework and is clear.
That raises option 1 in my eyes.

We still can add an enum validator, for those who want to use it.

bmoneke, 2022-11-06:
Another option is `get_process=SomeEnum`, or maybe in cast parameter. 

That is simple and is going to be tested in #736 to work correctly.

bilderbuchi, 2022-11-18:
Thinking further on this, an enum is *basically* a dictionary/mapping (with some more bells & whistles). This makes option 2 more attractive, as the usage could be equivalent to dict (i.e. supply `values` and `map_values`) -- users will already know how that works, and get the additional benefit of autocomplete and bit flag behaviour (where appropriate).

bilderbuchi, 2022-11-18:
In any case, let us postpone this until after #740 is in, as that touches the same code region and that would get confusing to keep straight.