pymeasure/pymeasure #734: Multiple plot widget

CasperSchippers, 2022-11-05:
A widget with multiple plots can be quite beneficial.

#722 by @mcdo0486 takes care of one implementation, by having multiple, dockable plotwidgets; each of these widgets itself is a full PlotWidget, so there is quite a bit of space-wise overhead. To have a widget with multiple plots without the spatial space-wise overhead, it would be nice to not just lump together multiple plotwidgets, but rather have a pyqtgraph widget with multiple plots; there are a few ways this can be done, by have multiple graphs sharing the same x-axis (e.g. having different left and right y-axes), or having adjoined subplots (e.g. having several full axes, but omitting the ticks on the x-axes of all but the bottom x-axis to save space). Below a few thoughts from the closed PR #723:

_Originally posted by @msmttchr in https://github.com/pymeasure/pymeasure/issues/723#issuecomment-1298394099:_
If you checkout https://github.com/pyqtgraph/pyqtgraph/pull/1359, `pyqtgraph` people are trying to add this feature and hopefully this will converge soon.
My opinion is either:
- we close this PR [#723] in favor of #722 
- we wait for https://github.com/pyqtgraph/pyqtgraph/pull/1359 to be merged and then base this PR on it.
- we base implementation on https://github.com/pyqtgraph/pyqtgraph/blob/master/pyqtgraph/examples/MultiplePlotAxes.py

_Originally posted by @mcdo0486 in https://github.com/pymeasure/pymeasure/issues/723#issuecomment-1299324175:_
From talking with a couple users today, the use of multiple curves on a single graph doesn't have as much value as separate graphs. The measurements are often independent and aren't correlated so sharing curves on one graph isn't as useful as simplicity of multiple graphs. This is just random feedback I got today. If there is interest from the users then I'll definitely checkout the multiaxis plot implementation for a widget, thanks for linking to the pyqtgraph thread.

_Originally posted by @CasperSchippers in https://github.com/pymeasure/pymeasure/pull/723#pullrequestreview-1160034773:_
If, however, it would be possible to really merge the graphs (e.g. reduce/remove the space between the graph, only ticks and labels on the lower x-axis to save space, synchronised x-axes such that zooming/shifting in one axis works for all axes) I would see a tremendous advantage! (The holy grail in this sense I think would be to have the option to add or remove graphs on the go, but this might be (I don't know) extremely difficult).
This might be achievable using the  pyqtgraph.GraphicsLayoutWidget or the pyqtgraph.MultiPlotItem.

bilderbuchi, 2022-11-05:
I had a similar design problem in the past. We had a GUI where we were plotting live time series of instrument parameters with magnitudes ranging from kV to nA, so this was not easy to meaningfully and clearly display in one common plot. We ended up scaling them up/down to a 1-10 magnitude (and live-adapted the legend captions to something like `1e4 V`), it did the job.

However, some improvement I never got around to implementing, that could be relevant here, was the following procedure:
* Scale all quantities to "engineering magnitudes", so in steps of 3 magnitudes (mV, V, kV,...). These are familiar to users already.
* With this, all magnitudes end up between 1 and 1000. This is too much to meaningfully display on one plot.
* The insight (thanks to [this awesome book](https://direct.mit.edu/books/oa-monograph/5345/The-Art-of-Insight-in-Science-and)) was to subdivide this _geometrically_ into _two_ plots from 1-30 and 30-1000, which preserves enough visual detail to be workable. 
* This should yield a dense but readable display for a large number of wildly different values over arbitrary magnitudes.
* Now, what remains is to assign every trace to the "lower" or "upper" plot according to some measure (values in the visible range, recent values, ...)

Sadly, I never got around to fully implementing and testing this, but it might be a good solution here.

goodboy, 2022-11-07:
Isn't it more useful to just have `N`-axes per co-domain and then just compress which are shown depending on the *context* of the current visualization?

Like, even just as example, say you had 100 different measurements (and all 100 are disjoint in terms of co-domain range) that you wanted to show collectively on one "view". Why not just have 100 axes where `100 - N_shown` are hidden (say 4) and then depending on what you want shown you just only show the 4 *relevant* axes to the current comparison - whilst still showing all 100 curves, overlay-ed.

I realize this approach is not so conducive to static pen-and-paper style figures but with computers this kind of thing is suuuper simple. Also, when would you ever need to see all 100 axes at once; it's more likely the contextual comparison is simpler.

bilderbuchi, 2022-11-07:
> you just only show the 4 relevant axes to the current comparison - whilst still showing all 100 curves, overlay-ed.

I'm not sure what you are proposing here? Showing all 100 curves overlaid in a single plot, but stacked y-axes for only 4 of them?

> Also, when would you ever need to see all 100 axes at once; it's more likely the contextual comparison is simpler.

I'm not sure who talked about 100 axes. At least in my case, we were working on electrostatic (spacecraft) thrusters, and the telemetry had somewhere north of 50 parameters overall. During testing, you were typically continually interested in about 10 of them (from nano- to kilo- range), to be able to notice as soon as something goes sideways.
Having 10 axes (if you mean them in the Matplotlib sense, i.e. plotting areas) was not feasible for that.