pymeasure/pymeasure #736: Better way of encapsulating enums

bilderbuchi, 2022-11-06:
When defining enums that belong to a class (e.g. error codes), it would be clean design if the enums are defined inside the class. However, as our property factories run at import time, not all attributes and scopes are set up yet, so I've often run into problems doing this, which is why I recommend to define them at module level, instead.
This is not a concern when they are used inside methods.

@bmoneke suggested to test if we can actually use them as arguments to the property factories, when they are defined inside the class. Does that work for all aspects (plain args, callables like `cast`, lambdas)? If yes, define some tests confirming the behaviour, and document the improved approach. Consider if we should move existing documents.

_Originally posted by @bmoneke in https://github.com/pymeasure/pymeasure/pull/712#discussion_r1014733992_

bmoneke, 2022-11-06:
There is another reason for adding them to class:

Let's say we have a control for the waveform of the generator.
In order to set the control, it would be more easy to set it with `inst.waveform = inst.Waveforms.SINE` instead of importing the different enums from the instrument file itself `from pymeasure.instruments.manufacturer.device import *`.

bmoneke, 2022-11-17:
### Difficulties

I did some tests with enums and found some difficulties:

- Using `cast=SomeEnum` requires the enum to be known before the Instrument is defined. Therefore, it cannot be defined in the Instrument itself, but globally, you can, however use `get_process=lambda v: SomeEnum(v)`.
- `StrEnum` is a python 3.11 feature, therefore `set_process=lambda v: v.value` is necessary for `Enum`s defining strings.
- If you use `set_process=lambda v: v.value` (as necessary for plain `Enum`), you cannot use strings anymore. (if you require an IntFlag, the integers themselves are valid values as well.
- In the getter, IntEnum (and IntFlag) require ints, therefore you have to cast to int before.

### Example

Here is a working example of some enums and an instrument:
~~~
class MODES(enum.Enum):
    VOLTAGE = "MV"
    CURRENT = "MC"

class EnumInstrument(Instrument):

    def __init__(self, adapter, *args, **kwargs):
        super().__init__(adapter, "Test", *args, **kwargs)

    class OPTIONS(enum.IntEnum):
        ONE = 1
        SEVEN = 7

    class STATUS(enum.IntFlag):
        POWER_SHORTAGE = 1
        OVER_CURRENT = 2
        INTERLOCK = 4

    class WAVEFORMS(enum.Enum):
        FUNNY = "FY"
        SINE = "sn"
        TRIANGLE = "tr"

    mode = Instrument.control("M?", "M %s", """Control the mode with a global enum.""",
                              cast=MODES,
                              set_process=lambda v: v.value)

    option = Instrument.control("O?", "O %d", """Control an integer option with an IntEnum.""",
                                get_process=lambda v: EnumInstrument.OPTIONS(v))

    status = Instrument.measurement("STB?", """Measure a bitwise status, returns IntFlag.""",
                                    cast=int,
                                    get_process=lambda v: EnumInstrument.STATUS(v))

    waveform = Instrument.control("WAVE?", "WAVE %s", """Control a waveform with Enum.""",
                                  set_process=lambda v: v.value,
                                  get_process=lambda v: EnumInstrument.WAVEFORMS(v))
~~~

### Usage hints:

- Use `get_process=lambda v: SomeEnum(v)` if you want to return an Enum.
- Use additionally `cast=int` if you want to use an integer based Enum.
- Use `set_process=lambda v: v.value`, if you do use a plain Enum (not IntEnum or StrEnum)
- If you have a globally defined Enum (i.e. not in the same Instrument defined), you can use `cast=SomeEnum` instead of `get_process`, however, the members of `SomeEnum` have to be strings.

### Questions regarding best use

- Do we propose enums at module or class level? In the first case, do we import them to the main namespace?

msmttchr, 2022-11-17:
@bmoneke, based on your explanation, it seems very tricky to use enum properly.
I see the long explanation as an alternative of the short explanation: "don't use enum" :)
Jokes apart, I wonder if the `set_process=lambda v: v.value` can be avoided and add a check inside the code of `Instrument.control` something like:
```
if isinstance(value, Enum):
   value = value.value
```
I am not sure if we can get rid of the `get_process` as well.
Regarding the other questions:
I would select class level, since name may be too generic to be imported in the global namespace.
Possibly following python documentation style may be best.

bmoneke, 2022-11-17:
The goal was, to find out, how we could use enums without changing code. Now we can look for ways to improve it.

> @bmoneke, based on your explanation, it seems very tricky to use enum properly. I see the long explanation as an alternative of the short explanation: "don't use enum" :) Jokes apart, I wonder if [...]

The initial idea was, to not introduce to much logic. However, as StrEnum will do that job transparently, the code won't have to stay forever, just until 3.10 won't be supported anymore (i.e. still some time to go).

> I am not sure if we can get rid of the `get_process` as well. Regarding the other questions: I would select class level, since name may be too generic to be imported in the global namespace. Possibly following python documentation style may be best.

I found another (complex and not very elegant) way to get `cast` to work: We define a class containing enums and then the instrument under scrutiny inherits from that class es well:
~~~
class Enums:
    class Test(enum.Enum):
        ABC = "test"

class EnumInstrument(Instrument, Enums):
    test = Instrument.measurement("T?", """Testing""", cast=Enums.Test)
~~~
This works and you could (in case of a control), use also `inst.Test.ABC` for setting some value. However, as an IntFlag (very useful for status bytes) requires a cast to int beforehand, get_process would be necessary again.

Now the question is: Do we want to propose the properties from my example (which are not larger than the ones they replace with a mapped dictionary etc.) or do we want to change the property creators?
In #733 I had already started some ideas of how to change the property creators.

If we want to do more than tests and documentation (scope of this issue), we should use the other issue.

msmttchr, 2022-11-17:
I am not sure about @bilderbuchi opinion, but from my point of view, it is too complex to use Enum as of today (I didn't realize it prior to your explanation). So, I would be inclined to say to change property creators and drop this documentation effort.

bilderbuchi, 2022-11-18:
> * Using `cast=SomeEnum` requires the enum to be known before the Instrument is defined.

Yes, this is what I meant above
> at import time, not all attributes and scopes are set up yet, 

> In the getter, IntEnum (and IntFlag) require ints, therefore you have to cast to int before.

You sure about this? I just tried, and for floats that convert to ints, this just works:
``` 
In [2]: class bla(IntEnum):
   ...:     a = 3
   ...: 

In [3]: bla(3)
Out[3]: <bla.a: 3>

In [4]: bla(3.0)
Out[4]: <bla.a: 3>
```

>    Do we propose enums at module or class level? In the first case, do we import them to the main namespace?

I think it's fine to continue using enums as we already do in the codebase, at module level (primarily it's `IntEnum` and `IntFlag` I think). It would be nicer at class level, but in the end this is a language limitation. The drawbacks are contained because the module they end up in is typically a model-specific one containing only the class, so we don't wildly pollute namespaces.

I would not import them into the manufacturer's main namespace (like we do the instruments), because then they stay associated with the correct instrument, which should further limit any potential confusion.

>    Naming scheme: Enum in CamelCase (like all classes) and members in UPPERCASE_WITH_UNDERSCORE (similar to the python enum documentation)?

Yes

bilderbuchi, 2022-11-18:
Ha, nice to learn about StrEnum, I was not aware that that existed! I think we should mostly be fine because we primarily use `IntEnum` and `IntFlag`, which transparently behave like ints, so aside from the module-level definition, it's imo not that ugly or complicated to use. I'd rather avoid wrapping this into lambdas just to be able to put them in a class.

> I found another (complex and not very elegant) way to get cast to work: We define a class containing enums and then the instrument under scrutiny inherits from that class es well

Yes, that's a possibility. It might be quite natural if there is already a base class for a model family or whatnot. Or maybe a plain `Config` class that the instrument also inherits from, with the enums as attributes. Possibly more care w.r.t. MRO and init and super calls might be necessary, so not sure I want to establish multiple inheritance in the docs just yet.

bilderbuchi, 2022-11-18:
> Now the question is: Do we want to propose the properties from my example (which are not larger than the ones they replace with a mapped dictionary etc.) or do we want to change the property creators?

I'm not sure I understand. Those questions seem more connected with #733?
AFAIUI, the topic here was a better way of encapsulating enums, and it turns out this does not work as we thought in the OP, so we can't really encapsulate better, so we leave it as-is?

bmoneke, 2022-11-18:
> In the getter, IntEnum (and IntFlag) require ints, therefore you have to cast to int before.

I got error messages, that "3.0 is not an int", but your example works on my private computer as well. I don't know

> I'm not sure I understand. Those questions seem more connected with #733?

Exactly. That was my introduction to raise the question, whether we want to talk about the property creator changes (which I later point to that issue) or we just write some documentation.

I guess I misunderstood you. I understood "encapsulating" as including in the class **and** in the property creators, while you looked at the first question only.

bmoneke, 2022-11-18:
Another option: you define the enums at module level, but add them to the instrument class as well.

bmoneke, 2022-11-23:
I just found out (thanks to @AidenDawn ), that `cast=SomeEnum` works, if the members are strings. You just have to address them properly (see the edited example and property `waveforms2`), i.e. not as a class attribute but directly.